zephyrproject-rtos/zephyr #81713: Zephyr 4.1 Release Checklist

fabiobaltieri, 2024-11-21:
#### As soon as possible
- [] Ensure there is a draft `doc/releases/release-notes-4.0.rst` and `doc/releases/migration-guide-4.0.rst` with feature summary from the previous one and if not create one (previous #69418)
- [] Ensure MAINTAINERS.yml "Release Notes"'s maintainers to point at the current release owners (#76876)
- [] Set the current release managers as assignee's in https://github.com/zephyrproject-rtos/zephyr/issues/76943 and ensure it is linked from the [release plan](https://github.com/orgs/zephyrproject-rtos/projects/13).

#### From day one but with extra focus as we approach RC1 and after:
- [] Check the CI [bi-weekly](https://github.com/zephyrproject-rtos/zephyr/actions/workflows/twister.yaml?query=event%3Aschedule) run for build failures that should be fixed and report or fix them.
- [] Make sure the release plan is kept up to date:
   - See https://github.com/orgs/zephyrproject-rtos/projects/13
- [] Ensure the [release management page](https://github.com/zephyrproject-rtos/zephyr/wiki/Release-Management) has an up-to-date schedule for the release

#### One month before RC1:
- [] Ping west project maintainers to remind them of the RC1/freeze deadline and to consider if their modules should be updated (you may want to extract all west projects' maintainers names from the MAINTAINERS files and create an issue like) (previous https://github.com/zephyrproject-rtos/zephyr/issues/73834)
- [] Start giving heads ups on the TSC meeting about the incoming freeze.

#### Two weeks before RC1:
- [] Send a heads ups about the incoming freeze to dev-mailing list and Discord General channel

#### Pre RC1
   - [] Create next release (v4.2.0) milestone so that features that don't make it to the release can be assigned a milestone
   - [] Verify that all security-critical external components and external dependencies are up to date 
     - (3.7) https://github.com/zephyrproject-rtos/zephyr/issues/73703
     - (3.7) https://github.com/zephyrproject-rtos/zephyr/issues/73878
     - (4.0) https://github.com/zephyrproject-rtos/zephyr/issues/80489
     - (4.0) https://github.com/zephyrproject-rtos/zephyr/issues/80491
   - [] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as un-exploitable (ask Security working group).
   - [] Make sure there's no pending ["Release Blocker"](https://github.com/zephyrproject-rtos/zephyr/issues?q=is%3Aissue+is%3Aopen+label%3A%22Release+Blocker%22) issues
   - [] Ensure that the latest Zephyr SDK contains all the changes necessary to support the new features included in, or planned for this release; if not, ping the SDK maintainer (@stephanosio) to release a new SDK version.
       - [] Check if any toolchain-related enhancements and bug fixes are required for any of the pull requests targeting this release. If any of these enhancements and bug fixes are not part of the latest Zephyr SDK, a new SDK version must be released.
       - [] Check that the Picolibc version in the latest SDK matches the Picolibc module version in `west.yml`. If they do not match, a new SDK version must be released. (ping @stephanosio and @keith-packard )
   - [ ] Create a "Deprecated APIs/options to be removed by x.y" issue for 4.3 (N+3), linking to it from & to the 4.2 one (https://github.com/zephyrproject-rtos/zephyr/issues/77070) . Remember to set the `deprecation_tracker` label.

See https://docs.zephyrproject.org/latest/project/release_process.html#tagging for tagging details, release specific filters under https://github.com/zephyrproject-rtos/zephyr/wiki/Filters to find pending PRs.

- [] Release Notes and Migration Guide WIP
  - [] Make sure to tell everybody to
    * Use the **past tense** (i.e. "X has been added" or "X is now Y")
    * Update `doc/releases/release-notes-4.0.rst` detailed sections (relative to 3.7.0 release) as important issues are addressed (subsystem owners). (previous: https://github.com/zephyrproject-rtos/zephyr/issues/54495)
  - [] Be sure to remind maintainers to update the release notes, at the very least in the rc1, rc2 and rc3 emails. But you may need to create a GH issue to follow up.

#### After RC3, before final release

- [] Finalize Release Notes and Migration Guide: (previous: #76113, #76295 )
  - [] Review `doc/releases/release-notes-4.0.rst` overview summary and details sections
    - [] Request summary details from subsystem maintainers
- [] Add new version in `doc/conf.py`, `doc/releases/index.rst`, and `doc/releases/eol_releases.rst` (previous  #76047)
- [] Remove EOL version(s) in `doc/conf.py`, `doc/releases/index.rst`, and `doc/releases/eol_releases.rst`  : (previous: #69404) .
- [] Update `doc/LICENSING.rst` for new components that do not have a license: (previous: #69392)
- [] Remaining Release Notes (previous #75933)
- [] Update version in VERSION ([EXTRAVERSION must be empty, not zero 0](https://docs.zephyrproject.org/latest/project/release_process.html#tagging)) (previous #76047 )
- [] Tag v4.1.0 (see https://docs.zephyrproject.org/latest/project/release_process.html#tagging)
- [] Update and publish the [draft GitHub release](https://github.com/zephyrproject-rtos/zephyr/releases) created by the CI.
- [] Tag v4.1.0 in [example-application](https://github.com/zephyrproject-rtos/example-application) (previous: https://github.com/zephyrproject-rtos/example-application/pull/73) and create github release from tag.

### After Release
  - [] Update patchlevel in VERSION (PATCHLEVEL = 99) (e.g. https://github.com/zephyrproject-rtos/zephyr/pull/76334) and tag `v4.1.99` (https://github.com/zephyrproject-rtos/zephyr/pull/81474)
  - [] Create `v4.0-branch` branch for 4.0.x releases (from v4.0.0 tag, don't include the switch to patchlevel 99)
  - [] Create `backport v4.0-branch` label
  - [] Move all PRs/Issues in the v4.0.0 milestone to the next milestone
  - [] Close v4.0.0 milestone 
  - [] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release and announce the opening of the merge window
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [] Marketing Blog for announcing release (https://www.zephyrproject.org/blog / https://www.zephyrproject.org/announcements/)
  - [] Post an update in https://github.com/zephyrproject-rtos/zephyr/discussions under General, have one of our administrators move it to Announcements and pin it.
    - []  (3.7 https://github.com/zephyrproject-rtos/zephyr/discussions/76336) 
    - [] (4.0: https://github.com/zephyrproject-rtos/zephyr/discussions/81499)
  - [] Post an announcement in discord in the #general & #release channels
  - [] Update MAINTAINERS.yml "Release Notes" maintainers to point at the next release owners (https://github.com/zephyrproject-rtos/zephyr/pull/81669)
  - [] Update current and supported version information in the [Security Policy](https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/SECURITY.md). (https://github.com/zephyrproject-rtos/zephyr/pull/81614)
  - [] If you're a first time release manager, you may contact Tracey (or [info@zephyrproject.org](mailto:info@zephyrproject.org)) to claim your Release Manager badge.
- [] Create draft `doc/releases/release-notes-4.1.rst` and `doc/releases/migration-guide-4.1.rst` with feature summary from the previous one (if not there already) #76344

Previous checklist: #76338

fabled, 2024-11-22:
@fabiobaltieri Seems you assigned this me in mistake, no?

fabiobaltieri, 2024-11-22:
> @fabiobaltieri Seems you assigned this to me in mistake, no?

woops, yes that was meant to go to myself :-) thanks

closed 2025-03-10 by fabiobaltieri