zephyrproject-rtos/zephyr #81716: Docs: Enums should be shown with one value per line

Thalley, 2024-11-21:
**Is your enhancement proposal related to a problem? Please describe.**
Today enums are just a long string it seems, which isn't very readable. 

For example at https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html we have 
[image]

If you go to a specific enum it's show as 
[image]

which is much more readable. 

**Describe the solution you'd like**
If we want to keep the enum declarations like above, then instead of 
```
[bt_audio_codec_cap_type](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#gaf58823a56eb36cfd22e582ab1ea3d015) {
  [BT_AUDIO_CODEC_CAP_TYPE_FREQ](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015af68f3f5665446de9b71c2f70e7aa762b) = 0x01 , [BT_AUDIO_CODEC_CAP_TYPE_DURATION](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015a4a90a37cd6d8ca9d1ed984e4a122d510) = 0x02 , [BT_AUDIO_CODEC_CAP_TYPE_CHAN_COUNT](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015a0cd3a3d31febc9e30137e9e21d4f7191) = 0x03 , [BT_AUDIO_CODEC_CAP_TYPE_FRAME_LEN](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015ad074bed5f79946e2388584336d30cf46) = 0x04 ,
  [BT_AUDIO_CODEC_CAP_TYPE_FRAME_COUNT](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015acba980896f080ac2ba81d1b66305bb41) = 0x05
}
```
it should be
```
bt_audio_codec_cap_type](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#gaf58823a56eb36cfd22e582ab1ea3d015) {
  [BT_AUDIO_CODEC_CAP_TYPE_FREQ](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015af68f3f5665446de9b71c2f70e7aa762b) = 0x01 ,
 [BT_AUDIO_CODEC_CAP_TYPE_DURATION](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015a4a90a37cd6d8ca9d1ed984e4a122d510) = 0x02 ,
 [BT_AUDIO_CODEC_CAP_TYPE_CHAN_COUNT](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015a0cd3a3d31febc9e30137e9e21d4f7191) = 0x03 ,
 [BT_AUDIO_CODEC_CAP_TYPE_FRAME_LEN](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015ad074bed5f79946e2388584336d30cf46) = 0x04 ,
  [BT_AUDIO_CODEC_CAP_TYPE_FRAME_COUNT](https://docs.zephyrproject.org/latest/doxygen/html/group__bt__audio.html#ggaf58823a56eb36cfd22e582ab1ea3d015acba980896f080ac2ba81d1b66305bb41) = 0x05
}

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

kartben, 2024-11-21:
What you're suggesting is doable, but in the interest of keeping things more concise in the "summary" section, how about just dropping the enum values, since can always click to get the nicely formatted (and more detailed) table?

[image]

Your proposition/ask (note how this takes A LOT more vertical space, with limited added value IMO)

[image]

Thalley, 2024-11-21:
Yeah, that works too, since we have a nice overview when clicking on a specific enum.

Not sure whether not having the literal values visible is a downside for some users though

kartben, 2025-06-19:
TBH I am tempted to close as WONTFIX, yes the UI/UX could be improved but given there's at least one place where enums are listed in a very well organized way, I think we can live with it.