zephyrproject-rtos/zephyr #81722: Bluetooth: ISO: Remove CONFIG_BT_ISO_TX_MTU

Thalley, 2024-11-21:
**Is your enhancement proposal related to a problem? Please describe.**
`CONFIG_BT_ISO_TX_MTU` today is not used by the host. 

It has 3 uses
1) Many samples and tests are using this value to define their TX buffer pools
2) hci_raw.c uses it to define the size of the `hci_iso_pool`. 
3) The controller optionally uses it to determine the HCI ISO TX buffer size, but `CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE` can be directly for that too

**Describe the solution you'd like**
Remove the Kconfig as the host doesn't it and the applications shouldn't. For hci_raw and the HCI ISO tx buffer sizes we can/should use `CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE`. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Thalley, 2024-11-26:
PR adding the fragmented test: https://github.com/zephyrproject-rtos/zephyr/pull/81871

cvinayak, 2025-02-18:
PR #82026 may address failure in #81871