zephyrproject-rtos/zephyr #81733: NXP FRDM-MCXA156: no UART RX

DerekSnell, 2024-11-21:
LPUART0, used for the debug console and shell, does not receive characters.  It does transmit, and will print to the terminal.

To replicate, the sample samples/drivers/uart/echo_bot shows the issue.  It prints to the terminal, but will not echo back any characters.  Perhaps there is an issue with the RX pin.

I first noticed this trying to run the flash_shell sample.  And then found that I did not see UART RX work in any sample I tested.  I tested with commit https://github.com/zephyrproject-rtos/zephyr/commit/74bb5b334db1749d9ff30218d89e922f1409849f when this board was first introduced, and it has the same issue.  

@NeilChen93 , can you review this issue?

NeilChen93, 2024-11-22:
Hi @DerekSnell , get it, I will check and fix this issue.

ehughes, 2024-11-24:
@DerekSnell  @NeilChen93 I am bringing up an MCXA156 (BGA version)  now (but saw this on the FRDM board as well.

The issue is is that the input buffer enable bit is not getting set.
```
#define A15X_MUX(port, pin, mux)		\
	(((((port) - '0') & 0xF) << 28) |	\
	(((pin) & 0x3F) << 22) |		\
	(((mux) & 0xF) << 8))
```

Bit 12 needs set to enable the input buffer.

I did a quick overlay like this:

`&pinctrl {
	pinmux_lpuart0: pinmux_lpuart0 {
		group0 {
			pinmux = <(LPUART0_RXD_P0_2 | (1<<12))>,
					 <LPUART0_TXD_P0_3>;
			drive-strength = "low";
			slew-rate = "fast";
		};
	};

};`

for a quick fix.

DerekSnell, 2024-11-25:
Hi @ehughes ,
@NeilChen93 already has a PR submitted at https://github.com/zephyrproject-rtos/zephyr/pull/81752.
Best regards

closed 2024-11-27 by fabiobaltieri