zephyrproject-rtos/zephyr #81744: Sample WiFi:apsta_mode didn't work with ESP32S3

kien-exact, 2024-11-22:
**Describe the bug**
I am running a dual mode (AP and STA) function test with the ESP32S3. However, it encounters an issue on the DHCPV4 Server side.
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378ca5
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc95ab0,len:0x3b40
load:0x40374000,len:0x11a90
SHA-256 comparison failed:
Calculated: e969b2232861ba27434428cd9a81d3779c983520abb3ef697b278bc2e40e97c2
Expected: 00000000f0a90000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x4037c7b0
I (100) boot: ESP Simple boot
I (100) boot: compile time Nov 22 2024 11:07:27
W (100) boot: Unicore bootloader
I (101) spi_flash: detected chip: gd
I (102) spi_flash: flash io: dio
W (105) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (121) boot: chip revision: v0.2
I (121) boot.esp32s3: Boot SPI Speed : 40MHz
I (124) boot.esp32s3: SPI Mode       : DIO
I (128) boot.esp32s3: SPI Flash Size : 8MB
I (132) boot: Enabling RNG early entropy source...
I (136) boot: DRAM: lma 0x00000020 vma 0x3fc95ab0 len 0x3b40   (15168)
I (143) boot: IRAM: lma 0x00003b68 vma 0x40374000 len 0x11a90  (72336)
I (149) boot: padd: lma 0x00015608 vma 0x00000000 len 0xa9f0   (43504)
I (155) boot: IMAP: lma 0x00020000 vma 0x42000000 len 0x4ff18  (327448)
I (162) boot: padd: lma 0x0006ff20 vma 0x00000000 len 0xd8     (216)
I (168) boot: DMAP: lma 0x00070000 vma 0x3c050000 len 0xc7b4   (51124)
I (174) boot: Image with 6 segments
I (177) boot: DROM segment: paddr=00070000h, vaddr=3c050000h, size=0C7C0h ( 51136) map
I (185) boot: IROM segment: paddr=00020000h, vaddr=42000000h, size=4FF16h (327446) map
I (228) heap_runtime: ESP heap runtime init at 0x3fcad628 size 240 kB.

[00:00:00.232,000] <inf> wifi_init: rx ba win: 6
[00:00:00.232,000] <inf> wifi_init: WiFi IRAM OP enabled
*** Booting Zephyr OS build v4.0.0 ***
[00:00:05.234,000] <inf> MAIN: Turning on AP Mode
[00:00:05.235,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: Started DHCPv4 server, address pool:
[00:00:05.235,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:     0: 192.168.4.11
[00:00:05.235,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:     1: 192.168.4.12
[00:00:05.235,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:     2: 192.168.4.13
[00:00:05.235,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:     3: 192.168.4.14
[00:00:05.235,000] <inf> MAIN: DHCPv4 server started...

[00:00:05.267,000] <err> esp32_wifi: Failed to send packet
[00:00:05.268,000] <err> esp32_wifi: Failed to send packet
[00:00:05.268,000] <err> esp32_wifi: Failed to send packet
[00:00:05.268,000] <err> esp32_wifi: Failed to send packet
[00:00:05.269,000] <inf> MAIN: AP Mode is enabled. Waiting for station to connect
[00:00:05.269,000] <inf> MAIN: Connecting to SSID: SSID

[00:00:05.271,000] <err> esp32_wifi: Failed to send packet
[00:00:05.272,000] <err> esp32_wifi: Failed to send packet
[00:00:05.272,000] <err> esp32_wifi: Failed to send packet
[00:00:05.272,000] <err> esp32_wifi: Failed to send packet
[00:00:05.273,000] <err> esp32_wifi: Wi-Fi not in station mode
[00:00:05.273,000] <inf> MAIN: Connected to SSID
[00:00:05.273,000] <err> MAIN: Unable to Connect to (SSID)
[00:00:06.269,000] <err> esp32_wifi: Failed to send packet
[00:00:06.273,000] <err> esp32_wifi: Failed to send packet
[00:00:07.269,000] <err> esp32_wifi: Failed to send packet
[00:00:07.273,000] <err> esp32_wifi: Failed to send packet
[00:00:10.239,000] <inf> MAIN: station: 42:31:98:67:F1:5D joined
[00:00:10.655,000] <err> esp32_wifi: Failed to send packet
[00:00:10.655,000] <dbg> net_dhcpv4_server: dhcpv4_handle_discover: DHCPv4 processing Discover - reserved 192.168.4.11
[00:00:11.657,000] <err> esp32_wifi: Failed to send packet
[00:00:11.663,000] <err> esp32_wifi: Failed to send packet
[00:00:11.663,000] <dbg> net_dhcpv4_server: dhcpv4_handle_discover: DHCPv4 processing Discover - reserved 192.168.4.11
[00:00:13.570,000] <err> esp32_wifi: Failed to send packet
[00:00:13.570,000] <dbg> net_dhcpv4_server: dhcpv4_handle_discover: DHCPv4 processing Discover - reserved 192.168.4.11
[00:00:17.620,000] <err> esp32_wifi: Failed to send packet
[00:00:17.620,000] <dbg> net_dhcpv4_server: dhcpv4_handle_discover: DHCPv4 processing Discover - reserved 192.168.4.11
[00:00:25.418,000] <err> esp32_wifi: Failed to send packet
[00:00:25.418,000] <dbg> net_dhcpv4_server: dhcpv4_handle_discover: DHCPv4 processing Discover - reserved 192.168.4.11
[00:00:28.503,000] <inf> MAIN: station: 42:31:98:67:F1:5D leave
[00:00:55.418,000] <dbg> net_dhcpv4_server: dhcpv4_server_timeout: Address 192.168.4.11 expired
[00:00:55.419,000] <dbg> net_dhcpv4_server: dhcpv4_server_timeout_recalc: No more addresses, canceling timer
```
**To Reproduce**
Simply build the settings example and run it:
```
west build -p always zephyr/samples/net/wifi/apsta_mode -b esp32s3_devkitm/esp32s3/procpu && west flash && west espressif monitor
```

**Expected behavior**
The ESP32S3 should create a connection with the SSID: ESP32-AP and allow connections (successfully assigning IP).

**Impact**
Version v3.7.0 runs quite stable with the features of Access Point & DHCPV4 server. When upgrading to version v4.0.0, it encounters this issue, and I think it significantly affects programs that need to use the DHCPV4 server.

**Environment (please complete the following information):**
 - OS: MacOS, Linux or Windows
 - Toolchain: Zephyr SDK v0.16.8
 - Zephyr v4.0.0

kien-exact, 2024-11-27:
Hi @sylvioalves 
I would like to add a few points that I found.
- The AP feature doesn't seem to be working in the current version 4.0.0.
- I created a basic example for initializing AP through a static IP, but stations still cannot connect.
Here is my sample app.
[esp-apsta.zip](https://github.com/user-attachments/files/17927411/esp-apsta.zip)

kaessert, 2024-11-30:
I'm having trouble with this example and the station mode as well. I'm seeing `esp32_wifi: Failed to send packet` a few times and it doesn't get an ip. Worked fine with 3.7.

elXaptation, 2025-01-07:
I ran into this issue last night. Searched and found this report.

My redimentry debugging seems to indicate esp_wifi_drv.c is receiving an empty data pointer for (at least) dhcp related packets.

What sends data to esp_wifi_drv.c?

Was gonna try 3.7.0 but couldn't figure out how to roll my env back without starting from scratch or totally breaking my env. Found some docs from 2021.. but didn't have a high level of confidence in them being up to date. Any pointers would be appreciated.  Sorry.. I'm just learning about all this.

andi-exact, 2025-01-07:
@elXaptation I heard on the Zephyr Discord that the issues with concurrent WiFi AP and STA might be fixed on `main`. I haven't had a chance to test this yet, but you are welcome to try the main branch and see if the issue is fixed.

osmakari, 2025-02-02:
Seems like esp_wifi_drv.c runs the API init function `esp32_wifi_init` first for the STA device and sets `esp32_wifi_iface_ap` to the station's interface. This causes the driver to check against the wrong device's status, resulting in "Wi-Fi not in station mode" error. Changing the instance numbers between the two `NET_DEVICE_DT_INST_DEFINE` macros fixed it for me:

```
...
NET_DEVICE_DT_INST_DEFINE(1,  // This was 0 
		esp32_wifi_dev_init, NULL,
		&esp32_data, NULL, CONFIG_WIFI_INIT_PRIORITY,
		&esp32_api, ETHERNET_L2,
		NET_L2_GET_CTX_TYPE(ETHERNET_L2), NET_ETH_MTU);

#if defined(CONFIG_ESP32_WIFI_AP_STA_MODE)
NET_DEVICE_DT_INST_DEFINE(0, // This was 1
		NULL, NULL,
		&esp32_ap_sta_data, NULL, CONFIG_WIFI_INIT_PRIORITY,
		&esp32_api, ETHERNET_L2,
		NET_L2_GET_CTX_TYPE(ETHERNET_L2), NET_ETH_MTU);
...
```

PR #85220 (2025-02-18): drivers: wifi: esp32: Fix AP-STA mode interface init order
Wi-Fi AP-STA mode fails on ESP32 due to incorrect interface initialization. In AP-STA mode, esp32_wifi_init first sets the driver's AP interface, followed by the STA interface. However, the driver initializes them in reverse order, causing both to read incorrect device data and resulting in a "Wi-Fi not in station mode" error.
This fix handles STA and AP initialization in separate functions and checks for correct mode in esp32_wifi_status. Tested on a custom ESP32-S3-WROOM-1-N4R2 board, where both STA and AP can connect, send, and rece [... cut]
closed 2025-02-18 by kartben