zephyrproject-rtos/zephyr #81791: SensorTile.box PRO I2C node disables USB output

rmarcus1tul, 2024-11-22:
**Test Conditions**
Hardware
- STEVAL-MKBOXPRO (sensortile_box_pro dts)
- Micro-USB cable

Software
- Zephyr 3.7.0 and 4.0.0
- sensors-on-board sample
- Ubuntu 24.04 and Windows 11 OS
- dfu-util for flashing

**Background**
Using the default configuration for the SensorTile.box PRO with its associated sensors-on-board demo, no USB device is detected. I'm testing on Ubuntu but also confirmed USB virtual COM is disabled when attempting on Windows. 

I traced this to some conflict between the USB driver and the I2C sensors. Adding an overlay file with the 3 sensors on I2C1 disabled re-enables USB:
```
&{/soc/i2c@40005400/lps22df@5d} {
    status = "disabled";
};

&{/soc/i2c@40005400/lis2mdl@1e} {
    status = "disabled";
};

&{/soc/i2c@40005400/stts22h@38} {
    status = "disabled";
};
```

No errors or warnings are flagged when building and there is no indication of a conflict. I tried to get UART output using a JLink Base with VCOM enabled, and following the instructions in [the README](https://docs.zephyrproject.org/latest/samples/boards/st/sensortile_box_pro/sensors-on-board/README.html), but I could not get an output even on a functional build with the I2C sensors disabled. I then attempted to get outputs over RTT but `CONFIG_USE_SEGGER_RTT=y` produces compilation errors.

avisconti, 2025-02-18:
@rmarcus1tul what happens if you build and run hello_world sample? I would assume that this print fine on USB console, correct?

```west -b sensortile_box_pro samples/hello_world```

avisconti, 2025-02-18:
@rmarcus1tul you are mentioning ```Micro-USB cable```, but sensortile_box_pro has a type-c connector. Can you confirm you are using the [sensortile_box_pro](https://docs.zephyrproject.org/latest/boards/st/sensortile_box_pro/doc/index.html#sensortile_box_pro) and not the old [sensortile_box](https://docs.zephyrproject.org/latest/boards/st/sensortile_box/doc/index.html) ?

Sorry for the stupid question, but, again, I'm trying to figure out how it is possible to have this strange behaviour...

closed 2025-08-14 by github-actions[bot]