zephyrproject-rtos/zephyr #81797: west: can no longer flash stm32 target

cfriedt, 2024-11-22:
**Describe the bug**

Previously, I was able to flash a board with west. Now I cannot.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? `nucleo_l496zg`
 - What have you tried to diagnose or workaround this issue? bisect
 - Is this a regression? Yes 133665e55a707e6eb58425298779e3b3d680d319

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p -b nucleo_l496zg samples/hello_world`
2. `west flash`
3. See error

**Expected behavior**
The board is flashed.

**Impact**
This is a showstopper for testing on HW.

**Logs and console output**
```shell
Traceback (most recent call last):
  File "/opt/zephyr/.venv/lib/python3.12/site-packages/west/commands.py", line 559, in __call__
    mod = _commands_module_from_file(self.py_file)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/zephyr/.venv/lib/python3.12/site-packages/west/commands.py", line 714, in _commands_module_from_file
    spec.loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/cfriedt/zephyrproject/zephyr/scripts/west_commands/flash.py", line 11, in <module>
    from run_common import add_parser_common, do_run_common, get_build_dir
  File "/home/cfriedt/zephyrproject/zephyr/scripts/west_commands/run_common.py", line 37, in <module>
    from runners import get_runner_cls, ZephyrBinaryRunner, MissingProgram
ImportError: cannot import name 'MissingProgram' from 'runners' (/home/cfriedt/zephyrproject/zephyr/scripts/west_commands/runners/__init__.py)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/zephyr/.venv/lib/python3.12/site-packages/west/app/main.py", line 561, in run_command
    self.run_extension(args.command, argv)
  File "/opt/zephyr/.venv/lib/python3.12/site-packages/west/app/main.py", line 685, in run_extension
    self.cmd = self.extensions[name].factory()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/zephyr/.venv/lib/python3.12/site-packages/west/commands.py", line 561, in __call__
    raise ExtensionCommandError(
west.commands.ExtensionCommandError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/zephyr/.venv/bin/west", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/zephyr/.venv/lib/python3.12/site-packages/west/app/main.py", line 1176, in main
    app.run(argv or sys.argv[1:])
  File "/opt/zephyr/.venv/lib/python3.12/site-packages/west/app/main.py", line 255, in run
    self.run_command(argv, early_args)
  File "/opt/zephyr/.venv/lib/python3.12/site-packages/west/app/main.py", line 605, in run_command
    self.handle_extension_command_error(ece)
  File "/opt/zephyr/.venv/lib/python3.12/site-packages/west/app/main.py", line 431, in handle_extension_command_error
    self.cmd.err(msg, fatal=True)
    ^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'err'
```

**Environment (please complete the following information):**

 - OS: Ubuntu Linux 24.04
 - Toolchain: Zephyr SDK v0.17.0
 - Commit SHA: 054d60fb1c39db0f9884dd3f480425bae70f3ce4

**Additional context**
https://discord.com/channels/720317445772017664/733037524498382881/1309648854064566313

walidbadar, 2024-11-23:
I think this issue has affected all boards, I am unable to flash esp32 targets.

pdgendt, 2024-11-23:
> @pdgendt I think this issue has affected all boards, I am unable to flash esp32 targets.

https://github.com/zephyrproject-rtos/zephyr/pull/81810 should fix it

PR #81810 (2024-11-23): scripts: west_commands: Fix MissingProgram import
`MissingProgram` should be exported in `__all__`
Fixes #81797
closed 2024-11-23 by kartben