zephyrproject-rtos/zephyr #81802: rtio: completion queue event count overflow

JordanYates, 2024-11-23:
**Describe the bug**

The internal counter used to keep track of completion queue accounts is only ever incremented, never decreasing.

https://github.com/zephyrproject-rtos/zephyr/blob/3d6dde483234a69ffb80845a58bad8bf294b45b7/include/zephyr/rtio/rtio.h#L1180

This counter is used by `rtio_submit` to determine when the submission has completed:

https://github.com/zephyrproject-rtos/zephyr/blob/3d6dde483234a69ffb80845a58bad8bf294b45b7/include/zephyr/rtio/rtio.h#L1453
https://github.com/zephyrproject-rtos/zephyr/blob/3d6dde483234a69ffb80845a58bad8bf294b45b7/include/zephyr/rtio/rtio.h#L1473-L1476

A typical RTIO I2C transaction might like the following:
1. `rtio_sqe_prep_tiny_write`
2. `rtio_sqe_prep_read`
3. rtio_sqe_prep_callback_no_cqe`

So 2 CQE events per I2C transaction. If there were 50 I2C transactions per second, this counter would overflow in ~250 days (2**31 / 50 / 2 / 86400).

Increasing or decreasing the I2C transaction rate would scale the overflow duration linearly.
At overflow, the second check will pass immediately:
```
cq_count = INT32_MAX + 1; // INT32_MIN
while(INT32_MAX < INT32_MIN) {
   // Never enters
}
```

**Expected behavior**

Internal counter overflow should either be detected and handled, or reset in some way through `atomic_dec`.

**Environment (please complete the following information):**

 - Zephyr v4.0

teburd, 2024-11-23:
We need the counter to know when to return, so detecting the rollover is expected and should already work. Agreed a test should show this working

JordanYates, 2024-11-23:
> We need the counter to know when to return, so detecting the rollover is expected and should already work. Agreed a test should show this working

Testcase was added for this situation which fails with `rtio_submit` returning immediately without waiting for the completions: #81820

teburd, 2024-11-23:
Each struct rtio should be used from one, only one context for submitting. Like wise for checking on and getting completions. They do not have to be the same call context. There’s unfortunately no great way of enforcing this lifetime of the struct.

PR #85079 (2025-02-12): rtio: Fix completion count overflow issue
As noted by @JordanYates in #81802 cq_count being overflowed wasn't properly dealt with. This now correctly accounts for the completion count overflowing and wrapping.

fixes #81802
closed 2025-02-12 by kartben