zephyrproject-rtos/zephyr #81824: twister: snippets ignored in filtering

JordanYates, 2024-11-24:
**Is your enhancement proposal related to a problem? Please describe.**

Currently `required_snippets` are not considered when determining whether a given testcase passes filters.
As a trivial example, modifying `zephyr/samples/hello_world/sample.yaml` to the following:
```
tests:
  sample.demo:
    filter: CONFIG_RTT_CONSOLE
    required_snippets:
      - rtt-console
```
Will result in all platforms being skipped, despite the symbol being set by the snippet:
https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd158e7cec50761fece73b1d4835a2e/snippets/rtt-console/rtt-console.conf#L3

```
./zephyr/scripts/twister -T ./zephyr/samples/hello_world/ --platform nrf52840dk/nrf52840 -v
...
INFO    - 1/1 nrf52840dk/nrf52840       samples/hello_world/sample.demo                    SKIPPED (runtime filter)
```

This also applies to devicetree filtering.

**Describe the solution you'd like**

`filter` parameters should pass when the required options/nodes are provided by a required snippet.

**Additional context**

Naively extending the cmake modules evaluated by the filter logic to include `snippets` is not sufficient:
https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd158e7cec50761fece73b1d4835a2e/scripts/pylib/twister/twisterlib/runner.py#L1873-L1876

While the following command finds the snippet, the overlays are not applied for reasons unknown to me :)
```
cmake -B build -DBOARD=nrf52840dk/nrf52840 -S samples/hello_world/ -DSNIPPET=rtt-console -DMODULES=snippets,dts -P./cmake/package_helper.cmake

-- Board: nrf52840dk, qualifiers: nrf52840
-- Snippet(s): rtt-console
-- Found host-tools: zephyr 0.16.8 (/home/jordan/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/jordan/zephyr-sdk-0.16.8)
-- Found Dtc: /opt/zephyr-sdk/0.9/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/jordan/code/zephyr/zephyr/boards/nordic/nrf52840dk/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: /home/jordan/code/zephyr/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/jordan/code/zephyr/zephyr/build/zephyr/edt.pickle
-- Generated zephyr.dts: /home/jordan/code/zephyr/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/jordan/code/zephyr/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/jordan/code/zephyr/zephyr/build/zephyr/dts.cmake
```

nashif, 2025-01-26:
@JordanYates I am not able to reproduce, adding this:

```patch
diff --git a/samples/hello_world/sample.yaml b/samples/hello_world/sample.yaml
index 1bcb7db62f2..f009004b98f 100644
--- a/samples/hello_world/sample.yaml
+++ b/samples/hello_world/sample.yaml
@@ -4,8 +4,11 @@ sample:
   name: hello world
 common:
   tags: introduction
+  filter: CONFIG_RTT_CONSOLE
   integration_platforms:
     - native_sim
+  required_snippets:
+    - rtt-console
   harness: console
   harness_config:
     type: one_line
```

and then running 
`./scripts/twister -T samples/hello_world/ --platform nrf52840dk/nrf52840 -v` I get:

```
INFO    - 1/1 nrf52840dk/nrf52840       samples/hello_world/sample.basic.helloworld        NOT RUN (build <zephyr>)
```

So, the runtime filter works fine

JordanYates, 2025-01-26:
So it looks like it works on `main`, it does not on the `v4.0.0` tag. Do you know which PR fixed the behavior?

nashif, 2025-01-27:
triggered a backport https://github.com/zephyrproject-rtos/zephyr/pull/84603

not sure what other impact it might have, lets see the CI results.

nashif, 2025-01-27:
re-assigning as this is not a twister bug and was fixed in the build system already.

tejlmand, 2025-02-20:
Fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/84603

closed 2025-02-20 by tejlmand