zephyrproject-rtos/zephyr #81851: Bluetooth: Rename CONFIG_BT_HAS_HCI_VS

Thalley, 2024-11-25:
**Is your enhancement proposal related to a problem? Please describe.**
`CONFIG_BT_HAS_HCI_VS` is used for vendor specific commands, but the prefix clashes with the Bluetooth Hearing Access Service (BT_HAS). 

**Describe the solution you'd like**
Deprecate the Kconfig option and add a new one called `CONFIG_BT_HCI_HAS_VS`

**Describe alternatives you've considered**
Other names are possible as well, but the `CONFIG_BT_HCI` prefix makes more sense to have. 

**Additional context**
https://github.com/zephyrproject-rtos/zephyr/pull/81828 originally discussed here

jhedberg, 2024-11-25:
I can pick up this task once #81828 has been merged.

alwa-nordic, 2024-11-28:
I think it makes more sense to put the `HAS` in front: `HAS_BT_HCI_VS`.

jhedberg, 2024-11-28:
> I think it makes more sense to put the `HAS` in front: `HAS_BT_HCI_VS`.

I wasn't aware of that being an approved reserved Kconfig namespace, however now that I did `git grep 'config HAS_'` there does seem to be plenty of usage of the name space in multiple different subsystems. So yes, that seems like a better option. I wonder if we have other similar symbols in the Bluetooth subsystem that should be updated?