zephyrproject-rtos/zephyr #81893: ESP32S3_DEVKITC LED Strip Sample - RGB Pin # Wrong in Overlay

burke-david, 2024-11-25:
**Describe the bug**

In running the [led_strip sample project](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/led/led_strip) on the [Espressif ESP32S3_DEVKITC platform](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide.html#hardware-reference), I noticed that the defined pin for the RGB LED in the [esp32s3_devkitc_procpu.overlay](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/drivers/led/led_strip/boards/esp32s3_devkitc_procpu.overlay) file should be `GPIO38`, not `GPIO39`.
So instead of 
```
pinmux = <I2S0_O_SD_GPIO39>;
```
Line 42 should read
```
pinmux = <I2S0_O_SD_GPIO38>;
```

**To Reproduce**

**Expected behavior**

**Impact**

The example will not function properly without the pin being changed to `GPIO38` on the esp32s3_devkitc boards.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

Discovered using the latest commit as of yesterday [b1def71](https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd158e7cec50761fece73b1d4835a2e)

**Additional context**

kartben, 2024-11-25:
Thanks @burke-david, do you want to send a pull request to get this fixed?

burke-david, 2024-11-26:
@kartben, yes, thank you. Should have done that in the first place. [PR](https://github.com/zephyrproject-rtos/zephyr/pull/82002) submitted referencing this issue.

closed 2024-12-02 by kartben