zephyrproject-rtos/zephyr #81902: [Coverity CID: 434745] Unchecked return value in drivers/sensor/sensirion/scd4x/scd4x.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/v4.0.0/drivers/sensor/sensirion/scd4x/scd4x.c#L190

Category: Error handling issues
Function: `scd4x_set_idle_mode`
Component: Drivers
CID: [434745](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434745)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/v4.0.0/drivers/sensor/sensirion/scd4x/scd4x.c#L190

```
184     {
185      const struct scd4x_config *cfg = dev->config;
186      int ret;
187    
188      if (cfg->mode == SCD4X_MODE_SINGLE_SHOT) {
189              /*send wake up command twice because of an expected nack return in power down mode*/
>>>     CID 434745:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "scd4x_write_command" without checking return value (as is done elsewhere 21 out of 23 times).
190              scd4x_write_command(dev, SCD4X_CMD_WAKE_UP);
191              ret = scd4x_write_command(dev, SCD4X_CMD_WAKE_UP);
192              if (ret < 0) {
193                      LOG_ERR("Failed write wake_up command.");
194                      return ret;
195              }
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CHECKED_RETURN). ([CWE-252](http://cwe.mitre.org/data/definitions/252.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434745

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

MaureenHelm, 2024-12-02:
@jfaeh can you take a look?

jfaeh, 2024-12-09:
@MaureenHelm  This is intentional and can be ignored. I updated the coverity parameters.

closed 2024-12-20 by teburd