zephyrproject-rtos/zephyr #81908: [Coverity CID: 434739] Unchecked return value in drivers/sensor/sensirion/scd4x/scd4x.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/v4.0.0/drivers/sensor/sensirion/scd4x/scd4x.c#L862

Category: Error handling issues
Function: `scd4x_init`
Component: Drivers
CID: [434739](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434739)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/v4.0.0/drivers/sensor/sensirion/scd4x/scd4x.c#L862

```
856              return -ENODEV;
857      }
858    
859      ret = scd4x_write_command(dev, SCD4X_CMD_STOP_PERIODIC_MEASUREMENT);
860      if (ret < 0) {
861              /*send wake up command twice because of an expected nack return in power down mode*/
>>>     CID 434739:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "scd4x_write_command" without checking return value (as is done elsewhere 21 out of 23 times).
862              scd4x_write_command(dev, SCD4X_CMD_WAKE_UP);
863              ret = scd4x_write_command(dev, SCD4X_CMD_WAKE_UP);
864              if (ret < 0) {
865                      LOG_ERR("Failed to put the device in idle mode.");
866                      return ret;
867              }
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CHECKED_RETURN). ([CWE-252](http://cwe.mitre.org/data/definitions/252.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434739

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

MaureenHelm, 2024-12-02:
@jfaeh this came from #77673, can you take a look?

jfaeh, 2024-12-09:
@MaureenHelm  This is intentional and can be ignored. I updated the coverity parameters.

closed 2025-01-28 by zephyrbot