zephyrproject-rtos/zephyr #81930: [Coverity CID: 434686] Out-of-bounds access in drivers/counter/rtc_mcp7940n.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/counter/rtc_mcp7940n.c

Category: Memory - corruptions
Function: `write_data_block`
Component: Drivers
CID: [434686](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434686)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/counter/rtc_mcp7940n.c#L289

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434686

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-07-05:
Analysis: 
-------------
The function `write_data_block()` is only seen within this compilation unit.
It is called 2 times as follows:
```
rc = write_data_block(dev, REG_RTC_SEC, RTC_TIME_REGISTERS_SIZE);
rc = write_data_block(dev, alarm_base_address, RTC_ALARM_REGISTERS_SIZE);
```
The problem reported by Coverity is a False Positive when alarm_base_address (which is either REG_ALM0_SEC or REG_ALM1_SEC) is passed. Coverity sees that size could be up to 9 bytes because of this check on size:
```
if (size > MAX_WRITE_SIZE) {
   return -EINVAL;
}
```
This could potentially cause a problem for REG_ALM[0|1]_SEC, since the destination register is only 6 bytes. Of course, it's not the case as write_data_block() is called with size==RTC_ALARM_REGISTERS_SIZE (6 bytes) in this case.

ljd42, 2025-07-05:
We can mark this as False Positive in Coverity.
But to be honest, I really like the PR proposed by @Lefucjusz .

@nordic-krch : Any chance to revive this PR?