zephyrproject-rtos/zephyr #81931: [Coverity CID: 434682] Logically dead code in drivers/gpio/gpio_max149x6.h

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/gpio/gpio_max149x6.h

Category: Control flow issues
Function: `max149x6_reg_transceive`
Component: Drivers
CID: [434682](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434682)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/gpio/gpio_max149x6.h#L111

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434682

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_DEADCODE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

jilaypandya, 2024-12-07:
Ping @bogdanovs, could you have a look at this?

bogdanovs, 2024-12-08:
@jilaypandya yes , this is leftover from some testing phase. I will submit PR with fix for is next couple of days.
Is that sufficient or I need to handle this issue more specific case , I am not familiar with the flow.

jilaypandya, 2024-12-18:
> @jilaypandya yes , this is leftover from some testing phase. I will submit PR with fix for is next couple of days. Is that sufficient or I need to handle this issue more specific case , I am not familiar with the flow.

yeah you can just add a PR and link it to this issue, once the PR is merged, this issue will be automatically closed

bogdanovs, 2025-01-06:
@jilaypandya Sorry for long delay on this minor fix here is the PR https://github.com/zephyrproject-rtos/zephyr/pull/83578

PR #83578 (2025-01-09): drivers: gpio: Remove logically dead code for MAX149x6
Remove logically dead code detected from Coverity.

Fixes: #81931
closed 2025-01-09 by kartben