zephyrproject-rtos/zephyr #81948: [Coverity CID: 434632] Logically dead code in drivers/lora/sx127x.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/lora/sx127x.c#L592

Category: Control flow issues
Function: `sx127x_lora_init`
Component: Drivers
CID: [434632](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434632)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/lora/sx127x.c#L592

```
586              LOG_ERR("SPI device not ready");
587              return -ENODEV;
588      }
589    
590      ret = sx12xx_configure_pin(tcxo_power, GPIO_OUTPUT_INACTIVE);
591      if (ret) {
>>>     CID 434632:  Control flow issues  (DEADCODE)
>>>     Execution cannot reach this statement: "return ret;".
592              return ret;
593      }
594    
595      /* Setup Reset gpio and perform soft reset */
596      ret = sx12xx_configure_pin(reset, GPIO_OUTPUT_ACTIVE);
597      if (ret) {
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_DEADCODE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434632

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-06-23:
This is a false positive. 

Coverity evaluates `ret` to 0, because of the way `sx12xx_configure_pin` is defined  (in drivers/lora/sx12xx_common.h)
```
#define sx12xx_configure_pin(_name, _flags)				\
	COND_CODE_1(DT_INST_NODE_HAS_PROP(0, _name##_gpios),		\
		    (__sx12xx_configure_pin(&dev_config._name, _flags)),\
		    (0))
```
This will evaluate to a call to `__sx12xx_configure_pin` only when the semtech,sx1272 is defined in the Devicetree. 
This can be verifiied with the following (build) test: 
```
west twister -s sample.driver.lora.build.sx1272
```

JordanYates, 2025-06-23:
Yes I agree, this driver will disappear in due time anyway with other changes that are happening.

closed 2025-06-24 by kartben