zephyrproject-rtos/zephyr #81960: net: perf: Running on MacOS and NRF52840-DK

JayToltTech, 2024-11-25:
Note: long discussion in Discord prior to filing a bug

**Describe the bug**

My ultimate goal is to use USB-CDC NCM as a networking adapter to communicate with iPadOS via UDP sockets.  I am struggling to get this stack working, so my first step is to get the net/zperf sample to compile and be recognized by MacOS as a network device.

The current state is that I am using a nRF52840-DK as a reference board and while I can compile and run zperf, I have found a few issues that in the sample with stack overflow and once these are fixed, MacOS recognizes the NCM device but does not show it as Active so it cannot be connected to.  Instead, I see:

[image]

and attempts to get the device manually connected (Limit IP address tracking -> Off, TCP/IP Configure IPv4 -> Manually, IP Address 192.0.2.2) have not worked.

See also:

Discussion on CDC-ECM working with MacOS (NetworkConnection Management needed)
https://github.com/zephyrproject-rtos/zephyr/issues/69209

Platforms such as macOS, iOS and iPadOS will not perform DHCP negotiation with autoconf fallback without the interface being "connected".
https://github.com/zephyrproject-rtos/zephyr/discussions/69211 @donusry-privoro

 - What target platform are you using?

MacOS Sequoia 15.1.1, Apple M1
nRF52840-DK

 - What have you tried to diagnose or workaround this issue?

Switch to USB-Next Device Stack
Simplify approach to use off the shelf sample rather than writing code from scratch
Increase log level
Long discussions on Discord with @josuah 
Review all previous discussions I could find of USB ECM/NCM and MacOS/iPadOS
Increase Stack Sizes to mitigate stack overflows in off the shelf sample code
Run under Ozone debugger
Install Arch Linux and use it to get a 'Linux eye view' of networking and usb device diagnostics
Patch usbd_cdc_ncm.c to set errno = ENOTSUP; in usbd_cdc_ncm_ctd(...) per discussion with @josuah 

 - Is this a regression?

No, AFAIK until recently MacOS/iPadOS interoperability was not possible due to ECM limitations in the USB Previous stack.

**To Reproduce**

```
west build -p -b nrf52840dk/nrf52840 samples/net/zperf -DEXTRA_CONF_FILE="overlay-usbd_next.conf" -DDTC_OVERLAY_FILE="usbd_next_ncm.overlay"
west flash
```

Modifications to prj.conf to deal with memfault stack overflows detected when running under Ozone debugger with RTT debug logging enabled.  Found stack overflows on both netmgmt and usbd threads.

```
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_NET_MGMT_EVENT_STACK_SIZE=4096
CONFIG_USBD_THREAD_STACK_SIZE=4096
```

Modifications to pro.conf to add DBG_LOG levels and switch to using RTT logging:

```
CONFIG_DEBUG=y
CONFIG_DEBUG_INFO=y
CONFIG_DEBUG_OPTIMIZATIONS=y

CONFIG_PRINTK=y
CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_LOG_BACKEND_SHOW_COLOR=n
CONFIG_LOG_PRINTK=y

CONFIG_USE_SEGGER_RTT=y
CONFIG_LOG_BACKEND_RTT=y
[... 96 lines cut ...]
[00:00:02.116,790] <dbg> net_if: net_if_ipv6_start_dad: (main): Interface 0x2000173c ll addr 02:00:00:1D:27:06 tentative IPv6 addr 2001:db8::1
[00:00:02.117,675] <dbg> net_if: net_if_tx: (tx_q[0]): Processing (pkt 0x20024568, prio 1) network packet iface 1 (0x2000173c)
[00:00:02.118,408] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x2000173c) src 02:00:00:1D:27:06 dst 33:33:FF:00:00:01 type 0x86dd len 14
[00:00:02.119,232] <dbg> cdc_ncm: cdc_ncm_send: Configuration is not enabled or interface not ready (0 / 1)
[00:00:02.205,657] <inf> cdc_ncm: USB configuration is not enabled
[00:00:02.214,782] <dbg> cdc_ncm: cdc_ncm_send_speed_change: Cannot send speed change (-16)
[00:00:02.226,196] <dbg> cdc_ncm: ncm_send_notification: Cannot send speed change (-16)
[00:00:02.313,385] <dbg> cdc_ncm: usbd_cdc_ncm_enable: Configuration enabled
[00:00:02.337,402] <dbg> cdc_ncm: ncm_send_notification: Speed change sent
[00:00:02.347,473] <dbg> cdc_ncm: usbd_cdc_ncm_cth: 1: 128 28 0 0
[00:00:02.356,567] <dbg> cdc_ncm: usbd_cdc_ncm_cth: GET_NTB_PARAMETERS
[00:00:02.429,443] <dbg> cdc_ncm: usbd_cdc_ncm_request: finished 81 len 16
[00:00:02.447,387] <dbg> cdc_ncm: ncm_send_notification: Connected status sent
[00:00:02.461,425] <dbg> cdc_ncm: usbd_cdc_ncm_request: finished 81 len 8
[00:00:02.707,519] <inf> cdc_ncm: New configuration, interface 1 alternate 1
[00:00:03.107,666] <dbg> net_if: rs_timeout: (sysworkq): RS no respond iface 0x2000173c count 2
[00:00:03.146,514] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x2000173c) src 02:00:00:1D:27:06 dst 33:33:00:00:00:02 type 0x86dd len 14
[00:00:04.131,713] <dbg> net_if: rs_timeout: (sysworkq): RS no respond iface 0x2000173c count 3
```

**Environment (please complete the following information):**

 - OS: MacOS Sequoia 15.1.1, Apple M1
 - Toolchain: Zephyr SDK,Zephyr OS build v4.0.0-827-g305ceba9204d
 - Commit SHA: 305ceba9204d9a7ee4bae0a6103ef4cbd9a1c54e

JayToltTech, 2024-11-25:
Additional logs moved over from Discord discussion, taken prior to diagnosing and fixing stack overflow issues:

Upon initial connection to the Arch VM:
```
00:00:18.694,244] <inf> cdc_ncm: New configuration, interface 1 alternate 0
[00:00:18.694,274] <dbg> cdc_ncm: usbd_cdc_ncm_update: Skip iface 1 alternate 0
[00:00:18.694,305] <dbg> cdc_ncm: usbd_cdc_ncm_request: finished 1 len 0
[00:00:18.696,075] <dbg> cdc_ncm: usbd_cdc_ncm_cth: 1: 128 28 0 0
[00:00:18.696,105] <dbg> cdc_ncm: usbd_cdc_ncm_cth: GET_NTB_PARAMETERS
[00:00:18.719,940] <inf> cdc_ncm: New configuration, interface 1 alternate 1
[00:00:18.719,970] <dbg> cdc_ncm: cdc_ncm_out_start: enqueue out 2048
[00:00:18.731,048] <dbg> cdc_ncm: usbd_cdc_ncm_ctd: bRequest 0x43 (SetPacketFilter) not implemented
```

Arch dmesg looks healthy:
```
[19313.087220] usb 1-2.2: new full-speed USB device number 49 using xhci_hcd
[19313.175077] usb 1-2.2: New USB device found, idVendor=2fe3, idProduct=0001, bcdDevice= 4.00
[19313.175094] usb 1-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19313.175098] usb 1-2.2: Product: USBD sample
[19313.175104] usb 1-2.2: Manufacturer: Zephyr Project
[19313.175107] usb 1-2.2: SerialNumber: E48335DD75B0C1CF
[19313.185928] cdc_ether 1-2.2:1.0 eth0: register 'cdc_ether' at usb-0000:00:03.0-2.2, CDC Ethernet Device, 00:00:5e:00:53:01
[19313.202475] cdc_ether 1-2.2:1.0 enp0s3u2u2: renamed from eth0
[20193.551181] usb 1-2: USB disconnect, device number 48
[20193.551187] usb 1-2.2: USB disconnect, device number 49
[20193.551259] cdc_ether 1-2.2:1.0 enp0s3u2u2: unregister 'cdc_ether' usb-0000:00:03.0-2.2, CDC Ethernet Device
```

After calling:
```
sudo ip link set enp0s3u2u2 up
```

```
[00:01:13.236,663] <dbg> cdc_ncm: usbd_cdc_ncm_ctd: bRequest 0x43 (SetPacketFilter) not implemented
```

Discussion on SetPacketFilter bug in Linux: https://github.com/hathach/tinyusb/issues/2010

Discussion on ENOTSUP: https://github.com/zephyrproject-rtos/zephyr/commit/a5493a5c71dc0a09d4c7523f70c9ba0d0721b04d

JayToltTech, 2024-11-25:
Modifications to usbd_cdc_ncm.c to add ENOTSUP:

```
static int usbd_cdc_ncm_ctd(struct usbd_class_data *const c_data,
			    const struct usb_setup_packet *const setup,
			    const struct net_buf *const buf)
{
	if (setup->RequestType.recipient == USB_REQTYPE_RECIPIENT_INTERFACE) {
		if (setup->bRequest == SET_ETHERNET_PACKET_FILTER) {
			LOG_DBG("bRequest 0x%02x (%s) not implemented",
				setup->bRequest, "SetPacketFilter");
				errno = ENOTSUP;
			return 0;
		}

		if (setup->bRequest == SET_NTB_INPUT_SIZE) {
			LOG_DBG("bRequest 0x%02x (%s) not implemented",
				setup->bRequest, "SetNtbInputSize");
				errno = ENOTSUP;
			return 0;
		}

		if (setup->bRequest == SET_NTB_FORMAT) {
			LOG_DBG("bRequest 0x%02x (%s) not implemented",
				setup->bRequest, "SetNtbFormat");
				errno = ENOTSUP;
			return 0;
		}
	}

	LOG_DBG("bmRequestType 0x%02x bRequest 0x%02x unsupported",
		setup->bmRequestType, setup->bRequest);
	errno = -ENOTSUP;

	return 0;
}
```

JayToltTech, 2024-11-25:
Modified to use DHCPv4 by setting:

```
CONFIG_NET_DHCPV4=y
CONFIG_NET_CONFIG_SETTINGS=n

CONFIG_UDC_NRF_THREAD_STACK_SIZE=4096
```

Log when connected to MacOS:

```
00> [00:00:00.254,943] <dbg> net_tcp: net_tcp_init: (main): Workq started. Thread ID: 0x20003b88
00> [00:00:00.255,615] <dbg> net_if: net_if_init: (main): 
00> [00:00:00.255,920] <err> net_if: There is no network interface to work with!
00> [00:00:00.256,561] <dbg> net_if: net_if_post_init: (main): 
00> *** Booting Zephyr OS build v4.0.0-827-g305ceba9204d ***
00> [00:00:00.257,293] <wrn> net_config: No auto-started network interface - network-bound app initialization skipped.
00> [00:00:02.170,837] <inf> net_echo_server_sample: Run echo server
00> [00:00:02.171,234] <inf> net_echo_server_sample: Waiting network to be connected
00> [00:00:02.171,966] <wrn> net_config: No auto-started network interface - network-bound app initialization skipped.
00> [00:00:00.447,601] <dbg> cdc_ncm: usbd_cdc_ncm_preinit: CDC NCM device initialized
00> [00:00:00.448,974] <dbg> net_tcp: net_tcp_init: (main): Workq started. Thread ID: 0x20002c00
00> [00:00:00.449,920] <dbg> net_if: net_if_init: (main): 
00> [00:00:00.450,439] <dbg> net_if: init_iface: (main): On iface 0x20001720
00> [00:00:00.450,897] <dbg> net_ethernet: ethernet_init: (main): Initializing Ethernet L2 0x20003c10 for iface 1 (0x20001720)
00> [00:00:00.451,629] <dbg> net_if: update_operational_state: (main): iface 1 (0x20001720), oper state DOWN admin DOWN carrier OFF dormant OFF
00> [00:00:00.452,331] <dbg> cdc_ncm: cdc_ncm_iface_init: CDC NCM interface initialized
00> [00:00:00.452,850] <dbg> net_if: net_if_ipv6_calc_reachable_time: (main): min_reachable:15000 max_reachable:45000
00> [00:00:00.453,674] <dbg> net_if: net_if_post_init: (main): 
00> [00:00:00.454,071] <dbg> net_if: net_if_up: (main): iface 1 (0x20001720)
00> [00:00:00.454,498] <dbg> cdc_ncm: cdc_ncm_iface_start: Start interface 1
00> [00:00:00.454,956] <dbg> net_if: update_o[00:00:04.510,742] <inf> cdc_ncm: USB configuration is not enabled
00> [00:00:04.511,169] <dbg> cdc_ncm: cdc_ncm_send_speed_change: Cannot send speed change (-16)
00> [00:00:04.511,657] <dbg> cdc_ncm: ncm_send_notification: Cannot send speed change (-16)
00> [00:00:04.512,207] <dbg> net_if: dad_timeout: (sysworkq): DAD succeeded for fe80::ff:fe85:f49c at interface 1
00> [00:00:04.512,847] <dbg> net_if: rs_timeout: (sysworkq): RS no respond iface 0x20001720 count 1
00> [00:00:04.513,397] <dbg> net_if: net_if_start_rs: (sysworkq): Starting ND/RS for iface 0x20001720
00> [00:00:04.514,129] <dbg> net_if: net_if_tx: (tx_q[0]): Processing (pkt 0x200245a8, prio 1) network packet iface 1 (0x20001720)
00> [00:00:04.514,923] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x20001720) src 02:00:00:85:F4:9C dst 33:33:00:00:00:02 type 0x86dd len 14
00> [00:00:04.515,747] <dbg> cdc_ncm: cdc_ncm_send: Configuration is not enabled or interface not ready (0 / 1)
00> [00:00:04.516,448] <dbg> cdc_ncm: usbd_cdc_ncm_init: CDC NCM class initialized
00> [00:00:04.612,213] <inf> cdc_ncm: USB configuration is not enabled
00> [00:00:04.621,337] <dbg> cdc_ncm: cdc_ncm_send_speed_change: Cannot send speed change (-16)
00> [00:00:04.632,720] <dbg> cdc_ncm: ncm_send_notification: Cannot send speed change (-16)
00> [00:00:04.718,292] <dbg> cdc_ncm: usbd_cdc_ncm_enable: Configuration enabled
00> [00:00:04.743,865] <dbg> cdc_ncm: ncm_send_notification: Speed change sent
00> [00:00:04.756,561] <dbg> cdc_ncm: usbd_cdc_ncm_cth: 1: 128 28 0 0
00> [00:00:04.765,625] <dbg> cdc_ncm: usbd_cdc_ncm_cth: GET_NTB_PARAMETERS
00> [00:00:04.838,500] <dbg> cdc_ncm: usbd_cdc_ncm_request: finished 81 len 16
00> [00:00:04.853,851] <dbg> cdc_ncm: ncm_send_notification: Connected status sent
00> [00:00:04.870,483] <dbg> cdc_ncm: usbd_cdc_ncm_request: finished 81 len 8
00> [00:00:05.514,190] <dbg> net_if: rs_timeout: (sysworkq): RS no respond iface 0x20001720 count 2
00> [00:00:05.526,000] <dbg> net_if: net_if_start_rs: (sysworkq): Starting ND/RS for iface 0x20001720
00> [00:00:05.538,299] <dbg> net_if: net_if_tx: (tx_q[0]): Processing (pkt 0x200245a8, prio 1) network packet iface 1 (0x20001720)
00> [00:00:05.553,039] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x20001720) src 02:00:00:85:F4:9C dst 33:33:00:00:00:02 type 0x86dd len 14
00> [00:00:05.737,609] <inf> cdc_ncm: New configuration, interface 1 alternate 1
00> [00:00:05.747,711] <dbg> cdc_ncm: cdc_ncm_out_start: enqueue out 2048
00> [00:00:28.930,633] <err> os: ***** MPU FAULT *****
00> [00:00:28.938,293] <err> os:   Stacking error (context area might be not valid)
00> [00:00:28.948,577] <err> os:   Data Access Violation
00> [00:00:28.956,420] <err> os:   MMFAR Address: 0x2000baf0
00> [00:00:28.964,660] <err> os: r0/a1:  0x11b801c9  r1/a2:  0xb87c5eac  r2/a3:  0x920b96cf
00> [00:00:28.975,738] <err> os: r3/a4:  0xf895349a r12/ip:  0xf0af691d r14/lr:  0x8558d2bc
00> [00:00:28.986,816] <err> os:  xpsr:  0x8af3d600
00> [00:00:28.994,201] <err> os: Faulting instruction address (r15/pc): 0xa3069130
00> [00:00:29.004,455] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
00> [00:00:29.014,648] <err> os: Current thread: 0x200031c0 (udc_nrfx)
00> [00:00:29.023,773] <err> os: Halting system
00> [00:00:00.354,034] <dbg> cdc_ncm: usbd_cdc_ncm_preinit: CDC NCM device initialized
[... 88 lines cut ...]
00> [00:02:36.245,727] <dbg> cdc_ncm: usbd_cdc_ncm_ctd: bRequest 0x43 (SetPacketFilter) not implemented
```

`ip link`

[... cut ...]
[ 1469.217365] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1469.217366] usb 1-2.1: Product: USBD sample
[ 1469.217367] usb 1-2.1: Manufacturer: Zephyr Project
[ 1469.217368] usb 1-2.1: SerialNumber: F9374730D1CE8A63
[ 1469.344002] cdc_ncm 1-2.1:1.0: MAC-Address: 00:00:5e:00:53:01
[ 1469.344276] cdc_ncm 1-2.1:1.0 eth0: register 'cdc_ncm' at usb-0000:00:03.0-2.1, CDC NCM (NO ZLP), 00:00:5e:00:53:01
[ 1469.347876] cdc_ncm 1-2.1:1.0 enp0s3u2u1: renamed from eth0
```

still no joy.

JayToltTech, 2024-11-26:
Have tried two variations to see if I can narrow down the bug: samples/net/sockets/echo_server and using an Nucleo-H533RE or nRF53-DK.

samples/net/zperf is too large to fit in the SRAM of the H533RE (RAM/SRAM bank is 128k) without modification to use other banks (SRAM2/SRAM3).

samples/net/sockets/echo_server does not start the network adapter by default in my testing when using NCM/USBNext, so this sample will need modifications to be able to 'work out of the box' as a simple repo test harness.

JayToltTech, 2024-11-26:
I tried moving to CDC-ECM from CDC-NCM to see if that would clear things up, e.g.

```
west build -p -b nrf52840dk/nrf52840 samples/net/zperf -DEXTRA_CONF_FILE="overlay-usbd_next.conf" -DDTC_OVERLAY_FILE="usbd_next_ecm.overlay"
```

This at least got me to see outgoing DHCP query packets from the device on Wireshark running on MacOS.

I then tried to convert it over to running `CONFIG_NET_DHCPV4_SERVER=y` and pasting in `enable_dhcpv4_server()` following the code from 'samples/net/wifi/apsta_mode'.  This didn't seem to work at all, after switching MacOS back to DHCP on the network interface, Wireshark did not pick up any traffic.  I'm seeing errors on both the zephyr logs and when I try to connect from Arch Linux I see errors in dmesg as well:

```
00:04:13.744,110] <inf> cdc_ecm: bRequest 0x43 (SetPacketFilter) not implemented
[00:00:02.355,285] <dbg> net_if: update_operational_state: (usbd): iface 1 (0x20001698), oper state UP admin UP carrier ON dormant OFF
[00:00:02.370,697] <dbg> net_if: net_if_ipv6_maddr_add: (usbd): [0] interface 1 (0x20001698) address ff02::1 added
[00:00:02.384,429] <dbg> net_if: net_if_start_dad: (usbd): Starting DAD for iface 0x20001698
[00:00:02.411,437] <dbg> net_if: net_if_ipv6_maddr_add: (usbd): [1] interface 1 (0x20001698) address ff02::1:ff23:d3d8 added
[00:00:02.426,147] <dbg> net_if: join_mcast_solicit_node: (usbd): Join solicit node address ff02::1:ff23:d3d8 (ifindex 1)
[00:00:02.457,244] <dbg> net_if: net_if_start_rs: (usbd): Starting ND/RS for iface 0x20001698
[00:00:02.469,207] <dbg> net_if: net_if_tx: (tx_q[0]): Processing (pkt 0x20024340, prio 1) network packet iface 1 (0x20001698)
[00:00:02.483,917] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x20001698) src 02:00:00:23:D3:D8 dst 33:33:00:00:00:16 type 0x86dd len 14
[00:00:02.501,434] <inf> cdc_ecm: bRequest 0x43 (SetPacketFilter) not implemented
[00:00:02.557,342] <dbg> net_if: dad_timeout: (sysworkq): DAD succeeded for fe80::ff:fe23:d3d8 at interface 1
[00:00:03.271,179] <dbg> net_if: net_if_tx: (tx_q[0]): Processing (pkt 0x20024384, prio 1) network packet iface 1 (0x20001698)
[00:00:03.285,919] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x20001698) src 02:00:00:23:D3:D8 dst 33:33:00:00:00:16 type 0x86dd len 14
[00:00:03.303,649] <dbg> net_if: net_if_tx: (tx_q[0]): Processing (pkt 0x200243c8, prio 1) network packet iface 1 (0x20001698)
[00:00:03.318,389] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x20001698) src 02:00:00:23:D3:D8 dst 33:33:FF:23:D3:D8 type 0x86dd len 14
[00:00:03.336,120] <dbg> net_if: net_if_tx: (tx_q[0]): Processing (pkt 0x2002440c, prio 1) network packet iface 1 (0x20001698)
[00:00:03.350,830] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x20001698) src 02:00:00:23:D3:D8 dst 33:33:00:00:00:02 type 0x86dd len 14
80,895] <dbg> net_if: net_if_start_rs: (sysworkq): Starting ND/RS for iface 0x20001698
[00:00:03.493,133] <dbg> net_if: net_if_tx: (tx_q[0]): Processing (pkt 0x2002440c, prio 1) network packet iface 1 (0x20001698)
[00:00:03.507,873] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x20001698) src 02:00:00:23:D3:D8 dst 33:33:00:00:00:02 type 0x86dd len 14
[00:00:04.493,103] <dbg> net_if: rs_timeout: (sysworkq): RS no respond iface 0x20001698 count 2
[00:00:04.504,913] <dbg> net_if: net_if_start_rs: (sysworkq): Starting ND/RS for iface 0x20001698
[00:00:04.517,150] <dbg> net_if: net_if_tx: (tx_q[0]): Processing (pkt 0x2002440c, prio 1) network packet iface 1 (0x20001698)
[00:00:04.531,860] <dbg> net_ethernet: ethernet_fill_header: (tx_q[0]): iface 1 (0x20001698) src 02:00:00:23:D3:D8 dst 33:33:00:00:00:02 type 0x86dd len 14
[00:00:05.517,120] <dbg> net_if: rs_timeout: (sysworkq): RS no respond iface 0x20001698 count 3
```

```
24412.552363] cdc_ether 1-2.1:1.0 eth0: register 'cdc_ether' at usb-0000:00:03.0-2.1, CDC Ethernet Device, 00:00:5e:00:53:01
[24412.555723] cdc_ether 1-2.1:1.0 enp0s3u2u1: renamed from eth0
[24419.748935] cdc_ether 1-2.1:1.0 enp0s3u2u1: NETDEV WATCHDOG: CPU: 1: transmit queue 0 timed out 5508 ms
[24424.930043] cdc_ether 1-2.1:1.0 enp0s3u2u1: NETDEV WATCHDOG: CPU: 0: transmit queue 0 timed out 5180 ms
[24430.048339] cdc_ether 1-2.1:1.0 enp0s3u2u1: NETDEV WATCHDOG: CPU: 2: transmit queue 0 timed out 5118 ms
[24436.193294] cdc_ether 1-2.1:1.0 enp0s3u2u1: NETDEV WATCHDOG: CPU: 1: transmit queue 0 timed out 6143 ms
[24441.826239] cdc_ether 1-2.1:1.0 enp0s3u2u1: NETDEV WATCHDOG: CPU: 2: transmit queue 0 timed out 5631 ms
[24476.707089] cdc_ether 1-2.1:1.0 enp0s3u2u1: NETDEV WATCHDOG: CPU: 3: transmit queue 0 timed out 5326 ms
[24609.761827] cdc_ether 1-2.1:1.0 enp0s3u2u1: NETDEV WATCHDOG: CPU: 1: transmit queue 0 timed out 5474 ms
[24801.782313] cdc_ether 1-2.1:1.0 enp0s3u2u1: NETDEV WATCHDOG: CPU: 0: transmit queue 0 timed out 5003 ms
```

JayToltTech, 2024-11-26:
DHCPV4_SERVER is failing to initialize, can't quite figure out what step(s) are needed to enable it to work:

```
00> [00:00:04.069,366] <dbg> cdc_ecm: usbd_cdc_ecm_init: CDC ECM class initialized
00> [00:00:05.099,700] <dbg> net_if: net_if_ipv4_addr_add: (main): [0] interface 1 (0x20001698) address 192.168.4.1 type MANUAL added
00> [00:00:09.308,013] <dbg> net_sock: zsock_socket_internal: (main): socket: ctx=0x20002478, fd=3
00> [00:00:09.308,715] <dbg> net_ctx: net_context_bind: (main): Context 0x20002478 binding to UDP 0.0.0.0:67 iface 1 (0x20001698)
00> [00:00:09.309,448] <dbg> net_conn: net_conn_change_callback: (main): [7] connection handler 0x200050fc changed callback
00> [00:00:09.310,150] <dbg> net_conn: net_conn_change_remote: (main): [7] connection handler 0x200050fc changed remote
00> [00:00:09.311,035] <dbg> net_conn: conn_register_debug: (main): [0x200050fc/17/1/0x15] remote -/0 
00> [00:00:09.311,706] <dbg> net_conn: conn_register_debug: (main):   local 0.0.0.0/67 cb 0x159c1 ud 0
00> [00:00:09.312,286] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: Started DHCPv4 server, address pool:
00> [00:00:09.312,896] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:    0: 192.168.4.11
00> [00:00:09.313,415] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:    1: 192.168.4.12
00> [00:00:09.314,117] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:    2: 192.168.4.13
00> [00:00:09.314,697] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:    3: 192.168.4.14
00> [00:00:14.053,771] <err> net_sock_svc: Socket service thread not running, service 0x3c7e4 register fails.
00> [00:00:19.480,010] <err> net_dhcpv4_server: Failed to register socket service, -5
```

jfischer-no, 2024-11-26:
> and attempts to get the device manually connected (Limit IP address tracking -> Off, TCP/IP Configure IPv4 -> Manually, IP Address 192.0.2.2) have not worked.

Has it been resolved or is manual configuration not possible on this operating system?

> **To Reproduce**
> 
> ```
> west build -p -b nrf52840dk/nrf52840 samples/net/zperf -DEXTRA_CONF_FILE="overlay-usbd_next.conf" -DDTC_OVERLAY_FILE="usbd_next_ncm.overlay"
> west flash
> ```

I am not experiencing memory overflows in the default sample configuration, have you changed the sample configuration?

I suggest we try to solve it one step at a time, have you been able to get zperf sample to run on MacOS? As an alternative, there is also samples/net/sockets/echo_server/ and you can find http_server [here](https://github.com/zephyrproject-rtos/zephyr/pull/80985/commits/e52009147b942d1f4dbc27b186c362675c83443c). No idea why you need a DHCP server, but I would get simpler things working first.

JayToltTech, 2024-11-27:
Here are the overlay files I am using with the zperf sample:
[overlays.zip](https://github.com/user-attachments/files/17936068/overlays.zip)

I am calling west with:

```
west build -p -b nrf52840dk/nrf52840 samples/net/zperf -DEXTRA_CONF_FILE="overlay-usbd_next.conf overlay-debug.conf overlay-log.conf overlay-max-stacks.conf" -DDTC_OVERLAY_FILE=usbd_next_ncm.overlay
```

I've updated the Discord discussion with some investigations around "cdc_ncm: USB configuration is not enabled" as this seems to be a fatal issue.  I'm not successful in parsing the code to determine why `usbd_cdc_ncm_enable(...)` does not seem to get called.

```
00:00:02.299,133] <dbg> udc: udc_debug_ep_enqueue: ep 0x80 queue is empty
[00:00:02.317,657] <dbg> usbd_core: usbd_event_handler: RESUME event
[00:00:02.327,301] <dbg> udc: udc_ep_buf_alloc: Allocate net_buf, ep 0x00, size 8
[00:00:02.357,971] <dbg> udc: udc_ep_buf_alloc: Allocate net_buf, ep 0x00, size 8
[00:00:02.368,682] <inf> udc: Sequence 3 not completed
[00:00:02.560,699] <dbg> cdc_ncm: cdc_ncm_send_speed_change: Cannot send speed change (-16)
[00:00:02.816,101] <inf> cdc_ncm: USB configuration is not enabled
[00:00:02.825,378] <dbg> cdc_ncm: cdc_ncm_send[00:00:03.742,370] <inf> cdc_ncm: USB configuration is not enabled
[00:00:03.874,694] <inf> cdc_ncm: USB configuration is not enabled
[00:00:03.895,629] <dbg> cdc_ncm: ncm_send_notification: Cannot send speed change (-16)
[00:00:04.148,651] <dbg> cdc_ncm: cdc_ncm_send_speed_change: Cannot send speed change (-16)
[00:00:04.160,308] <dbg> [00:00:04.424,987] <dbg> cdc_ncm: ncm_send_notification: Cannot send speed change (-16)
[00:00:04.933,441] <inf> cdc_ncm: USB configuration is not enabled
[00:00:05.330,535] <inf> cdc_ncm: USB configuration is not enabled
```

JayToltTech, 2024-11-27:
Using ECM, I am able to ping from my Arch Linux VM when using v4.0-branch:

```
west build -p -b nrf52840dk/nrf52840 samples/net/zperf -DEXTRA_CONF_FILE="overlay-usbd_next.conf" -DDTC_OVERLAY_FILE=usbd_next_ecm.overlay
```

```
> dmesg -w

[154370.335751] usb 1-2.3: new full-speed USB device number 45 using xhci_hcd
[154370.420913] usb 1-2.3: New USB device found, idVendor=2fe3, idProduct=0001, bcdDevice= 4.00
[154370.420920] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[154370.420921] usb 1-2.3: Product: USBD sample
[154370.420922] usb 1-2.3: Manufacturer: Zephyr Project
[154370.420923] usb 1-2.3: SerialNumber: F9374730D1CE8A63
[154370.428982] cdc_ether 1-2.3:1.0 eth0: register 'cdc_ether' at usb-0000:00:03.0-2.3, CDC Ethernet Device, 00:00:5e:00:53:01
[154370.432437] cdc_ether 1-2.3:1.0 enp0s3u2u3: renamed from eth0

> ip link show enp0s3u2u3
16: enp0s3u2u3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:5e:00:53:01 brd ff:ff:ff:ff:ff:ff

> sudo ip address add 192.0.2.2/24 dev enp0s3u2u3
> sudo ip link set enp0s3u2u3 up 
> ping 192.0.2.1
PING 192.0.2.1 (192.0.2.1) 56(84) bytes of data.
64 bytes from 192.0.2.1: icmp_seq=1 ttl=64 time=0.991 ms
```

Not successful with NCM in same situation:

```
west build -p -b nrf52840dk/nrf52840 samples/net/zperf -DEXTRA_CONF_FILE="overlay-usbd_next.conf" -DDTC_OVERLAY_FILE=usbd_next_ncm.overlay
```

```
> dmesg -w

[154772.130744] usb 1-2: new full-speed USB device number 46 using xhci_hcd
[154772.257443] usb 1-2: New USB device found, idVendor=203a, idProduct=fffe, bcdDevice= 1.01
[154772.257470] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[154772.257472] usb 1-2: Product: Virtual USB1.1 HUB
[154772.257473] usb 1-2: Manufacturer: Parallels
[154772.257473] usb 1-2: SerialNumber: PW3.0
[154772.258902] hub 1-2:1.0: USB hub found
[154772.258930] hub 1-2:1.0: 15 ports detected
[154772.527919] usb 1-2.1: new full-speed USB device number 47 using xhci_hcd
[154772.612009] usb 1-2.1: New USB device found, idVendor=2fe3, idProduct=0001, bcdDevice= 4.00
[154772.612015] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[154772.612016] usb 1-2.1: Product: USBD sample
[154772.612017] usb 1-2.1: Manufacturer: Zephyr Project
[154772.612018] usb 1-2.1: SerialNumber: F9374730D1CE8A63
[154772.644467] cdc_ncm 1-2.1:1.0: MAC-Address: 00:00:5e:00:53:01
[154772.644609] cdc_ncm 1-2.1:1.0 eth0: register 'cdc_ncm' at usb-0000:00:03.0-2.1, CDC NCM (NO ZLP), 00:00:5e:00:53:01
[154772.648277] cdc_ncm 1-2.1:1.0 enp0s3u2u1: renamed from eth0

> ip link show enp0s3u2u1

17: enp0s3u2u1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:5e:00:53:01 brd ff:ff:ff:ff:ff:ff

> sudo ip address add 192.0.2.2/24 dev enp0s3u2u1
> sudo ip link set enp0s3u2u1 up
> ping 192.0.2.1
PING 192.0.2.1 (192.0.2.1) 56(84) bytes of data.
From 192.0.2.2 icmp_seq=1 Destination Host Unreachable
> ip link show enp0s3u2u1
17: enp0s3u2u1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:5e:00:53:01 brd ff:ff:ff:ff:ff:ff
```

jfischer-no, 2024-11-28:
I found a few wrong assumptions in the NCM implementation, still investigating them.

JayToltTech, 2024-11-28:
Will give this a shot @jfischer-no, thanks.

JayToltTech, 2024-11-28:
@jfischer-no No success, behavior the same, unable to get Arch Linux to bring the device from state DOWN to UP.

```
git remote add jfischer-no git@github.com:jfischer-no/zephyr.git
git fetch jfischer-no
git checkout pr-device_next-cdc_ncm-nth

west build -p -b nrf52840dk/nrf52840 samples/net/zperf -DEXTRA_CONF_FILE="overlay-usbd_next.conf" -DDTC_OVERLAY_FILE=usbd_next_ncm.overlay

00:00:00.444,427] <inf> cdc_ncm: USB configuration is not enabled
*** Booting Zephyr OS build v4.0.0-1057-g3899c66a2b56 ***
[00:00:04.586,730] <inf> cdc_ncm: USB configuration is not enabled
[00:00:04.587,005] <inf> net_config: Initializing network
[00:00:04.587,066] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:04.686,798] <inf> cdc_ncm: USB configuration is not enabled
[00:00:04.687,744] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:04.786,865] <inf> cdc_ncm: USB configuration is not enabled
[00:00:05.738,098] <inf> cdc_ncm: New configuration, interface 1 alternate 1
[00:00:05.740,051] <wrn> udc_nrf: aborted IN ep 0x82
[00:00:05.740,112] <inf> cdc_ncm: New configuration, interface 1 alternate 0
[00:00:05.740,142] <err> cdc_ncm: Bulk OUT transfer error or zero length
[00:00:05.764,648] <inf> cdc_ncm: New configuration, interface 1 alternate 1

> ip link show

4: enp0s3u2u3: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:5e:00:53:01 brd ff:ff:ff:ff:ff:ff
    
> sudo ip link set enp0s3u2u3 up; echo $?
0

> ip link show

4: enp0s3u2u3: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:5e:00:53:01 brd ff:ff:ff:ff:ff:ff

> sudo ip address add 192.0.2.2/24 dev enp0s3u2u3; echo $?
0

> ping 192.0.2.1

PING 192.0.2.1 (192.0.2.1) 56(84) bytes of data.
From 192.0.2.2 icmp_seq=1 Destination Host Unreachable

```

jfischer-no, 2024-11-29:
> No success, behavior the same, unable to get Arch Linux to bring the device from state DOWN to UP.
> FYI, I'm testing first on Arch Linux because the diagnostic tools are a bit easier to use there :-)

This should work right out of the box.

```
ip address add dev enp0s3u2u3 192.0.2.2/24
ip link set enp0s3u2u3 up
ping 192.0.2.1
```

jfischer-no, 2024-11-29:
@JayToltTech On macOS Ventura
[image]
Will update PR later.

JayToltTech, 2024-12-01:
Nice work!  Looking forward to it.  @jfischer-no using a real live nRF52840-DK, along with a few other DK board I'm testing with (nRF53 and various STM32H5's).

JayToltTech, 2024-12-02:
@jfischer-no Any chance you could push your work up to your GitHub repo that unblocked MacOS?  Would love to take a look at this with you.  I checked your repo for additional NCM related branches and didn't see anything to play with.

jfischer-no, 2024-12-02:
> @jfischer-no Any chance you could push your work up to your GitHub repo that unblocked MacOS? Would love to take a look at this with you. I checked your repo for additional NCM related branches and didn't see anything to play with.

Updated https://github.com/zephyrproject-rtos/zephyr/pull/82282. If you still have problems, please provide some logs, but with only CONFIG_USBD_CDC_NCM_LOG_LEVEL_DBG enabled and everything else USB related set to *_LOG_LEVEL_ERR.

JayToltTech, 2024-12-02:
@jfischer-no you are the bomb.  Up and working here from nRF52840-DK to MacOS.  Moving on to testing with additional OSes and devices.

Many thanks for your attention and skill!

jeffwelder-ellenbytech, 2024-12-02:
I can use this to get to the point of pinging each other (MacOS to nrf52840dk), but can you actually share internet to the DK?

JayToltTech, 2024-12-03:
@jeffwelder-ellenbytech that would be Internet Connection Sharing, https://support.apple.com/guide/mac-help/share-internet-connection-mac-network-users-mchlp1540/mac.

jfischer-no, 2024-12-03:
> ... Moving on to testing with additional OSes and devices.

The Windows 10 (UsbNcm) driver is not working flawlessly with our implementation, and I am reluctant to have a workaround on our end. The issue is now it handles segment delineation, I observed that the Windows 10 driver does not use ZLP or padding to create a short packet for the case the transfer is multiple of MPS but smaller than the dwNtbOutMaxSize. (Linux kernel driver uses padding for "Generic CDC-NCM devices", macOS driver (Ventura) and our implementation uses ZLP). I have not had time to test Windows 11 yet.

jfischer-no, 2024-12-03:
> An 'out of the box' example of USB CDC ECM/NCM that interoperates with MacOS, either samples/net/zperf or samples/net/sockets/echo_server, preferably including autoconf or dhcp so that it 'just works'.

Updated https://github.com/zephyrproject-rtos/zephyr/pull/82282. CDC ECM should work fine on Linux and macOS (tested Ventura).

JayToltTech, 2024-12-04:
Sounds like this issue with ZLP may have been corrected in Windows 11:

https://learn.microsoft.com/en-us/answers/questions/1531925/difference-in-usb-ncm-class-driver-behavior-for-wi

> As per this Ethernet Control Model specification, the host driver of an NCM device’s communication interface is expected to send a Zero-Length Packet (ZLP) when the Nth packet equals the maximum packet size. This specification is correctly adhered to by the host driver in Windows 11, which sends the ZLP as expected. However, this is not the case in Windows 10. We have confirmed this discrepancy through Lacro and Wireshark logs. Consequently, our compact USB NCM device does not operate correctly with the UsbNcm.sys driver in Windows 10.

JayToltTech, 2024-12-04:
FYI, as of this morning, I have MacOS working well with the NRF52840-DK using your provided PR branch @jfischer-no, including DHCP server so MacOS auto configures to speak with the device.

I can also attach my nRF52850-DK to an iPad and run a little ping utility there to confirm basic connectivity.

I consider the bug to be solved and look forward to it landing in 4.1.  Please close unless you want to hold open to track the Win10/11 discussion that started here, I'm happy to continue to try and make a connection there to help improve Windows compatibility if you want.

Note: I have seen a little instability with NTH16 verification of packets during ad hoc testing.  A reset seems to 'fix' the problem and I have not yet found a reproducible case for how to induce the error.

[image]

PR #82282 (2024-12-18): usb: device_next: fix CDC NCM datagram pointers processing 
The implementation should not assume that NDP is placed immediately
after NTH. According to the specification (revision 1.0), NDP can be
anywhere in the transfer buffer. There is also always a "terminating
zero datagram pointer entry", which we should also use to terminate the
datagram processing loop.

Fixes: #81960
closed 2024-12-18 by kartben