zephyrproject-rtos/zephyr #81963: [Coverity CID: 434591] Unintentional integer overflow in drivers/gpio/gpio_pca_series.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/gpio/gpio_pca_series.c

Category: Integer handling issues
Function: `gpio_pca_series_pin_interrupt_configure_extended`
Component: Drivers
CID: [434591](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434591)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/gpio/gpio_pca_series.c#L1362

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434591

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERFLOW_BEFORE_WIDEN). ([CWE-190](http://cwe.mitre.org/data/definitions/190.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-06-23:
Can be simply solved by casting 0b11  to an uint64_t: 
```
uint64_t edge_cfg_mask = (uint64_t)0b11 << edge_cfg_shift;
```

ljd42, 2025-06-23:
Hi @kartben : first good issue :wink:

ljd42, 2025-06-23:
My bad, I don't know yet the coding guideline by heart 🙈

sudarsan-22, 2025-06-24:
fixes: #92064

commit c9be1972 (2025-06-27): gpio: pca_series: fix shift overflow and use 3ULL instead of 0b11

Replaces binary literal with 3ULL to avoid shift overflow and align
with Zephyr coding style.

Fixes: #81963
Fixes: CID 434591
closed 2025-06-27 by dkalowsk