zephyrproject-rtos/zephyr #81967: [Coverity CID: 434580] Untrusted divisor in drivers/sensor/bosch/bmp180/bmp180.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/sensor/bosch/bmp180/bmp180.c

Category: Insecure data handling
Function: `bmp180_press_channel_get`
Component: Drivers
CID: [434580](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434580)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/sensor/bosch/bmp180/bmp180.c#L350

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434580

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_TAINTED_SCALAR). ([CWE-369](http://cwe.mitre.org/data/definitions/369.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

MaureenHelm, 2024-12-02:
@ruehlchris this came from #77182, can you take a look?

ruehlchris, 2024-12-03:
> @ruehlchris this came from #77182, can you take a look?

Yes, I will fix this.

ruehlchris, 2024-12-03:
Hi @MaureenHelm ,

the check hit on this lines.
partial_data1 = (data->raw_temp - cal->ac6) * cal->ac5 / 0x8000;
partial_data2 = cal->mc * 0x800 / (partial_data1 + cal->md);

where partial_data1 + cal->md  theoretically can be zero.

I will add a check that  partial_data1 + cal->md != 0 and for the unlikely case set partial_data2 = 0

closed 2025-05-25 by zephyrbot