zephyrproject-rtos/zephyr #81972: [Coverity CID: 434568] Logically dead code in drivers/lora/sx127x.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/lora/sx127x.c

Category: Control flow issues
Function: `sx127x_antenna_configure`
Component: Drivers
CID: [434568](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434568)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/lora/sx127x.c

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434568

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_DEADCODE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-06-23:
@JordanYates : same story as in #81948

ljd42, 2025-06-24:
Marked in Coverity. Findings will be now dismissed.

@JordanYates : Ticket can be closed with the appropriate labeling.

closed 2025-06-24 by kartben