zephyrproject-rtos/zephyr #81974: [Coverity CID: 434563] Operands don't affect result in drivers/counter/rtc_mcp7940n.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/counter/rtc_mcp7940n.c

Category: Integer handling issues
Function: `mcp7940n_rtc_set_time`
Component: Drivers
CID: [434563](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434563)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/counter/rtc_mcp7940n.c#L404

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434563

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-569](http://cwe.mitre.org/data/definitions/569.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-06-23:
@cfriedt : 
The issue reported by Coverity is related to the use of `time_t`, which is here an `uint32_t` and therefore will be never greater than UINT32_MAX... It looks  like the code is assuming that time_t could be an `uint64_t`.

While it's not a false-positive, I would silent this issue in coverity. 

However, the bigger question about `time_t`  - from what I can see on Discord - isn't yet resolved? I see that we have either 32- or 64-bits depending on the compilers. I don't have checked the signed, but POSIX for instance allow negative value for date prior to 01.01.1970 ( I reckon that it might not be very relevant in a RTC timer API 😆  )

cfriedt, 2025-06-23:
@ljd42 - time_t is dictated by ISO C.

We prefer 64-bits but it's up to the given C library.

ljd42, 2025-06-23:
I would also be expecting a int64_t TBH. Let me double check what's going on here...

ljd42, 2025-09-27:
Marked as False Positive in Coverity.

closed 2025-09-27 by ljd42