zephyrproject-rtos/zephyr #81978: [Coverity CID: 434555] Untrusted divisor in drivers/sensor/bosch/bme680/bme680.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/sensor/bosch/bme680/bme680.c

Category: Insecure data handling
Function: `bme680_power_up`
Component: Drivers
CID: [434555](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434555)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/sensor/bosch/bme680/bme680.c

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434555

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_TAINTED_SCALAR). ([CWE-369](http://cwe.mitre.org/data/definitions/369.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

MaureenHelm, 2024-12-02:
@JordanYates this looks related to #75886, can you take a look?

jilaypandya, 2024-12-03:
i dont know if i can post a screenshot over here, but the issue starts around line 385

JordanYates, 2025-06-26:
There are so, so many places in Zephyr where RAM corruption will result in hardware exceptions. Unless the plan is to try and guard every single one of those locations (with all the overhead that implies), I don't think there is much value in trying to guard against a huge class of errors in a single location.

ljd42, 2025-06-27:
Yes, of course! It's impossible to catch everything reliability. There are many places where things go wrong, and it's always about mitigation, not covering 100% of the case (which is impossible, even with hardware redundancy). 

Then let's mark it as false positive in Coverity as suggested some days ago. @jilaypandya or I can take care of it.

ljd42, 2025-07-09:
Analysis
------------
The data value used for the computation `data->res_heat_val` and others used in the function `bme680_calc_res_heat` are compensation data read from the sensor during the sensor's power up function. The only way these values can't be trusted is if the device does not refer to a proper BME680 sensor device, or if the application has corrupted the sensor's data e.g. due to a rogue thread. So it's an **application bug**.
 
There's no way to detect the corruption easily without implementing a safety pattern like one's complement. But then, it's another driver... 

Marked as False Positive in Coverity.

ljd42, 2025-07-09:
@JordanYates or @danieldegrasse : ticket can be closed.