zephyrproject-rtos/zephyr #81988: [Coverity CID: 434515] Unintentional integer overflow in drivers/sensor/bosch/bmp180/bmp180.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/sensor/bosch/bmp180/bmp180.c

Category: Integer handling issues
Function: `bmp180_compensate_press`
Component: Drivers
CID: [434515](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434515)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/sensor/bosch/bmp180/bmp180.c#L330

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434515

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERFLOW_BEFORE_WIDEN). ([CWE-190](http://cwe.mitre.org/data/definitions/190.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

MaureenHelm, 2024-12-02:
@ruehlchris can you take a look?

ruehlchris, 2024-12-03:
HI @MaureenHelm 

Issue triggered is this line,
partial_B4 = (uint64_t)(cal->ac4 * (partial_X3 + 32768)) / 0x8000;

theoretically int32_t partial_X3 + 32768 can overflow. To tackle cast  partial_X3 to int64_t.

partial_B4 = (uint64_t)(cal->ac4 * ((int64_t)(partial_X3) + 32768)) / 0x8000;

this should be fix the issue.

closed 2024-12-13 by kartben