zephyrproject-rtos/zephyr #81996: [Coverity CID: 353659] Out-of-bounds access in drivers/counter/counter_nrfx_rtc.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/drivers/counter/counter_nrfx_rtc.c

Category: Memory - corruptions
Function: `counter_1_init`
Component: Drivers
CID: [353659](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=353659)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/drivers/counter/counter_nrfx_rtc.c#L753

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=353659

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

nordic-piks, 2025-01-21:
[image]

aescolar, 2025-01-24:
This seems to be effectively the same as https://github.com/zephyrproject-rtos/zephyr/issues/68637

PR #84582 (2025-01-29): boards native_posix: Detect attempt to configure not existing int
Similar to 66a4fe32cef. Prevent overrunning the irq vector table in posix_irq_priority_set. This is not happening today in tree, but coverity thinks it may. Checking for it to prevent it is not a bad idea anyhow, so let's do it.

Fixes #81996.
Fixes #81914.
closed 2025-01-29 by kartben