zephyrproject-rtos/zephyr #81997: [Coverity CID: 353651] Dereference null return value in subsys/bluetooth/controller/ll_sw/ull_sync_iso.c

zephyrbot, 2024-11-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b1def7145fd/subsys/bluetooth/controller/ll_sw/ull_sync_iso.c

Category: Null pointer dereferences
Function: `ull_sync_iso_setup`
Component: Bluetooth
CID: [353651](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=353651)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b1def7145fd/subsys/bluetooth/controller/ll_sw/ull_sync_iso.c#L573

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=353651

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_NULL_RETURNS). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-07-09:
Not sure if it's the best way to solve this problem...

ljd42, 2025-09-22:
Analysis
------------

Coverity complains that `ull_sync_iso_stream_get()` might return NULL, and that corresponding checks are performed elsewhere:
 
https://github.com/zephyrproject-rtos/zephyr/blob/71f6840a1b1b802437848157968b45cf621d104a/subsys/bluetooth/controller/ll_sw/ull_sync_iso.c#L586 

@cvinayak : If NULL can be returned, how should the error be handled in this case? If that is not possible, how about using an __ASSERT() (or its bluetooth subsys equivalent)? 

With a proper guidance, I can take care of the PR if you wish.

cvinayak, 2025-09-22:
You can add a `LL_ASSERT` as the suggestion below, Thank you:

```c
        /* Skip to first selected BIS subevent */                                                   
        stream = ull_sync_iso_stream_get(lll->stream_handle[0]);
        LL_ASSERT(stream);

        if (lll->bis_spacing >= (lll->sub_interval * lll->nse)) {                                   
```

commit 21da2d05 (2025-10-01): Bluetooth: Controller: Fix missing null pointer check

Added a check for null pointer in ull_sync_iso.c file
to avoid potential crashes.

Fixes: #81997