zephyrproject-rtos/zephyr #82005: "west flash" is reporting error

aaronyegx, 2024-11-26:
**Describe the bug**

"west flash" is broken in the Zephyr latest main branch.

**To Reproduce**

1. west build apollo3_evb .\samples\hello_world -p
2. west flash

**Expected behavior**

No error reports, the binary can be programmed to the target board.

**Impact**

Unable to continue the development without "west flash" working.

**Logs and console output**

Traceback (most recent call last):
  File "C:\Python310\Lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python310\Lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Python310\Scripts\west.exe\__main__.py", line 7, in <module>
  File "C:\Python310\Lib\site-packages\west\app\main.py", line 866, in main
    app.run(argv or sys.argv[1:])
  File "C:\Python310\Lib\site-packages\west\app\main.py", line 111, in run
    self.run_command(argv)
  File "C:\Python310\Lib\site-packages\west\app\main.py", line 385, in run_command
    self.run_extension(args.command, argv)
  File "C:\Python310\Lib\site-packages\west\app\main.py", line 467, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "C:\Python310\Lib\site-packages\west\commands.py", line 194, in run
    self.do_run(args, unknown)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts\west_commands\flash.py", line 33, in do_run
    do_run_common(self, my_args, runner_args, domain_file=domains_file)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\run_common.py", line 314, in do_run_common
    prev_runner = do_run_common_image(command, user_args, user_runner_args, used_cmds,
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\run_common.py", line 489, in do_run_common_image
    runner.run(command_name)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\core.py", line 720, in run
    self.do_run(command, **kwargs)
    do_run_common(self, my_args, runner_args, domain_file=domains_file)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\run_common.py", line 314, in do_run_common
    prev_runner = do_run_common_image(command, user_args, user_runner_args, used_cmds,
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\run_common.py", line 489, in do_run_common_image
    runner.run(command_name)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\core.py", line 720, in run
    self.do_run(command, **kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\run_common.py", line 314, in do_run_common
    prev_runner = do_run_common_image(command, user_args, user_runner_args, used_cmds,
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\run_common.py", line 489, in do_run_common_image
    runner.run(command_name)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\core.py", line 720, in run
    self.do_run(command, **kwargs)
    prev_runner = do_run_common_image(command, user_args, user_runner_args, used_cmds,
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\run_common.py", line 489, in do_run_common_image
    runner.run(command_name)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\core.py", line 720, in run
    self.do_run(command, **kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\run_common.py", line 489, in do_run_common_image
    runner.run(command_name)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\core.py", line 720, in run
    self.do_run(command, **kwargs)
    runner.run(command_name)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\core.py", line 720, in run
    self.do_run(command, **kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\core.py", line 720, in run
    self.do_run(command, **kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\jlink.py", line 281, in do_run
    self.flash(**kwargs)
    self.do_run(command, **kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\jlink.py", line 281, in do_run
    self.flash(**kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\jlink.py", line 281, in do_run
    self.flash(**kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\jlink.py", line 451, in flash
    self.run_flash_cmd(fname, flash_file, **kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\jlink.py", line 438, in run_flash_cmd
    self.run_flash_cmd(fname, flash_file, **kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\jlink.py", line 438, in run_flash_cmd
    self.check_call(cmd, **kwargs)
  File "C:\ambiqcodes\radio_sw\zephyr\scripts/west_commands\runners\jlink.py", line 438, in run_flash_cmd
    self.check_call(cmd, **kwargs)
    self.check_call(cmd, **kwargs)
UnboundLocalError: local variable 'cmd' referenced before assignment

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain Zephyr SDK 0.17.0
 - Commit SHA or Version used: f90ce01d4ac507c99e3cddcaba6fd13239b06034

**Additional context**

aaronyegx, 2024-11-26:
It is likely the commit causes the problem:

Commit: dd5874b9242b943a43650f6aae4b0d5af0300da8
Parents: bb3bab7a3442115f0fc17c400696bf88da84329b
Author: Jean-Christophe Pince <[jean.christophe.pince@qorvo.com](mailto:jean.christophe.pince@qorvo.com)>
Author Date: Mon Sep 30 2024 17:32:16 GMT+0800 (China Standard Time)
Committer: Henrik Brix Andersen <[henrik@brixandersen.dk](mailto:henrik@brixandersen.dk)>
Committer Date: Tue Nov 26 2024 04:53:43 GMT+0800 (China Standard Time)
 

jlink: custom flashing support

Allows to use a custom flashing script with jlink runners.

I think this is the indent issue in the jlink.py file.
[image]

aaronyegx, 2024-11-26:
Hi @kartben @henrikbrixandersen , can you help assign the right person to fix this? Thanks.

pdgendt, 2024-11-26:
@aaronyegx please test with #82013

aaronyegx, 2024-11-26:
> @aaronyegx please test with #82013

This fix works.

PR #82013 (2024-11-26): scripts: west_commands: runners: Fix jlink unbound cmd
Commit e2a32d8b0ca596afeae48739c43817f693720a88 introduced an unbound `cmd` issue for `jlink` flashing.

Fixed by moving `cmd` assignment outside of condition.

Fixes #82005
closed 2024-11-26 by kartben