zephyrproject-rtos/zephyr #82039: MCU manager with STM32H750B_dk

MaazSk, 2024-11-26:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/81421

<div type='discussions-op-text'>

<sup>Originally posted by **MaazSk** November 15, 2024</sup>
I am using STM32H750B_DK which has an internal flash of 128K, I am trying to use `smp_svr` example located at `subsys/mgmt/mcumgr/smpsvr` My overlay file is as below

```

/ {
	chosen {
		zephyr,code-partition = &storage_partition;
	};
};

&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		slot0_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(109)>;
		};
		storage_partition: partition@1B400 {
				label = "image-0";
				reg = <0x0001B400 DT_SIZE_M(22)>;
		};
	};
};

&quadspi {
	pinctrl-names = "default";
	pinctrl-0 = <&quadspi_clk_pf10 &quadspi_bk1_ncs_pg6
		     &quadspi_bk1_io0_pd11 &quadspi_bk1_io1_pf9
		     &quadspi_bk1_io2_pf7 &quadspi_bk1_io3_pf6
		     &quadspi_bk2_io0_ph2 &quadspi_bk2_io1_ph3
		     &quadspi_bk2_io2_pg9 &quadspi_bk2_io3_pg14>;
	dual-flash;
	status = "okay";

	mt25ql512ab1: qspi-nor-flash-1@90000000 {
		compatible = "st,stm32-qspi-nor";
		reg = <0x90000000 DT_SIZE_M(64)>; /* 512 Mbits */
		qspi-max-frequency = <72000000>;
		spi-bus-width = <4>;
		reset-cmd;
		reset-cmd-wait = <20>;
		status = "okay";

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			slot1_partition: partition@0 {
				label = "image-0";
				reg = <0x00000000 0x1000000>;
			};

			//slot3_partition: partition@1000000{
			//	label = "image-1";
			//	reg = <0x1000000 DT_SIZE_M(64)>;
			//};
		};
	};
};
```

I am able to build the code but I am unable to sign it because of the memory that I have. Also, I am unable the slots to external memory.

The error with the above overlay I get is as below 
[image].

As it exceeded the memory location and If I change storage partition location to external flash I am unable to flash the code the error that I get is as follows Also the updated overlay is as
```
&quadspi {
	pinctrl-names = "default";
	pinctrl-0 = <&quadspi_clk_pf10 &quadspi_bk1_ncs_pg6
		     &quadspi_bk1_io0_pd11 &quadspi_bk1_io1_pf9
		     &quadspi_bk1_io2_pf7 &quadspi_bk1_io3_pf6
		     &quadspi_bk2_io0_ph2 &quadspi_bk2_io1_ph3
		     &quadspi_bk2_io2_pg9 &quadspi_bk2_io3_pg14>;
	dual-flash;
	status = "okay";

	mt25ql512ab1: qspi-nor-flash-1@90000000 {
		compatible = "st,stm32-qspi-nor";
		reg = <0x90000000 DT_SIZE_M(64)>; /* 512 Mbits */
		qspi-max-frequency = <72000000>;
		spi-bus-width = <4>;
		reset-cmd;
		reset-cmd-wait = <20>;
		status = "okay";

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			slot1_partition: partition@0 {
				label = "image-0";
				reg = <0x00000000 DT_SIZE_M(5)>;
			};

			storage_partition: partition@500000{
				label = "image-1";
				reg = <0x500000 DT_SIZE_M(5)>;
			};
		};
	};
};
```
[image]

What should  be a way forward. Also, if I remove `slot0` from internal flash to external flash I am unable to sign this is the error that I get
[image]

----**UPDATE 1**----
When I made the following changes I am able to sign the file even if I change the partition

```
/ {
	chosen {
		zephyr,code-partition = &slot0_partition;
		zephyr,flash-controller = &mt25ql512ab1;
		zephyr,flash = &mt25ql512ab1;
	};
};
```

----**UPDATE 2**----

I am able to update the code to the device but now I am having difficulty using AuTerm or zephyr mcumgr cli. Is there a tutorial or something how to use the mcumgr cli?

When ever I am using Auterm it is giving me a timeout error

[image]

</div>

nordicjm, 2024-11-26:
Your partitioning makes no sense, your device has 128KiB of flash, which is a single sector, meaning you can't have 109KiB allocated to it, it must be 128KiB for a single partition. Then seemingly you have a 22MiB area? Not sure what's going on there.

MaazSk, 2024-11-26:
Ahh! sorry sharing the updated dts it was in the discussion but forgot to mention it here.
```
/ {
	chosen {
		zephyr,entropy = &rng;
		zephyr,flash-controller = &mt25ql512ab1;
		zephyr,console = &usart3;
		zephyr,shell-uart = &usart3;
		zephyr,sram = &sram0;
		zephyr,flash = &mt25ql512ab1;
		zephyr,display = &ltdc;
		zephyr,code-partition = &slot1_partition;
		zephyr,uart-mcumgr = &usart1;
	};
};

&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(128)>;
		};
	};
};

&quadspi {
	pinctrl-names = "default";
	pinctrl-0 = <&quadspi_clk_pf10 &quadspi_bk1_ncs_pg6
		     &quadspi_bk1_io0_pd11 &quadspi_bk1_io1_pf9
		     &quadspi_bk1_io2_pf7 &quadspi_bk1_io3_pf6
		     &quadspi_bk2_io0_ph2 &quadspi_bk2_io1_ph3
		     &quadspi_bk2_io2_pg9 &quadspi_bk2_io3_pg14>;
	dual-flash;
	status = "okay";

	mt25ql512ab1: qspi-nor-flash-1@90000000 {
		compatible = "st,stm32-qspi-nor";
		reg = <0x90000000 DT_SIZE_M(64)>; /* 512 Mbits */
		qspi-max-frequency = <72000000>;
		spi-bus-width = <4>;
		reset-cmd;
		reset-cmd-wait = <20>;
		status = "okay";

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			slot0_partition: partition@0 {
				label = "image-0";
				reg = <0x00000000 DT_SIZE_M(5)>;
			};

			slot1_partition: partition@500000{
				label = "image-1";
				reg = <0x500000 DT_SIZE_M(5)>;
			};
			storage_partition: partition@A00000{
				label = "storage-1";
				reg = <0xA00000 DT_SIZE_M(5)>;
			};
		};
	};

	mt25ql512ab2: qspi-nor-flash-2@90000000 {
		compatible = "st,stm32-qspi-nor";
		reg = <0x0 DT_SIZE_M(64)>; /* 512 Mbits */
		qspi-max-frequency = <72000000>;
		reset-cmd;
		reset-cmd-wait = <20>;
		status = "okay";
	};
};

&usart1 {
			compatible = "st,stm32-usart", "st,stm32-uart";
			reg = < 0x40011000 0x400 >;
			clocks = < &rcc 0xf0 0x10 >;
			resets = < &rctl 0x1304 >;
			interrupts = < 0x25 0x0 >;
			status = "okay";
			pinctrl-0 = < &usart1_tx_pb6 &usart1_rx_pb7 >;
			pinctrl-names = "default";
			current-speed = < 0x1c200 >;
		};
```

nordicjm, 2024-11-26:
So you are using directXIP mode, I don't know how that works with the stm32 driver since you would need to enable the flash driver to do e.g. flash writing and reading perhaps @erwango can say if this is supported or has been tested with MCUboot.
MTU of 1024 is high, leave it at 256, and should attach a debugger when the application is running to see if it is actually running the application or not (and if so, where in the code it is)

MaazSk, 2024-11-26:
The code is working the error which I get is 
> [00:04:21.453,000] mcumgr_img_grp: Image upload inspect failed: 22
> [00:04:21.454,000] mcumgr_img_grp: Failed to open flash area ID 2: -19

I tried changing the MTU but did not helped.

nordicjm, 2024-11-26:
> The code is working the error which I get is
> 
> > [00:04:21.453,000] mcumgr_img_grp: Image upload inspect failed: 22
> > [00:04:21.454,000] mcumgr_img_grp: Failed to open flash area ID 2: -19
> 
> I tried changing the MTU but did not helped. Above output is from the sample code which I am running

`-19` is no dev, so what I would assume is that your external memory peripheral is exclusive, meaning you either use the SPI driver to write/read on the memory, or you use it in XIP mode, but not both. You could enable serial recovery mode in MCUboot which would allow you to update the application from that instead, then wouldn't need to have MCUmgr in the application. If this is the case then the same thing would apply to your storage partition, it will be wholly unusable

MaazSk, 2024-11-26:
Can I update the code from CAN also using that option?

So, I just need to enable Serial recovery in the MCUboot to update the application right? I will check that as well thanks

nordicjm, 2024-11-26:
> Can I update the code from CAN also using that option?

No, serial (UART/USB CDC) only

> So, I just need to enable Serial recovery in the MCUboot to update the application right? I will check that as well thanks

Yes

closed 2025-02-09 by github-actions[bot]