zephyrproject-rtos/zephyr #82049: Bluetooth: Controller: ISO: HCI ISO data packets use more buffers than what is required

Thalley, 2024-11-26:
**Describe the bug**
When sending SDUs on a BIS (and probably also a CIS), where the Maximum SDU size is larger than the HCI ISO data packet size, then each HCI ISO data packet size seems to take multiple buffers. 

For example if the HCI ISO data packet buffer size is 200 and the maximum SDU size is 624 then each HCI ISO data packet takes up 4 buffers, regardless of the TX size. E.g. when just sending a single octet in the SDU, it still takes 4 buffers. 

**To Reproduce**
1) Checkout https://github.com/zephyrproject-rtos/zephyr/pull/81871
2) run the `compile.sh` script
3) run the `bis_fragment.sh` script
4) See error on the 4th TX due to buffer overrun

**Expected behavior**
Expect that when sending an SDU smaller than the HCI ISO data packet, then only a single buffer is used. If the SDU size is larger than the HCI ISO buffer size but less than twice the size, 2 buffers should be used, etc. 

**Impact**
Significant memory increase in the controller if applications need to work around this issue. 

**Logs and console output**
See https://github.com/zephyrproject-rtos/zephyr/pull/81871 BSIM checks

**Environment (please complete the following information):**
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/81871

**Additional context**
N/A

Thalley, 2025-02-25:
@cvinayak should we consider https://github.com/zephyrproject-rtos/zephyr/pull/82026 closing this? 

There are still issues with fragmentation, but those are new ones and not related to the issues described by this issue

cvinayak, 2025-02-25:
> @cvinayak should we consider https://github.com/zephyrproject-rtos/zephyr/pull/82026 closing this?

It is related and just re-iterates the design in the Zephyr Controller, that,

> For example if the HCI ISO data packet buffer size is 200 and the maximum SDU size is 624 then each HCI ISO data packet takes up 4 buffers, regardless of the TX size. E.g. when just sending a single octet in the SDU, it still takes 4 buffers.

ISOAL prepares empty ISO PDUs for each remaining (BN - 1) subevent that constitutes the fragmented PDU for a SDU size of 624.

This issue is still relevant, ISOAL does not have to allocate empty ISO PDUs, the Lower Link Layer could use scratch buffer for empty PDUs.

Thalley, 2025-07-29:
Looks like it has been fixed at some point as the test is no longer failing

closed 2025-07-29 by Thalley