zephyrproject-rtos/zephyr #82098: wifi: Unable to connect to EAP-TTLS-MSCHAPv2 and EAP-PEAP-MSCHAPv2 using nrf7002dk

TjazVracko, 2024-11-27:
**Describe the bug**

We're currently testing Wi-Fi connectivity with enterprise methods using FreeRADIUS as an authentication server and the wifi shell sample to connect (on a nrf7002dk). We are on the master branch of Zephyr, commit 

We also have trouble connecting via EAP-TTLS-MSCHAPv2 and EAP-PEAP-MSCHAPv2.

**To Reproduce**

<details>
<summary>1. Generate a fresh set of certificates on freeRADIUS</summary>

The CA and the certificates that we're using were generated on the freeRADIUS server, using the provided `make` commands. We are using [freeRADIUS v3.2.6](https://github.com/FreeRADIUS/freeradius-server/tree/release_3_2_6/raddb/certs). The steps are:

1. Modify `ca.cnf`, `server.cnf` and `client.cnf` files
3. Generate .pem/.crt/.key files with `make ca`, `make server` and `make client`
4. Copy the `ca.pem`, `client.crt` and `client.key` to `zephyr/samples/net/wifi/test_certs`.

</details>

2.  build and flash with:
    ```bash
    west build -b nrf7002dk/nrf5340/cpuapp -- -DEXTRA_CONF_FILE=overlay-enterprise.conf
    west flash
    ```
3. Attempt to connect:
    ```bash
    # EAP-TTLS-MSCHAPv2
    wifi connect -s "MikroTikChateau-WLAN-2" -c 108 -k 14 -I "user123" -P "password123"
    
    # EAP-PEAP-MSCHAPv2
    wifi connect -s "MikroTikChateau-WLAN-2" -c 108 -k 12 -I "user123" -P "password123"
    ```

4. Te connection request fails:
    ```text
    Connection request failed (1)
    ```

**Expected behavior**

We would expect the connection to be successful, since we can connect from a Ubuntu PC just fine, using the following:
[image]
[image]

**Impact**

Showstopper. We would expect this to work.

**Logs and console output**
/

**Environment (please complete the following information):**

 - OS: Ubuntu 22.04
 - Zephyr master branch- commit 0bc3393fb112ec80ebeab48cd023d69b1e9db757

**Additional context**
/

krish2718, 2024-11-27:
@MaochenWang1 @fengming-ye  can you please take a look at this?

MaochenWang1, 2024-11-27:
Do you enable these?
`CONFIG_WIFI_NM_WPA_SUPPLICANT_CRYPTO_ALT=y
CONFIG_WIFI_NM_WPA_SUPPLICANT_CRYPTO_ENTERPRISE=y
CONFIG_WIFI_NM_WPA_SUPPLICANT_EAPOL=y
CONFIG_EAP_TLS=y
CONFIG_EAP_PEAP=y
CONFIG_EAP_TTLS=y
CONFIG_EAP_MSCHAPV2=y`

krish2718, 2024-11-27:
@MaochenWang1  FYI this was done on nRF7002 but we haven't tried it ourselves :), so, if you can confirm that the procedure is correct and it works on RW612 then I can give it a shot on NRF70 and see if something is amiss.

krish2718, 2024-11-27:
@TjazVracko can you please share the WPA supplicant debug logs? `CONFIG_WIFI_NM_WPA_SUPPLICANT_LOG_LEVEL_DBG=y`

MaochenWang1, 2024-11-28:
FYI @Rex-Chen-NXP

TiborSovilj, 2024-11-28:
@MaochenWang1, the following are enabled by default:
```text
CONFIG_WIFI_NM_WPA_SUPPLICANT_CRYPTO_ALT=y
CONFIG_WIFI_NM_WPA_SUPPLICANT_CRYPTO_ENTERPRISE=y
CONFIG_WIFI_NM_WPA_SUPPLICANT_EAPOL=y
CONFIG_EAP_TLS=y
```
which is sufficient for a successful TLS auth.

---

When we enabled CONFIG_EAP_TTLS=y we encountered the following error:
```text
in function `eap_ttls_phase2_request_chap':
/home/tibor/Work/my-workspace/modules/lib/hostap/src/eap_peer/eap_ttls.c:810: undefined reference to `chap_md5'
```
After that, we enabled `CONFIG_EAP_MSCHAPV2=y`, and the code compiled successfully. We can confirm that  TTLS authentication also went successfully.

---

Currently, we are still investigating how to authenticate using EAP-PEAP-MSCHAPv2. The authentication process cycles infinitely in approx. one minute intervals, the RADIUS server does not report any errors, and the WiFi shell repeatedly indicates a "Connected" state.  However, when running the `wifi status` command, we see:
```text
Status: successful
==================
State: SCANNING
```

We will try again with logging enabled and share soon.

TjazVracko, 2024-11-28:
Just for clarity, @TiborSovilj is my coworker who is testing this :)

krish2718, 2024-11-28:
> @krish2718 Should I open a new issue for this?

No need, we don't need bugs for each issue, its discretionary.

krish2718, 2024-11-28:
> But I can not say if this may hint at a bigger problem and that the whole CMakeLists.txt should be checked again (each specific CONFIG_EAP_* must add all of its dependencies.

TTLS and PEAP are newly added and much testing hasn't been done, your fix LGTM.

MaochenWang1, 2024-11-28:
> FYI this was done on nRF7002 but we haven't tried it ourselves :), so, if you can confirm that the procedure is correct and it works on RW612 then I can give it a shot on NRF70 and see if something is amiss.

Yes, in progress. Will update the result.

krish2718, 2024-11-28:
> Currently, we are still investigating how to authenticate using EAP-PEAP-MSCHAPv2. The authentication process cycles infinitely in approx. one minute intervals, the RADIUS server does not report any [...]

> Status: successful
==================
State: SCANNING
We will try again with logging enabled and share soon.

Please share WPA supplicant debug logs as requested.

Rex-Chen-NXP, 2024-11-29:
Hi @All both sap and sta test base on 2 RW612 boards, cmds show as below.
【EAP_TLS】:
WPA2:
date set 2024-10-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 7 -w 2 -K whatever -I client1

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 7 -w 2 -a client1 -K whatever

WPA3 suite B:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 7 -w 2 -K whatever -I client1 -S 1

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 7 -w 2 -a client1 -K whatever -S 1

WPA3 suite B-192:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 7 -w 2 -K whatever -I client1 -S 2

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 7 -w 2 -a client1 -K whatever -S 2

【PEAPv0-MSCHAPv2】:
WPA2:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 12 -w 2 -K whatever -V 0  -a client1 -I nxp -P wireless 

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 12 -w 2 -a client1 -K whatever -V 0 -I nxp -P wireless 

WPA3 suite B:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 12 -w 2 -K whatever -V 0  -a client1 -I nxp -P wireless  -S 1

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 12 -w 2 -a client1 -K whatever -V 0 -I nxp -P wireless  -S 1

WPA3 suite B-192:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 12 -w 2 -K whatever -V 0  -a client1 -I nxp -P wireless  -S 2

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 12 -w 2 -a client1 -K whatever -V 0 -I nxp -P wireless  -S 2

【EAP-PEAP-GTC】:
WPA2:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 13 -w 2 -K whatever -a client1 -I nxp -P wireless 

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 13 -w 2 -a client1 -K whatever -I nxp -P wireless 

WPA3 suite B:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 13 -w 2 -K whatever  -a client1 -I nxp -P wireless  -S 1

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 13 -w 2 -a client1 -K whatever -I nxp -P wireless  -S 1

WPA3 suite B-192:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 13 -w 2 -K whatever -a client1 -I nxp -P wireless  -S 2

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 13 -w 2 -a client1 -K whatever -I nxp -P wireless  -S 2

【EAP-TTLS-MSCHAPv2】:
WPA2:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 14 -w 2 -K whatever  -a client1 -I nxp -P wireless 

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 14 -w 2 -a client1 -K whatever -I nxp -P wireless 

WPA3 suite B:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 14 -w 2 -K whatever  -a client1 -I nxp -P wireless  -S 1

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 14 -w 2 -a client1 -K whatever -I nxp -P wireless  -S 1

WPA3 suite B-192:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 14 -w 2 -K whatever  -a client1 -I nxp -P wireless  -S 2

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 14 -w 2 -a client1 -K whatever -I nxp -P wireless  -S 2

【EAP-PEAP-TLS】:
WPA2:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 15 -w 2 -K whatever -a client1 -I nxp -P wireless 

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 15 -w 2 -a client1 --key1-pwd whatever  --key2-pwd whatever -I nxp -P wireless

WPA3 suite B:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 15 -w 2 -K whatever -a client1 -I nxp -P wireless  -S 1

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 15 -w 2 -a client1 --key1-pwd whatever  --key2-pwd whatever -I nxp -P wireless  -S 1  

WPA3 suite B-192:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 15 -w 2 -K whatever  -a client1 -I nxp -P wireless  -S 2

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 15 -w 2 -a client1 --key1-pwd whatever  --key2-pwd whatever -I nxp -P wireless  -S 2

【EAP_TLS_SHA256】:
date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B8
wifi ap enable -s EapNet_AP -c 149 -k 16 -w 2 -K whatever -I client1

date set 2024-09-27 00:00:00
nxp_wifi set-mac 00:50:43:02:FE:B9
wifi connect -s EapNet_AP -c 149 -k 16 -w 2 -a client1 -K whatever

Rex-Chen-NXP, 2024-11-29:
Hi all, both sap and sta test base on 2 boards.
【EAP_TLS】:
WPA2:
date set 2024-10-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 7 -w 2 -K whatever -I client1

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 7 -w 2 -a client1 -K whatever

WPA3 suite B:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 7 -w 2 -K whatever -I client1 -S 1

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 7 -w 2 -a client1 -K whatever -S 1

WPA3 suite B-192:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 7 -w 2 -K whatever -I client1 -S 2

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 7 -w 2 -a client1 -K whatever -S 2

【PEAPv0-MSCHAPv2】:
WPA2:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 12 -w 2 -K whatever -V 0  -a client1 -I user -P pswd 

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 12 -w 2 -a client1 -K whatever -V 0 -I user -P pswd 

WPA3 suite B:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 12 -w 2 -K whatever -V 0  -a client1 -I user -P pswd  -S 1

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 12 -w 2 -a client1 -K whatever -V 0 -I user -P pswd  -S 1

WPA3 suite B-192:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 12 -w 2 -K whatever -V 0  -a client1 -I user -P pswd  -S 2

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 12 -w 2 -a client1 -K whatever -V 0 -I user -P pswd  -S 2

【EAP-PEAP-GTC】:
WPA2:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 13 -w 2 -K whatever -a client1 -I user -P pswd 

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 13 -w 2 -a client1 -K whatever -I user -P pswd 

WPA3 suite B:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 13 -w 2 -K whatever  -a client1 -I user -P pswd  -S 1

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 13 -w 2 -a client1 -K whatever -I user -P pswd  -S 1

WPA3 suite B-192:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 13 -w 2 -K whatever -a client1 -I user -P pswd  -S 2

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 13 -w 2 -a client1 -K whatever -I user -P pswd  -S 2

【EAP-TTLS-MSCHAPv2】:
WPA2:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 14 -w 2 -K whatever  -a client1 -I user -P pswd 

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 14 -w 2 -a client1 -K whatever -I user -P pswd 

WPA3 suite B:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 14 -w 2 -K whatever  -a client1 -I user -P pswd  -S 1

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 14 -w 2 -a client1 -K whatever -I user -P pswd  -S 1

WPA3 suite B-192:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 14 -w 2 -K whatever  -a client1 -I user -P pswd  -S 2

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 14 -w 2 -a client1 -K whatever -I user -P pswd  -S 2

【EAP-PEAP-TLS】:
WPA2:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 15 -w 2 -K whatever -a client1 -I user -P pswd 

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 15 -w 2 -a client1 --key1-pwd whatever  --key2-pwd whatever -I user -P pswd

WPA3 suite B:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 15 -w 2 -K whatever -a client1 -I user -P pswd  -S 1

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 15 -w 2 -a client1 --key1-pwd whatever  --key2-pwd whatever -I user -P pswd  -S 1  

WPA3 suite B-192:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 15 -w 2 -K whatever  -a client1 -I user -P pswd  -S 2

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 15 -w 2 -a client1 --key1-pwd whatever  --key2-pwd whatever -I user -P pswd  -S 2

【EAP_TLS_SHA256】:
date set 2024-09-27 00:00:00
wifi ap enable -s EapNet_AP -c 149 -k 16 -w 2 -K whatever -I client1

date set 2024-09-27 00:00:00
wifi connect -s EapNet_AP -c 149 -k 16 -w 2 -a client1 -K whatever

TjazVracko, 2024-11-29:
@krish2718 PR is here: https://github.com/zephyrproject-rtos/zephyr/pull/82309

krish2718, 2024-11-29:
@TjazVracko Thanks for the PR, with that is the connection working for you? or are you still seeing the issue above? If yes, please do share WPA supplicant debug logs.

TiborSovilj, 2024-11-29:
@krish2718  @MaochenWang1 
For the EAP-PEAP-MSCHAPv2 method:
```text
uart:~$ wifi connect -s MikroTikChateau-WLAN-2 -c 108 -k 12 -a user-tls -K sanri
 -I user123 -P password123
Connection requested
--- 267 messages dropped ---
[00:09:07.391,326] <dbg> wpa_supp: wpa_printf_impl: argv[1]: 0
[00:09:07.392,059] <dbg> wpa_supp: wpa_printf_impl: wlan0: Control interface command 'SELECT_NETWORK 0'
[00:09:07.392,211] <dbg> wpa_supp: wpa_printf_impl: CTRL_IFACE: SELECT_NETWORK id=0
[00:09:07.392,333] <dbg> wpa_supp: wpa_printf_impl: Fast associate: Old scan results
[00:09:07.392,517] <dbg> wpa_supp: wpa_printf_impl: wlan0: Setting scan request: 0.000000 sec
--- 55 messages dropped ---
[00:09:07.392,944] <dbg> wpa_supp: wpa_printf_impl: wlan0: State: DISCONNECTED -> SCANNING
--- 14 messages dropped ---
[00:09:07.393,096] <dbg> wpa_supp: wpa_printf_impl: Scan SSID - hexdump_ascii(len=22):
[00:09:07.393,127] <dbg> wpa_supp: _wpa_hexdump_ascii: 
                                   4d 69 6b 72 6f 54 69 6b  43 68 61 74 65 61 75 2d |MikroTik Chateau-
                                   57 4c 41 4e 2d 32                                |WLAN-2           
[00:09:07.393,310] <dbg> wpa_supp: wpa_printf_impl: wlan0: Include wildcard SSID in the scan request
--- 27 messages dropped ---
[00:09:07.393,768] <dbg> wpa_supp: wpa_printf_impl: wlan0: Add radio work 'scan'@0x2003b638
[00:09:07.394,012] <dbg> wpa_supp: wpa_printf_impl: wlan0: First radio work item in the queue - schedule start immediately
[00:09:07.473,480] <dbg> wpa_supp: wpa_printf_impl: RRM: No RRM in network
[00:09:07.487,426] <dbg> wpa_supp: wpa_printf_impl: SME: Association Request IEs - hexdump(len=45): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 d
[00:09:07.508,605] <dbg> wpa_supp: wpa_printf_impl: EAPOL: SUPP_BE entering state IDLE
[00:09:07.508,697] <dbg> wpa_supp: wpa_printf_impl: EAP: EAP entering state INITIALIZE
[00:09:07.508,789] <dbg> wpa_supp: wpa_printf_impl: EAP: EAP entering state IDLE
[00:09:07.509,033] <dbg> wpa_supp: wpa_printf_impl: wlan0: Setting authentication timeout: 10 sec 0 usec
[00:09:07.509,216] <dbg> wpa_supp: wpa_printf_impl: wlan0: Cancelling scan request
[00:09:07.509,429] <inf> wpa_supp: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
[00:09:07.509,918] <dbg> wpa_supp: wpa_printf_impl: Received len: 33, msg_len:33 - Associated with 78:9a:18:86:7a:36->END
[00:09:07.510,162] <dbg> wpa_supp: wpa_printf_impl: Received len: 40, msg_len:40 - CTRL-EVENT-SUBNET-STATUS-UPDATE status=0->END
[00:09:07.510,375] <dbg> wpa_supp: wpa_printf_impl: Event not supported: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
--- 19 messages dropped ---
[00:09:08.505,157] <dbg> wpa_supp: wpa_printf_impl: EAP: EAP entering state SEND_RESPONSE
--- 36 messages dropped ---
[00:09:08.505,249] <dbg> wpa_supp: wpa_printf_impl: EAP: EAP entering state IDLE
[00:09:08.505,371] <dbg> wpa_supp: wpa_printf_impl: EAPOL: SUPP_BE entering state RESPONSE
[00:09:08.505,432] <dbg> wpa_supp: wpa_printf_impl: EAPOL: txSuppRsp
[00:09:08.505,584] <dbg> wpa_supp: wpa_printf_impl: TX EAPOL: dst=78:9a:18:86:7a:36
[00:09:08.505,950] <dbg> wpa_supp: wpa_printf_impl: TX EAPOL - hexdump(len=17): 01 00 00 0d 02 00 00 0d 01 75 73 65 72 2d 74 6c 73
[00:09:08.506,774] <dbg> wpa_supp: wpa_printf_impl: EAPOL: SUPP_BE entering state RECEIVE
[00:09:08.580,780] <dbg> wpa_supp: wpa_printf_impl: Event not supported: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=25
[00:09:08.581,054] <dbg> wpa_supp: wpa_printf_impl: Received len: 70, msg_len:70 - CTRL-EVENT-EAP-STATUS status='accept proposed method' parameter='D
[00:09:08.581,268] <dbg> wpa_supp: wpa_printf_impl: Event not supported: CTRL-EVENT-EAP-STATUS status='accept proposed method' parameter='PEAP'
[00:09:08.581,542] <dbg> wpa_supp: wpa_printf_impl: Received len: 60, msg_len:60 - CTRL-EVENT-EAP-METHOD EAP vendor 0 method 25 (PEAP) selected->END
[00:09:08.581,756] <dbg> wpa_supp: wpa_printf_impl: Event not supported: CTRL-EVENT-EAP-METHOD EAP vendor 0 method 25 (PEAP) selected
[00:09:08.582,000] <dbg> wpa_supp: wpa_printf_impl: Received len: 48, msg_len:48 - CTRL-EVENT-EAP-FAILURE EAP authentication failed->END
[00:09:08.582,183] <dbg> wpa_supp: wpa_printf_impl: Event not supported: CTRL-EVENT-EAP-FAILURE EAP authentication failed
[00:09:09.509,246] <dbg> wpa_supp: wpa_printf_impl: EAPOL: startWhen --> 0
Connected
--- 16 messages dropped ---
[00:09:10.591,522] <inf> wpa_supp: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MikroTikChateau-WLAN-2" auth_failures=2 duration=20 reason=CONN_FD
--- 27 messages dropped ---
[00:09:10.591,857] <dbg> wpa_supp: wpa_printf_impl: wlan0: Consecutive connection failures: 5 --> request scan in 10000 ms
[00:09:10.592,193] <dbg> wpa_supp: wpa_printf_impl: wlan0: Ignore new scan request for 10.000000 sec since an earlier request is scheduled to trigger
[00:09:10.592,376] <dbg> wpa_supp: wpa_printf_impl: wlan0: WPA: Clear old PMK and PTK
[00:09:10.602,447] <dbg> wpa_supp: wpa_printf_impl: Received len: 48, msg_len:48 - Authentication with 78:9a:18:86:7a:36 timed out.->END
[00:09:10.602,722] <dbg> wpa_supp: wpa_printf_impl: Received len: 76, msg_len:76 - CTRL-EVENT-DISCONNECTED bssid=78:9a:18:86:7a:36 reason=3 locally_D
--- 3 messages dropped ---
[00:09:10.603,088] <dbg> wpa_supp: wpa_printf_impl: Received len: 111, msg_len:111 - CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MikroTikChateau-WLAN-2D
[... cut ...]
[00:21:56.483,764] <dbg> wpa_supp: wpa_printf_impl: wlan0: No suitable network found
[00:21:56.484,008] <dbg> wpa_supp: wpa_printf_impl: wlan0: Setting scan request: 5.000000 sec
[00:21:56.484,527] <dbg> wpa_supp: wpa_printf_impl: Received len: 40, msg_len:40 - CTRL-EVENT-BSS-ADDED 1 78:9a:18:86:7a:36->END
[00:21:56.484,863] <dbg> wpa_supp: wpa_printf_impl: Received len: 24, msg_len:24 - CTRL-EVENT-SCAN-RESULTS ->END
[00:21:56.485,168] <dbg> wpa_supp: wpa_printf_impl: Received len: 29, msg_len:29 - CTRL-EVENT-NETWORK-NOT-FOUND ->END
```

TiborSovilj, 2024-11-29:
@krish2718  @MaochenWang1 
I've uploaded 2 log files for EAP-PEAP_MSCHAPv2 tests.
`mschap_logs.txt` is based on a `wifi connect` **without** `-V 0` and `-w 2` flags.
`mschap_logs_xtra_flags.txt` is based on a `wifi connect` **with** `-V 0` and `-w 2` flags.

[mschapv2_logs.txt](https://github.com/user-attachments/files/17956391/mschapv2_logs.txt)
[mschapv2_logs_xtra_flags.txt](https://github.com/user-attachments/files/17956392/mschapv2_logs_xtra_flags.txt)

krish2718, 2024-12-03:
+ @rado17 @sachinthegreen @bama-nordic

Rex-Chen-NXP, 2024-12-04:
> TiborSovilj

Hi, TiborSovilj, I have a simple check of the logs from you, and also try the cmds you shared base on RW612 boards, summary from my side:
A. For the [mschapv2_logs.txt], below logs observed, the default configured version as 1,  I think you miss the -V 0 option.
uart:~$ [00:00:59.440,124] <dbg> wpa_supp: wpa_printf_impl: EAP-PEAP: Start (server ver=0, own ver=1)
uart:~$ [00:00:59.440,216] <wrn> wpa_supp: EAP-PEAP: Failed to select forced PEAP version 1

B. For the [mschapv2_logs_xtra_flags.txt], ,below logs observed,  scan result not matched, I think you should remove the -w 2 option.
uart:~$ [00:01:28.502,563] <dbg> wpa_supp: wpa_printf_impl: wlan0:    skip RSN IE - no mgmt frame protection
uart:~$ [00:01:28.502,746] <dbg> wpa_supp: wpa_printf_impl: wlan0:    skip - MFP Required but network not MFP Capable
uart:~$ [00:01:28.503,082] <dbg> wpa_supp: wpa_printf_impl: wlan0: No suitable network found

Could you try the cmds shared from me few days ago in this issue ticket?
Or you can also try with the below cmd which add -V 0 option based on you shared the cmd for [mschapv2_logs.txt] :
wifi connect -s "MikroTikChateau-WLAN-2" -c 108 -k 12 -a user-tls -K san  -V 0

Rex-Chen-NXP, 2024-12-04:
Hi @krish2718 
Should I raise PR, add all the CMDs I shared in this issue ticket to doc/connectivity/networking/api/wifi.rst?

krish2718, 2024-12-04:
> Hi @krish2718 Should I raise PR, add all the CMDs I shared in this issue ticket as demos to doc/connectivity/networking/api/wifi.rst?

We have the main command with an example, IMHO that would suffice.

TiborSovilj, 2024-12-10:
@Rex-Chen-NXP @krish2718 @TjazVracko 

Sorry for the delayed reply,
It turns out that minimal working combination of flags for the EAP-PEAP-MSCHAPv2 is:

```bash
wifi connect -s "MikroTikChateau-WLAN-2" -k 12 -I "user123" -P "password123" -V 0 
```

i.e., with `-V 0`, but without `-w 2`.

Adding the `-a` and `-K` flags made no difference when we attempted to connect, i.e. the authentication was successful regardless.

Thank you for your input.

krish2718, 2024-12-11:
@TiborSovilj @TjazVracko  is this now working fine?

TiborSovilj, 2024-12-11:
@krish2718 

Here are the logs of the successful attempt on our network:
[mschapv2__new.txt](https://github.com/user-attachments/files/18091484/mschapv2__new.txt)

How do we determine the combination of flags required for any given network?
What information about the network do we require, and how does that map to the flags?

krish2718, 2024-12-15:
> How do we determine the combination of flags required for any given network?
> What information about the network do we require, and how does that map to the flags?

I don't understand, can you please elaborate on `flags`? 

If you are asking how does the client know the EAP type, then at Wi-Fi level it isn't possible, the scan can only display that the network in 802.1x (EAP), and then it's up to the supplicant and the AAA server to negotiate a particular method by means on [NAK](https://datatracker.ietf.org/doc/html/rfc3748#section-5.3).

In Zephyr we only support a single authentication mechanism as it's only an enumeration, but we could add something like `EAP_ALL` where supplicant enabled all methods and starts with highest and depending on the network AAA server NAK can select one. But this comes at the extra cost of memory as we enable all methods. If this is what you are after, I would suggest to create an enhancement request in Wi-Fi project.

krish2718, 2024-12-15:
> > How do we determine the combination of flags required for any given network?
> > What information about the network do we require, and how does that map to the flags?
> 
> I don't understand, can you please elaborate on `flags`?

Looking at previous replies, I guess by `flags` you mean the various shell command line options. In that case there isn't a defined rule set, the command line options should be compatible with the network settings (Authenticator/AP, AAA server). E.g., `-w 2` means STA is expecting `MFP` to be enabled by the AP but if it's not it won't connect. And similarly the `-V` would need to match the PEAP version being used by the network.

TjazVracko, 2024-12-16:
@krish2718 Thank you. That makes sense.

So currently, a connection attempt by the device has to match the network and AAA server settings **exactly**.

If, for example, we only know that the network is `EAP-PEAP-MSCHAPv2`, we could try all of the combinations of settings until we successfully connect?

------------------------------

Something like `EAP_ALL` would be very useful. Could it work so that it iterates only over the enabled methods and checks with the AAA server if any of them are supported?

krish2718, 2024-12-16:
> If, for example, we only know that the network is EAP-PEAP-MSCHAPv2, we could try all of the combinations of settings until we successfully connect?

If user configures EAP-PEAP-MSCHAPv2  supplicant will only try that and nothing else.

> Something like EAP_ALL would be very useful. Could it work so that it iterates only over the enabled methods and checks with the AAA server if any of them are supported?

Yes, `EAP_ALL` is useful but need to see how to handle the credentials, we would need all possible credentials from the user for which again he needs knowledge of network (from network admin), which kind of make its moot as security also can also be known with the credentials. But if we take `phase1, phase2 certs and passphrases` we could then selectively pick based on security method, we can check how supplicant behaves and replicate that.

@MaochenWang1 @Rex-Chen-NXP @jukkar WDYT?

MaochenWang1, 2024-12-16:
> If, for example, we only know that the network is `EAP-PEAP-MSCHAPv2`, we could try all of the combinations of settings until we successfully connect?
What `combinations of settings` do you mean?

Rex-Chen-NXP, 2024-12-17:
Hi @TiborSovilj
A. How do we determine the combination of flags required for any given network?
----I guess you want to know how to input the parameters by shell to configure the connection. As @krish2718 explained early( the command line options should be compatible with the network settings), network admin should know the details, for example ssid, EAP type, psk, etc. Then can configure it by shell correctly. 

B. What information about the network do we require, and how does that map to the flags?
a. What information about the network do we require
----This question is same as above A.
b. how does that map to the flags?
----shell usage provide simple description.

TjazVracko, 2024-12-18:
@Rex-Chen-NXP @MaochenWang1 @krish2718 

Thank you for all the answers. I think we have all the knowledge we need now.
We were missing knowledge about the EAP version and MFP.

Since everything seems to work, this issue can be closed.

closed 2024-12-18 by krish2718