zephyrproject-rtos/zephyr #82111: drivers: uart: uart_async_api: drivers.uart.async_api fails

katgiadla, 2024-11-27:
**Describe the bug**
Test  `tests/drivers/uart/uart_async_api/drivers.uart.async_api` fails.

Observed for `nrf54h20dk/nrf54h20/cpuapp`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54h20dk/nrf54h20/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -s drivers.uart.async_api -p nrf54h20dk/nrf54h20/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase" --fixture gpio_loopback`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-837-gb88d0bf8c7ee ***
Running TESTSUITE uart_async_chain_read
===================================================================
UART instance:uart@9d6000
START - test_chained_read
 PASS - test_chained_read in 0.427 seconds
===================================================================
TESTSUITE uart_async_chain_read succeeded
Running TESTSUITE uart_async_chain_read
===================================================================
UART instance:uart@8e6000
START - test_chained_read

    Assertion failed at zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:430: uart_async_chain_read_test_chained_read: (rx_data_idx not equal to sizeof(tx_buf))
Unexpected amount of data received 0 exp:10
 FAIL - test_chained_read in 0.096 seconds
===================================================================
TESTSUITE uart_async_chain_read failed.
Running TESTSUITE uart_async_chain_write
===================================================================
UART instance:uart@9d6000
START - test_chained_write
 PASS - test_chained_write in 0.003 seconds
===================================================================
TESTSUITE uart_async_chain_write succeeded
Running TESTSUITE uart_async_chain_write
===================================================================
UART instance:uart@8e6000
START - test_chained_write

    Assertion failed at zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:874: uart_async_chain_write_test_chained_write: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_chained_write in 0.125 seconds
===================================================================
TESTSUITE uart_async_chain_write failed.
Running TESTSUITE uart_async_double_buf
===================================================================
UART instance:uart@9d6000
START - test_double_buffer
 PASS - test_double_buffer in 0.975 seconds
===================================================================
TESTSUITE uart_async_double_buf succeeded
Running TESTSUITE uart_async_double_buf
===================================================================
UART instance:uart@8e6000
START - test_double_buffer

    Assertion failed at zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:507: uart_async_double_buf_test_double_buffer: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_double_buffer in 0.123 seconds
[... 126 lines cut ...]
 - FLAKY - [uart_async_single_read.test_single_read] - (Failed 1 of 2 attempts) - duration = 0.311 seconds

SUITE FAIL -   0.00% [uart_async_timeout]: pass = 0, fail = 1, skip = 0, total = 1 duration = 3.123 seconds
 - FLAKY - [uart_async_timeout.test_forever_timeout] - (Failed 1 of 2 attempts) - duration = 3.123 seconds

SUITE FAIL -   0.00% [uart_async_write_abort]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.123 seconds
 - FLAKY - [uart_async_write_abort.test_write_abort] - (Failed 1 of 2 attempts) - duration = 0.123 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: `v4.0.0-837-gb88d0b`

**Additional information:**
- used set of pins:
```
psels = <NRF_PSEL(UART_TX, 0, 6)>;
psels = <NRF_PSEL(UART_RX, 0, 7)>;
```

nordic-krch, 2025-01-16:
Is this still failing? On my setup it is passing.

nordic-piks, 2025-01-17:
Indeed, cannot reproduce.

closed 2025-01-17 by nordic-piks