zephyrproject-rtos/zephyr #82129: modbus_server.c: warning: 'err' may be used uninitialized

wniva, 2024-11-27:
**Describe the bug**
When using MODBUS I get warnings that "err" is not initialized.
This is a non-critical warning, but it would be nice if this could be corrected in the next release.

**To Reproduce**
My config:
CONFIG_MODBUS=y
CONFIG_MODBUS_ROLE_SERVER=y
CONFIG_MODBUS_FP_EXTENSIONS=n

**Logs and console output**
modbus_server.c: In function 'mbs_fc04_inreg_read':
modbus_server.c:500:20: warning: 'err' may be used uninitialized [-Wmaybe-uninitialized]
modbus_server.c:422:13: note: 'err' was declared here

**Environment:**
- OS: Windows 10
- nRF Connect SDK v2.7.0-5cb85570ca43
- Zephyr OS v3.6.99-100befc70c74

kartben, 2024-11-27:
@wniva a pull request fixing this would certainly be appreciated :)

jfischer-no, 2025-01-27:
Fixed in commit bda1970b330090414bc0c633e364d347d62cf984

closed 2025-01-27 by jfischer-no

wniva, 2025-01-28:
Thank you for the correction! :smiley:

jfischer-no, 2025-01-28:
> Thank you for the correction! 😃

Thanks to @gatzka 😄

wniva, 2025-07-02:
Note: Not yet included in the currently released Nordic v3.0.2 (Zephyr version 4.0.99) … :-(