zephyrproject-rtos/zephyr #82152: zcbor with CANONICAL mode fails when decoding some floats

tautologyclub, 2024-11-27:
Situation: Create a JSON `'{"my_float": 0.0}'` and convert it to CBOR using `zcbor.py`, then send the payload to Zephyr for decoding, with `CONFIG_ZCBOR_CANONICAL=y`

zcbor.py chooses to encode 0.0 as a 8 byte float - I'm honestly not sure if this is honestly INCORRECT, canonically speaking, because who can honestly say what an optimal way to encode a float is? I mean, in the integer case, it's obvious, but floats encode precision too, so it's not like it's STRICTLY incorrect to encode it as a longer float.

Anyway, with ZCBOR_CANONICAL, we hit

```
#ifdef ZCBOR_CANONICAL
		ZCBOR_ERR_IF((zcbor_header_len_ptr(result, result_len) != (len + 1)),
			ZCBOR_ERR_INVALID_VALUE_ENCODING);
#endif
````

inside `static bool value_extract(zcbor_state_t *state, void *const result, size_t result_len)` from `zcbor_decode.c`.

In my mind, canonical encoding of floats doesn't really make sense. 

Besides, this is the help text for the Kconfig:

```
config ZCBOR_CANONICAL
	bool "Produce canonical CBOR"
	help
	  Enabling this will prevent zcbor from creating lists and maps with
	  indefinite-length arrays (it will still decode them properly).
```

which is clearly an inaccurate description of the Kconfig.

oyvindronningstad, 2024-11-28:
Thanks for the report. What is happening here is that converting JSON to CBOR with zcbor.py does not have a CANONICAL mode. Only the C library does. You can turn off checking of canonical CBOR on the C library decoder at run time if you wish, via the `state->constant_state->enforce_canonical` flag. I will track this as a feature request for a canonical mode in the script.

oyvindronningstad, 2024-11-28:
Looking at it again, that error you are getting is part of an older version which does not have the `enforce_canonical` flag, which came in 0.9.0.

And yes, there is an "optimal" way to encode a float: It's the shortest variant that loses no precision to rounding, e.g. if converting from float64 to float16 back to float64 leaves the value unchanged, it can be encoded as float16 with no loss of precision, and should be under canonical CBOR.

tautologyclub, 2024-11-28:
I dunno, I read https://datatracker.ietf.org/doc/html/rfc7049#page-26 and to me the language is not precise enough to consider canonical floats well defined. I mean I would have preferred if the lib obeyed the Kconfig description because the Kconfig description is literally what we want - to be strict when encoding, but loose when decoding.

oyvindronningstad, 2024-11-29:
The runtime flag only affects decoding, so encoding will still be canonical.

oyvindronningstad, 2024-11-29:
https://github.com/zephyrproject-rtos/zephyr/pull/82318