zephyrproject-rtos/zephyr #82206: Bluetooth: Mesh: Proxy feature is broken in Zephyr 4.0.0

PavelVPV, 2024-11-28:
**Describe the bug**
After https://github.com/zephyrproject-rtos/zephyr/pull/78914 the Mesh Proxy feature is broken. It is not possible anymore to connect to the proxy node. This change can also result in mem fault: https://discord.com/channels/720317445772017664/733036850226266133/1311536117740474389

**To Reproduce**
1. Check out Zephyr 4.0.0
2. Build tests/bluetooth/mesh_shell
3. Start PB-GATT provisioning bearer by typing `mesh prov pb-gatt on` in shell
4. Use nRF Mesh mobile app for iOS or Android to provision and configure the device
5. Observe that configuration phase fails. The mobile app can't receive Composition Data of the device.

**Expected behavior**
Configuration works fine over mobile app.

**Impact**
The Mehs Proxy feature is broken in Zephyr 4.0.0

**Environment (please complete the following information):**

- Zephyr 4.0.0

LingaoM, 2024-11-28:
Currently there is no bbsim for proxy ?!

PavelVPV, 2024-11-28:
> Currently there is no bbsim for proxy ?!

Those we have test advertisement interval, advertised subnets, etc., but not data transfer between proxy client (mobile phone) and proxy server.

PR #82205 (2024-12-06): [Backport v4.0-branch] bluetooth: mesh: proxy_msg: Fix extracting role from k_work
Backport f5409bd3deba779263fb58bb06a0573e034795cf from #81900.

Fixes #82206
closed 2025-01-28 by PavelVPV