zephyrproject-rtos/zephyr #82221: Referencing ST morpho header in custom shield leads to devicetree error

PhilippFinke1988, 2024-11-28:
**Describe the bug**

I'm currently working on a shield definition for a custom shield targeted to be connected to the ST morpho headers of the _nucleo_g071rb_ and _nucleo_g0b1re_ boards (see boards folder in 
[application.zip](https://github.com/user-attachments/files/17946062/application.zip)). Both boards contain nexus nodes describing the pin mapping of their ST morpho header (see https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/st/nucleo_g071rb/st_morpho_connector.dtsi and https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/st/nucleo_g0b1re/st_morpho_connector.dtsi).

Trying to build my application by calling

```
west build -b nucleo_g071rb . -p  --build-dir build\debug --shield nucleo_my_shield
```

leads to a devicetree error

```
-- west build: making build dir C:\my-ws\application\build\debug pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/my-ws/application
-- CMake version: 3.30.4
-- Found Python3: c:/my-ws/.venv/Scripts/python.exe (found suitable version "3.12.6", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/my-ws/zephyr/.cache
-- Zephyr version: 3.7.0 (C:/my-ws/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: nucleo_g071rb, qualifiers: stm32g071xx
-- Shield(s): nucleo_my_shield
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (C:/Users/finkeph/.zephyr_ide/toolchains/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (C:/Users/finkeph/.zephyr_ide/toolchains/zephyr-sdk-0.16.8)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/my-ws/zephyr/boards/st/nucleo_g071rb/nucleo_g071rb.dts
-- Found devicetree overlay: C:/my-ws/application/boards/shields/nucleo_my_shield/nucleo_my_shield.overlay
devicetree error: expected '#gpio-cells' property on <Node /soc/interrupt-controller@e000e100 in 'C:/my-ws/zephyr/misc/empty_file.c'> (referenced by <Node /soc/spi@40013000 in 'C:/my-ws/zephyr/misc/empty_file.c'>)
-- In: C:/my-ws/application/build/debug/zephyr, command: c:/my-ws/.venv/Scripts/python.exe;C:/my-ws/zephyr/scripts/dts/gen_defines.py;--dts;C:/my-ws/application/build/debug/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;C:/my-ws/zephyr/dts/bindings;--header-out;C:/my-ws/application/build/debug/zephyr/include/generated/zephyr/devicetree_generated.h.new;--dts-out;C:/my-ws/application/build/debug/zephyr/zephyr.dts.new;--edt-pickle-out;C:/my-ws/application/build/debug/zephyr/edt.pickle;--vendor-prefixes;C:/my-ws/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at C:/my-ws/zephyr/cmake/modules/dts.cmake:296 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  C:/my-ws/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  C:/my-ws/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/my-ws/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
←[91mFATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' -DWEST_PYTHON=c:/my-ws/.venv/Scripts/python.exe '-BC:\my-ws\application\build\debug' -GNinja -DBOARD=nucleo_g071rb -DSHIELD=nucleo_my_shield '-SC:\my-ws\application'
```

Referencing the GPIO nodes directly instead of st_morpho_header works just fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Download [application.zip](https://github.com/user-attachments/files/17946062/application.zip)
2. Initialize a workspace for the application
3. Build the application using **west build -b nucleo_g071rb . -p  --build-dir build\debug --shield nucleo_my_shield**
4. See error

**Expected behavior**

The application should build without any error.

**Impact**

This issue is rather an annyance.

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain Zephyr SDK 0.6.18
 - Zephyr revision v3.7.0

PhilippFinke1988, 2024-11-28:
Sure, I pushed the code to https://github.com/PhilippFinke1988/nucleo_my_shield

rruuaanng, 2024-11-29:
I'm in the running of building, triggered it.
```
In file included from <command-line>:
C:/Users/errorput/Desktop/application/zephyr/boards/st/nucleo_g071rb/nucleo_g071rb.dts:10:10: fatal error: st/g0/stm32g071r(6-8-b)tx-pinctrl.dtsi: No such file or directory
   10 | #include <st/g0/stm32g071r(6-8-b)tx-pinctrl.dtsi>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMake Error at zephyr/cmake/modules/extensions.cmake:4562 (message):
  failed to preprocess devicetree files (error code 1):
  C:/Users/errorput/Desktop/application/zephyr/boards/st/nucleo_g071rb/nucleo_g071rb.dts;C:/Users/errorput/Desktop/application/boards/shields/nucleo_my_shield/nucleo_my_shield.overlay
Call Stack (most recent call first):
  zephyr/cmake/modules/dts.cmake:257 (zephyr_dt_preprocess)
  zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)
```

I'm questioning whether it exists
```
#include <st/g0/stm32g071r(6-8-b)tx-pinctrl.dtsi>
```

PhilippFinke1988, 2024-11-29:
This error is strange. The dtsi is part of the **hal_stm32** module (see https://github.com/zephyrproject-rtos/hal_stm32/blob/main/dts/st/g0/stm32g071r(6-8-b)tx-pinctrl.dtsi).

rruuaanng, 2024-11-29:
I downloaded the application you provided and entered `west init` in the directory. Everything was carried out according to what you provided. Maybe it was caused by my negligence somewhere.

FRASTM, 2024-11-29:
> I'm questioning whether it exists
> 
> ```
> #include <st/g0/stm32g071r(6-8-b)tx-pinctrl.dtsi>
> ```

Yes, see https://github.com/zephyrproject-rtos/hal_stm32/blob/main/dts/st/g0/stm32g071c(6-8-b)tx-pinctrl.dtsi

Did you west update ?

rruuaanng, 2024-11-29:
> The steps to reproduce I described might be a bit confusing. I updated them.

It disappears. Maybe the process could be a little more detailed. I'm wondering if it's really appropriate to use characters like ( for dtsi files? For the version we can choose `_` or `-`

PhilippFinke1988, 2024-12-02:
> If the nucleo_my_shield.overlay has `cs-gpios = <&st_morpho_header ST_MORPHO_L_9 GPIO_ACTIVE_LOW>;` it builds. But from **ST_MORPHO_L_11** if fails : `devicetree error: expected '#gpio-cells' property on <Node /soc/interrupt-controller@e000e100 in '~/zephyrproject/zephyr/misc/empty_file.c'> (referenced by <Node /soc/spi@40013000 in '~/zephyrproject/zephyr/misc/empty_file.c'>)`
> 
> --> <&st_morpho_header 9 or ST_MORPHO_L_10 seems to be a limit !?

I tested this myself and can confirm that there seems to be a limit.

I compared the definition of the ST morpho header with the arduino one in https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/st/nucleo_g071rb/arduino_r3_connector.dtsi and found out that _gpio-map-mask_ are different.

For arduino its

```
gpio-map-mask = <0xffffffff 0xffffffc0>;
```
and for ST morpho

```
gpio-map-mask = <ST_MORPHO_PIN_MASK 0x0>;
```

where ST_MORPHO_PIN_MASK is defined as 0xFF in https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/dt-bindings/gpio/st-morpho-header.h.

Might this be a problem?

FRASTM, 2024-12-02:
Can you pleasz check with #82410

PhilippFinke1988, 2024-12-02:
> Can you please check with #82410

That seems to work. But I don't understand why you removed L25 and L27. These pins are connected via a solder bridge that can be connected by the user.

FRASTM, 2024-12-02:
> > Can you please check with #82410
> 
> That seems to work. But I don't understand why you removed L25 and L27. These pins are connected via a solder bridge that can be connected by the user.

Ok for keeping Pin 25 & 27 unchanged

PhilippFinke1988, 2024-12-02:
Does it make sense to check for this kind of issue in other ST morpho header definitions?

PhilippFinke1988, 2024-12-02:
@FRASTM I checked all files with nodes using "st-morpho-header" as compatible and found that following nodes seem to have the same issue. Could you also fix these, please?

- https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/st/nucleo_g070rb/st_morpho_connector.dtsi
- https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/st/nucleo_g0b1re/st_morpho_connector.dtsi

FRASTM, 2024-12-02:
> @FRASTM I checked all files with nodes using "st-morpho-header" as compatible and found that following nodes seem to have the same issue. Could you also fix these, please?
>
> * [...]

done in the PR https://github.com/zephyrproject-rtos/zephyr/pull/82410

PhilippFinke1988, 2024-12-03:
@FRASTM Thank you! I approved PR #82410. I think we can close this issue after the PR has been merged

closed 2025-01-17 by kartben