zephyrproject-rtos/zephyr #82244: ESP32-S3: Application never enables quad flash mode

horakdavid, 2024-11-28:
**Describe the bug**
Currently, only the bootloader can configure flash in quad mode. When building application with quad flash mode enabled, flash operations may not work when bootloader was built with dual spi mode. Application should enable quad mode when it is being build with quad flash mode enabled.

**To Reproduce**
Build and flash mcuboot with `CONFIG_ESPTOOLPY_FLASHMODE_DIO=y` and application with `CONFIG_ESPTOOLPY_FLASHMODE_QIO=y`. It is not possible to read flash memory.

**Expected behavior**
Application should enable QIO mode and flash operations should work.

**Impact**
It is not possible to upgrade to app with quad spi flash enabled when mcuboot was built only with dual flash mode.

**Environment**
Zephyr v4.0.0

sylvioalves, 2024-11-28:
@wmrsouza

sylvioalves, 2024-11-28:
@horakdavid can you provide which SoC or board you have tested?

horakdavid, 2024-11-29:
@sylvioalves custom board with ESP32-S3-PICO-1-N8R2

horakdavid, 2024-11-29:
@sylvioalves I suppose this problem is related to all socs in esp32 family. 

The problem is that app access the flash as it is configured in quad spi mode, but it is actually in dual spi mode. 

Calling `bootloader_enable_qio_mode` on application startup before mapping flash segments and modifying linker script to link the function in ram solved this issue for me, but since this function is intended to be used only by bootloaders I don't think this is a proper solution.

sylvioalves, 2024-11-29:
@horakdavid, thanks, I'll take a look.

sylvioalves, 2024-12-02:
@horakdavid You mentioned you have tested with mcuboot. Is it Zephyr's MCUboot port, i.e, `--sysbuild` option or Espressif´'s MCUboot port?

horakdavid, 2024-12-02:
@sylvioalves I tried only the Zephyr's MCUboot port

PR #85625 (2025-02-14): soc: espressif: rework code startup to improve flash and spiram loader
This PR reworks all Espressif's SOC initialization code to fix booting issues regaring OCTAL FLASH/SPIRAM.
This also updates bootloader (either Simple Boot and MCUBoot) to use ROM functions during flash read/write operations, ensuring no dependencies of the flash driver initialization.
Runner handler is now common to all SoCs and fixes flash mode and frequency configuration, required for some flash memories.
Esptool is also updated to v4.8.1.

Fixes #64757
Fixes #85246
Fixes #83743
Fixes #82244
closed 2025-02-14 by kartben