zephyrproject-rtos/zephyr #82251: samples: subsys: mgmt: mcumgr: smp_svr: sample.mcumgr.smp_svr.* fail

katgiadla, 2024-11-28:
**Describe the bug**
Samples:

- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.cdc`
- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.fs`
- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.ram_load`
- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.serial`
- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.serial-console`
- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.shell`
- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.shell_mgmt`
- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.mcuboot_flags.direct_xip_withrevert`
- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.ram_load.serial`
- `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.ram_load.serial.fs.shell`

 fail.

Observed for `nrf52840dk/nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/subsys/mgmt/mcumgr/smp_svr -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
Logs for `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.cdc`:
```
*** Booting MCUboot eb9420679895 ***
*** Using Zephyr OS build v4.0.0-735-geb4f2fb44a0e ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Image index: 0, Swap type: none
I: Bootloader chainload address offset: 0x10000
I: Image version: v0.0.0
I: Jumping to the first image slot
*** Booting Zephyr OS build v4.0.0-735-geb4f2fb44a0e ***
[00:00:00.000,793] <inf> smp_sample: build time: Nov 24 2024 22:42:40
```

Logs for `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.ram_load.serial.fs.shell`:
```
*** Booting MCUboot eb9420679895 ***
*** Using Zephyr OS build v4.0.0-735-geb4f2fb44a0e ***
I: Starting bootloader
I: Primary   slot: version=0.0.0+0
I: Image 0 Secondary slot: Image not found
I: Image 0 RAM loading to 0x20006000 is succeeded.
I: Image 0 loaded from the primary slot
I: Bootloader chainload address offset: 0x20006000
I: Image version: v0.0.0
I: Jumping to the first image slot
```

Logs for `samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumgr.smp_svr.fs`:
```
*** Booting MCUboot eb9420679895 ***
*** Using Zephyr OS build v4.0.0-735-geb4f2fb44a0e ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Image index: 0, Swap type: none
I: Bootloader chainload address offset: 0xc000
I: Image version: v0.0.0
I: Jumping to the first image slot
*** Booting Zephyr OS build v4.0.0-735-geb4f2fb44a0e ***
[00:00:00.000,366] <inf> littlefs: LittleFS version 2.9, disk version 2.1
[00:00:00.000,610] <inf> littlefs: FS at flash-controller@4001e000:0xf8000 is 8 0x1000-byte blocks with 512 cycle
[00:00:00.000,640] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.000,732] <err> littlefs: ../modules/fs/littlefs/lfs.c:1374: Corrupted dir pair at {0x0, 0x1}
[00:00:00.000,762] <wrn> littlefs: can't mount (LFS -84); formatting
[00:00:00.172,149] <inf> littlefs: /lfs1 mounted
[00:00:00.172,180] <inf> smp_sample: build time: Nov 24 2024 19:21:02
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: ` v4.0.0-735-geb4f2`

**Additional information:**
Introduced by commit df32576cf579755ff2b62f87891cbf4341c714e4

butok, 2024-11-28:
Hello.

It cannot be introduced by tag: commit https://github.com/zephyrproject-rtos/zephyr/commit/df32576cf579755ff2b62f87891cbf4341c714e4

So reassigning it to Jamie.

butok, 2024-11-28:
@katgiadla 
Could you provide expected logs (without errors)?

katgiadla, 2024-11-28:
> @katgiadla Could you provide expected logs (without errors)?

I don't know because due to `harness: bluetooth` test has not been executed.

butok, 2024-11-28:
> > @katgiadla Could you provide expected logs (without errors)?
> 
> I don't know because due to `harness: bluetooth` test has not been executed.

Hm. Shouldn't Bluetooth only be used in BT specific test cases?
What about seral UART, USB and UDP?

PR #82314 (2024-12-03): samples: mgmt: mcumgr: smp_svr: Make build only
This is a sample, not a test, it should not run on devices and expect output with twister

Fixes #82251
closed 2024-12-03 by kartben