zephyrproject-rtos/zephyr #82265: Pins are left floating even with `GPIO_PULL_UP` on NXP RW612 (`rd_rw612_bga`)

nguterresn, 2024-11-28:
**Describe the bug**

As the title states, the pins are left floating even when defined to be pulled up while using the dt `rd_rw612_bga`. I've also measured with a multimeter.

**To Reproduce**

Here is a part of the `.overlay` file I'm using:

```dts
      chosen {
			// a few other things...
		      yellow_button = &button0;
		      red_button = &button1;
		      encoder0 = &encoder0;
		      encoder1 = &encoder1;
          };

	buttons {
		compatible = "gpio-keys";
		label = "Group of buttons";

		button0: button_0 {
			gpios = <&hsgpio0 10 GPIO_PULL_UP>; // D10
			zephyr,code = <INPUT_KEY_0>;
		};
		button1: button_1 {
			gpios = <&hsgpio0 2 GPIO_PULL_UP>; // D11 Does not work!
			zephyr,code = <INPUT_KEY_1>;
		};
		encoder0: encoder_0 {
			gpios = <&hsgpio0 17 GPIO_PULL_UP>; // A5
			zephyr,code = <INPUT_KEY_2>;
		};
		encoder1: encoder_1 {
			gpios = <&hsgpio0 16 GPIO_PULL_UP>; // A4
			zephyr,code = <INPUT_KEY_3>;
		};
	};
```

**Expected behavior**

Pins should be set with state HIGH.

**Impact**

I can't use all the pins and it slows down development.

**Logs and console output**

**Environment (please complete the following information):**

West version: 1.3.0
Zephyr SDK: 0.16.8

**Additional context**

rruuaanng, 2024-11-30:
Can you provide the full context? I would like to see the definition of `hsgpio0`

nguterresn, 2024-11-30:
[Here](https://github.dev/zephyrproject-rtos/zephyr/tree/main/boards/nxp/rd_rw612_bga).

```dts
hsgpio: hsgpio@100000 {
  compatible = "nxp,lpc-gpio";
  reg = <0x100000 0x4000>;
  #address-cells = <1>;
  #size-cells = <0>;

  hsgpio0: gpio@0 {
	  compatible = "nxp,lpc-gpio-port";
	  gpio-controller;
	  #gpio-cells = <2>;
	  reg = <0>;
	  int-source = "pint";
  };
  
  hsgpio1: gpio@1 {
	  compatible = "nxp,lpc-gpio-port";
	  gpio-controller;
	  #gpio-cells = <2>;
	  reg = <1>;
	  int-source = "pint";
  };
};
```

The context is: 

I'm trying to use a simple button. The button should be set internally as a pullup so that the other end is connected to GND. Upon press, the mcu triggers an interrupt ona failling edge.
I can't seem to set the internal GPIO pull-up resistor, it doesn't work.

mmahadevan108, 2024-12-05:
@nguterresn , can you help submit a PR with your proposed change.

nguterresn, 2024-12-11:
@mmahadevan108 I'm not sure how can I fix this. The proposed change is the fix itself — a pulled-up GPIO shouldn't be floating.

mmahadevan108, 2025-02-26:
@nguterresn , the bits in the register are set as expected when the `GPIO_PULL_UP` and `GPIO_PULL_DOWN` flags are specified via device tree. 
Could it be that something in your application is driving a value on the GPIO pins?

mmahadevan108, 2025-02-28:
I have also confirmed via measuring the GPIO pin that the signal toggles HIGH and LOW when I write to the PULL_UP and PULL_DOWN bits.

closed 2025-02-28 by mmahadevan108