zephyrproject-rtos/zephyr #82276: Nominate mysefl as contributor

mrodgers-witekio, 2024-11-28:
# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Matt Rodgers
1. GitHub username: mrodgers_witekio
1. Organization: Witekio

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pull/80395
https://github.com/zephyrproject-rtos/zephyr/pull/77982
https://github.com/zephyrproject-rtos/zephyr/pull/77125
https://github.com/zephyrproject-rtos/zephyr/pull/77003
https://github.com/zephyrproject-rtos/zephyr/pull/76850

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

jukkar, 2024-11-29:
@mrodgers-witekio I think you need to send a PR that adds yourself to desired places in MAINTAINERS file

mrodgers-witekio, 2024-11-29:
Based on this: https://docs.zephyrproject.org/latest/project/project_roles.html I think the MAINTAINERS file just needs to be changed for the collaborator or maintainer roles, rather than contributor.

jukkar, 2024-11-29:
Indeed, my bad. I mis read it as collaborator, probably need to get new glasses 😄

kartben, 2024-11-29:
> Indeed, my bad. I mis read it as collaborator, probably need to get new glasses 😄 

Or simply a sign that you'd apparently happily welcome a new collaborator :) hint hint @mrodgers-witekio

closed 2024-11-30 by nashif