zephyrproject-rtos/zephyr #82281: linker: ld: toolchain_linker_finalize duplicates <LINKER_FLAGS> leading to link issues in some situations

axelnxp, 2024-11-28:
**Describe the bug**

`toolchain_linker_finalize` was introduced by PR #78320 and according to its implementation for LD, it duplicates `<LINKER_FLAGS>`.
This is not a huge problem for common flags, but we got an issue when using `--whole-archive xxx --no-whole-archive`.
When this flag is duplicated, the linker exits with a duplicate definition error.
We've faced this issue in the connectedhomeip environment when migrating to 4.0 release.

cc @tejlmand 
 
**To Reproduce**
Use the following cmake instruction:
```
target_link_options(${target} INTERFACE -Wl,--whole-archive <some static lib> -Wl,--no-whole-archive
```
This flag will be duplicated in the link command, leading to duplicate definitions.

**Expected behavior**
Link flags added with `target_link_options` should be unique.

**Impact**
We can currently workaround by bypassing the `target_link_options` and manually link the symbols we want, but this is not acceptable in the long run.

**Logs and console output**
I printed the content of `CMAKE_CXX_LINK_EXECUTABLE` before and after `toolchain_link_options` is called:
```
-- CMAKE_CXX_LINK_EXECUTABLE: <CMAKE_CXX_COMPILER> <FLAGS> <CMAKE_CXX_LINK_FLAGS> <LINK_FLAGS> <OBJECTS> -o <TARGET> <LINK_LIBRARIES>
-- CMAKE_CXX_LINK_EXECUTABLE: <CMAKE_CXX_COMPILER> <FLAGS> <CMAKE_CXX_LINK_FLAGS> <LINK_FLAGS> <LINK_FLAGS> <OBJECTS> -o <TARGET> <LINK_LIBRARIES> -L"/opt/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v8-m.main+fp/hard" -lstdc++ -lm -lc -lgcc -lc
```
We can see `<LINK_FLAGS>` is duplicated by the function

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: `2e3873adde2386e0bc9acdbcdc8d389fdf4b9aee`

**Additional context**

Proposed patch:
```
diff --git a/cmake/linker/ld/target.cmake b/cmake/linker/ld/target.cmake
index 5e0a117c014..f9cdc938f54 100644
--- a/cmake/linker/ld/target.cmake
+++ b/cmake/linker/ld/target.cmake
@@ -148,7 +148,7 @@ macro(toolchain_linker_finalize)
   string(REPLACE ";" " " zephyr_std_libs "${zephyr_std_libs}")
 
   set(link_libraries "<LINK_FLAGS> <OBJECTS> -o <TARGET> <LINK_LIBRARIES> ${zephyr_std_libs}")
-  set(common_link "<LINK_FLAGS> ${link_libraries}")
+  set(common_link "${link_libraries}")
 
   set(CMAKE_ASM_LINK_EXECUTABLE "<CMAKE_ASM_COMPILER> <FLAGS> <CMAKE_ASM_LINK_FLAGS> ${common_link}")
   set(CMAKE_C_LINK_EXECUTABLE   "<CMAKE_C_COMPILER> <FLAGS> <CMAKE_C_LINK_FLAGS> ${common_link}")
```

axelnxp, 2024-11-28:
This part of the function is probably impacted too:
```
 set(cpp_link "<LINK_FLAGS> ${LIBGCC_DIR}/crtbegin.o ${link_libraries} ${LIBGCC_DIR}/crtend.o")
```

tejlmand, 2024-11-28:
> When this flag is duplicated, the linker exits with a duplicate definition error.

This flag is valid to duplicate.
From ld docs:
> > --whole-archive
> >
> > For each archive mentioned on the command line after the --whole-archive option, include every object file in the archive in the link, rather than searching the archive for the required object files. This is normally used to turn an archive file into a shared library, forcing every object to be included in the resulting shared library. **This option may be used more than once.** 

and you can try for yourself:
```
$ ccache /opt/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -gdwarf-4 -gdwarf-4 zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj -o zephyr/zephyr.elf zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj -T zephyr/linker.cmd -Wl,-Map=/projects/github/ncs/zephyr/build/zephyr/zephyr_final.map -Wl,--whole-archive app/libapp.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/libzephyr.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/arch/common/libarch__common.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/arch/arch/arm/core/libarch__arm__core.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/lib/libc/picolibc/liblib__libc__picolibc.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/lib/libc/common/liblib__libc__common.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/soc/soc/nrf52840/libsoc__nordic.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/drivers/clock_control/libdrivers__clock_control.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/drivers/console/libdrivers__console.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/drivers/gpio/libdrivers__gpio.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/drivers/pinctrl/libdrivers__pinctrl.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive -Wl,--whole-archive zephyr/drivers/timer/libdrivers__timer.a -Wl,--no-whole-archive -Wl,--whole-archive -Wl,--whole-archive modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a -Wl,--no-whole-archive -Wl,--whole-archive modules/segger/libmodules__segger.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a -L/projects/github/ncs/zephyr/build/zephyr -Wl,--print-memory-usage zephyr/arch/common/libisr_tables.a -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft -fuse-ld=bfd -Wl,--gc-sections -Wl,--build-id=none -Wl,--sort-common=descending -Wl,--sort-section=alignment -Wl,-u,_OffsetAbsSyms -Wl,-u,_ConfigAbsSyms -nostdlib -static -Wl,-X -Wl,-N -Wl,--orphan-handling=warn -Wl,-no-pie -specs=picolibc.specs -DPICOLIBC_LONG_LONG_PRINTF_SCANF -L"/opt/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp" -lc -lgcc
Memory region         Used Size  Region Size  %age Used
           FLASH:       18684 B         1 MB      1.78%
             RAM:        5696 B       256 KB      2.17%
        IDT_LIST:          0 GB        32 KB      0.00%
```
Use of multiple `--whole-archive` / `--no-whole-archive` works as expected.

So something else must be happening in this case.

tejlmand, 2024-11-28:
> Use the following cmake instruction:
> 
> ```
> target_link_options(${target} INTERFACE -Wl,--whole-archive <some static lib> -Wl,--no-whole-archive
> ```
> This flag will be duplicated in the link command, leading to duplicate definitions.

Perhaps your code was saved by CMake's deduplication feature in the past where the whole-archive flags probably was being de-duplicated because they were also added by Zephyr link line.

tejlmand, 2024-11-28:
> I printed the content of CMAKE_CXX_LINK_EXECUTABLE before and after toolchain_link_options is called:
> ....
> We can see <LINK_FLAGS> is duplicated by the function

That does indeed look suspicious.

axelnxp, 2024-11-29:
> This flag is valid to duplicate. From ld docs:
>
>
> and you can try for yourself:
>
> ```
> $ ccache /opt/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -gdwarf-4 -gdwarf-4 [...]

Sorry this was not very clear in my description, what I meant is the issue occurs if you duplicate the flag for the same library, like:
```
--whole-archive libfoo.a --no-whole-archive --whole-archive libfoo.a --no-whole-archive
```
In your example, I don't think that's the case from what I've seen.

tejlmand, 2024-11-29:
> In your example, I don't think that's the case from what I've seen.

Correct.

commit 84131133 (2024-12-02): cmake: linker: ld: fix duplicate LINK_FLAGS

Original implementation of `toolchain_linker_finalize` duplicates
the `LINK_FLAGS` in the link command. This can cause some problems
like duplicate definitions when using link options like
`--whole-archive`.

This commit fixes it by removing the duplicate `LINK_FLAGS`.

Fixes #82281
closed 2024-12-02 by fabiobaltieri