zephyrproject-rtos/zephyr #82323: Bluetooth: Mesh: Revert removal of tests with TinyCrypt

PavelVPV, 2024-11-29:
## Introduction

After https://github.com/zephyrproject-rtos/zephyr/pull/79931 PR the Bluetooth mesh option with TinyCrypt crypto library `CONFIG_BT_MESH_USES_TINYCRYPT` has been deprecated. Due to the Zephyr deprecation procedure all tests (unit and BabbleSim) that use this option were switched from TinyCrypt to PSA Crypto API with mbedTLS library (`CONFIG_BT_MESH_USES_MBEDTLS_PSA`).

### Problem description

Managing the encryption keys in the Bluetooth mesh with TinyCrypt and PSA Crypto API is different. The TinyCrypt keys are kept in RAM as a plain text, while PSA Crypto API provides key ids and the mesh stack doesn't have direct access to the keys. This also applies to the way the keys are stored in a non-volatile memory. When a mesh device boots up, it restores TinyCrypt keys as plain text by reading them through the settings subsystem. With PSA Crypto API only key ids are restored by the mesh stack. This makes impossible to flash a new firmware with mesh stack built with PSA Crypto API support on top of mesh stack built with TinyCrypt without proper precautions. As a minimum viable solution, users must unprovision mesh devices before flashing the new firmware with PSA Crypto API support.

The `CONFIG_BT_MESH_USES_TINYCRYPT` option stays for 2 more releases, however we can't guarantee that even minum viable solution works since all tests were removed.

The key points to revert tests are:
- Users may still rely on the TinyCrypt-based option for legacy projects or ongoing deployments. Ensuring tests remain for TinyCrypt helps maintain backward compatibility for these users during the transition period. It gives confidence that the feature works as intended, even if it’s deprecated, until the complete removal occurs.
- This introduces a risk of regressions or untested code paths, when shared components of the mesh stack are modified.
- Testing both options within the same release ensures that users can verify the correctness of their transition processes (e.g., unprovisioning/reprovisioning devices). Without these tests, debugging issues related to mixed environments becomes much harder.
- Keeping tests enables better documentation, helping users understand how to migrate and verify their setups. This avoids potential confusion or misconfigurations during the transition.
- Keeping tests during the deprecation period ensures that the feature is not only present but also functional and reliable until its formal removal.

### Proposed change

We want to provide smooth and predictable transition for users which we can't guarantee without running tests with TinyCrypt. Therefore we would like to revert the removal of the Bluetooth mesh tests (unit + bsim) with TinyCrypt and keep them running until TinyCrypt is finally removed.

alxelax, 2024-11-29:
I'd suggest to restore only bsim tests. Unit tests checks special units those depend on crypto algorithms indirectly.

d3zd3z, 2024-12-04:
Summary from the TSC Discussion:
- The deprecation of tiny crypt requires no Zephyr code to use these APIs.
- The fundamental issue is that the BT mesh code change how keys are stored with the removal of Tinycrypt, which makes the deprecation of Tinycrypt challenging.

Proposed two solutions:
- Allow `CONFIG_BT_MESH_USES_TINYCRYPT` to be deprecated at the same time as `CONFIG_TINYCRYPT` to be deprecated. This would require what is suggested by this change, allowing the tests that remain. We need CI changes to allow tests to remain of deprecated code.
-  Remove `CONFIG_BT_MESH_USES_TINYCRYPT`, but add code to the BT mesh code that adds support for reading/importing the tinycrypt format keys when TINYCRYPT is not selected. This would allow existing users and devices to migrate to the non tinycrypt support.

PavelVPV, 2025-02-10:
@nashif , @d3zd3z , @carlescufi , we have created a PR to revert removed bsim tests with enabled TinyCrypt: https://github.com/zephyrproject-rtos/zephyr/pull/84237. Bsim tests CI doesn't require any workaround for deprecated warning. We'd like to get this merged for 4.1.0. Maintaining this PR takes our time and increases risk of broken mesh with TC for 4.1.0. Mesh with TC support is still present as an option: https://github.com/zephyrproject-rtos/zephyr/blob/92dd07e51b9ef6c8fc515913c12e920e30d84981/subsys/bluetooth/mesh/Kconfig#L1490-L1491

closed 2025-02-17 by kartben