zephyrproject-rtos/zephyr #82331: Twister fails to flash on STM32 board when 2 development boards are connected to the system

RadhikaMaheshwari-eaton, 2024-11-29:
**BUG DESCRIPTION**
When we connect two STM32 controllers, specifically the STM32 Nucleo_H743ZI and the STM32 Nucleo_U575ZI_Q, the Twister script fails to flash the device whose COM port is specified in the command. Instead, it flashes the other board connected to the system, resulting in a flash error.

For example, if we execute the command:
python twister --device-testing --device-serial COM18 -T C:\edge-rtos-babelfish-v3\examples\services\Alarm_example -p nucleo_h743zi

the script attempts to flash the STM32 Nucleo_U575ZI_Q board on COM port 6, causing a flash error.

[image]

Additionally, the `device.log` file generated after the build and flash process shows the ST-firmware version for the STM32 Nucleo_U575ZI_Q board.

NUCLEOU575ZIQ - ST firmware version V3J10M3
NUCLEOH743ZI - ST firmware version V3J5M2

device.log 
```
Open On-Chip Debugger 0.12.0 (2023-01-14-23:37)

Licensed under GNU GPL v2

For bug reports, read

http://openocd.org/doc/doxygen/bugs.html

Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD

init

Info : clock speed 1800 kHz

Info : STLINK V3J10M3 (API v3) VID:PID 0483:374E

Info : Target voltage: 3.295808

Warn : UNEXPECTED idcode: 0x0be12477

Error: expected 1 of 1: 0x6ba02477

in procedure 'init'

FATAL ERROR: command exited with status 1: 'C:/Program Files/openocd-v0.12.0-i686-w64-mingw32/bin/openocd.exe' -s 'C:/blr_deps/zephyr-v3.6.0/zephyr/boards/arm/nucleo_h743zi\support' -s C:/blr_deps/zsdk/zephyr-sdk-0.16.5/sysroots/AMD64-pokysdk-linux/usr/share/openocd/scripts -f 'C:/blr_deps/zephyr-v3.6.0/zephyr/boards/arm/nucleo_h743zi\support\openocd.cfg' '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase C:/blr_deps/zephyr-v3.6.0/zephyr/scripts/twister-out/nucleo_h743zi/examples.services.Alarm_example/zephyr/zephyr.hex' -c 'reset halt' -c 'reset run' -c shutdown
```

We are using Zephyr 3.6.0 with Zephyr SDK version 0.16.5 on a Windows 10 system.

We have also attempted to upgrade the ST firmware versions for both the H7 and U5 boards, but this did not resolve the issue.

The expected behavior is that the Twister script should build and flash the device specified by the COM port in the command, regardless of any other ST devices connected to the system.

Attached twister log as well.
[twister.log](https://github.com/user-attachments/files/17959686/twister.log)

erwango, 2024-12-02:
@RadhikaMaheshwari-eaton To use twister on several boards, you need to use hardware-map feature.
See https://docs.zephyrproject.org/latest/develop/test/twister.html#executing-tests-on-multiple-devices.

Using this, we're running tests with twister on 20 boards in parallel. Note that we're using a Linux. I expect this would work on Windows as well.

Please have a try and report back.

RadhikaMaheshwari-eaton, 2024-12-03:
@erwango @gautierg-st If we use the hardware-map feature file, it works with two STM32 development boards connected. But is it mandatory to use this hardware-map.yml file?

What happens if we don't use the hardware-map feature file and instead specify the serial COM port in the command as mentioned in the document? What does that imply? 
```
python .\scripts\twister --device-testing --device-serial COM1 \
--device-serial-baud 115200 -p frdm_k64f  -T tests/kernel
```
[image]

Could you also provide some suggestions on how we can deploy this in a DevOps pipeline? Changing the hardware (i.e., development board) each time requires updating the hardware-map file, which isn't very practical.

erwango, 2024-12-03:
> What happens if we don't use the hardware-map feature file and instead specify the serial COM port in the command as mentioned in the document? What does that imply?

This has to be discussed with twister maintainer(s).

> Could you also provide some suggestions on how we can deploy this in a DevOps pipeline? Changing the hardware (i.e., development board) each time requires updating the hardware-map file, which isn't very practical.

On Linux, when generating hardware-map, you could use `--persistent-hardware-map` feature which will store device reference using immutable devid so that you only need to fill-in the hw-map once with boards details and it will be known even after unplug/plug.

This being said all Window related limitations or ways to set up a DevOps pipeline are not related to STM32. I'd suggest to get it in touch with relevant people through Discord.

erwango, 2024-12-03:
@RadhikaMaheshwari-eaton I've re-assigned the ticket for clarity and treatment efficientness. I'd propose also to update the issue title for clarity. Or even close this one and open a new one.

RadhikaMaheshwari-eaton, 2024-12-03:
Hi @erwango @gautierg-st I've observed that the hardware-map file works with one combination of STM32 H7 and U5, but it causes a flash error with another combination. In the problematic setup, the ST firmware for both H7 and U5 is V3J10M3.
[image]

This is the command which I am using `python twister --device-testing --hardware-map map.yml -T C:\edge-rtos-babelfish-v3-repo\edge-rtos-babelfish-v3\examples\services\Alarm_example -c -p nucleo_h743zi `
I only intend to flash the STM32 H7, not the U5, even though the U5 is connected to my system. This shouldn't cause any issues, right? Here is the hardware-map file.
```
- connected: true
  id: 004200224741500720383733
  platform: nucleo_u575zi_q
  product: ST-link
  runner: openocd
  serial: COM6
- connected: true
  id: 0024001E3532511131333430
  platform: nucleo_h743zi
  product: ST-link
  runner: openocd
  serial: COM18
```

This is the device.log file and twister.log file.
[twister.log](https://github.com/user-attachments/files/17992466/twister.log)
[device.log](https://github.com/user-attachments/files/17992469/device.log)
I'd also like to learn how to use west flash for Twister, as mentioned in the help section.
```
--west-flash [WEST_FLASH]
                        Uses west instead of ninja or make to flash when running with --device-testing. Supports comma-separated argument list. E.g "twister --device-testing --device-serial /dev/ttyACM0 --west-flash="--board-id=foobar,
                        --erase" will translate to "west flash -- --board-id=foobar --erase" NOTE: device-testing must be enabled to use this option.
```
What does the board ID mean, and how can we obtain it? I've tried using the ID from the generated map file. For example:

`python twister --device-testing --device-serial COM18 --west-flash="--board-id=0024001E3532511131333430, --erase" -T C:\edge-rtos-babelfish-v3-repo\edge-rtos-babelfish-v3\examples\services\Alarm_example -p nucleo_h743zi`

However, we receive an error log stating
 `FATAL ERROR: runner openocd received unknown arguments: ['--board-id=0024001E3532511131333430', ' --erase']`

RadhikaMaheshwari-eaton, 2024-12-04:
Hi @erwango @gautierg-st @nashif Are there any updates on the issue mentioned earlier?

RadhikaMaheshwari-eaton, 2024-12-05:
Upon further debugging, I discovered that the issue might be related to OpenOCD. Specifically, if we include the serial number of the STM32H7 board in the openocd.cfg file, then build and flash the Twister, it correctly flashes to the intended device.
[image]
This raises the question of how we can specify the board's serial number in the Twister command without modifying the openocd.cfg file. If you have any suggestions, please share them here. @erwango @nashif @gautierg-st

erwango, 2024-12-05:
When using Linux, `--board-id` argument is not needed for this use case.

If you really need `--board-id`, then you'll need to implement support in openocd runner, as it is not available yet.
From what I can see `--board-id` is only supported for pyocd and uf2 runners.
Alternatively, use pyocd runner.

RadhikaMaheshwari-eaton, 2024-12-05:
@erwango However, the STM32 H7 series does not support the PyOCD runner.
```
←[92m-- west flash: using runner pyocd
←[0m←[91mFATAL ERROR: board nucleo_h743zi does not support runner pyocd
←[0mTo fix, configure this runner in C:\blr_deps\zephyr-v3.6.0\zephyr\boards\arm\nucleo_h743zi\board.cmake and rebuild.
```
In the board.cmake file, only three runners are specified: OpenOCD, J-Link, and STM32CubeProgrammer.
[image]

How can we now incorporate the serial number of the STM32 board into the Twister command for a Windows system?

FRASTM, 2024-12-06:
> @erwango However, the STM32 H7 series does not support the PyOCD runner.
> 
with `pyocd pack install stm32h7`
the "west flash: using runner pyocd" can flash the nucleo_h743zi when board.cmake has 
```
board_runner_args(pyocd "--target=stm32h743zitx")
include(${ZEPHYR_BASE}/boards/common/pyocd.board.cmake)
```

RadhikaMaheshwari-eaton, 2024-12-09:
@erwango @FRASTM I followed the steps mentioned above, but unfortunately, none of them worked.
Build command: 
```
C:\blr_deps\zephyr-v3.6.0\zephyr\scripts>
python twister --device-testing --device-serial COM18 --west-flash="--board-id=0024001E3532511131333430" --west-runner pyocd -T C:\edge-rtos-babelfish-v3-repo\edge-rtos-babelfish-v3-work1\examples\services\Alarm_example -p nucleo_h743zi
```

board.cmake
[image]

Device.log 
[image]

It appears that the pyocd runner or twister command does not accept the --board-id arguments directly or with --west-flash on a Windows system. 
**Note**: I have two STM32 boards connected to my system: one STM32H7 and one STM32U5. However, I am trying to flash only the H7 board using the COM port and board-id.

RadhikaMaheshwari-eaton, 2024-12-17:
Hi @erwango @FRASTM 

Build Command:
`python twister --device-testing --hardware-map map.yml -T C:\edge-rtos-babelfish-v3-repo\edge-rtos-babelfish-v3\examples\services\Alarm_example -c -p nucleo_h743zi`

Hardware Map (map.yml):
```
- connected: true
  id: 0024001E3532511131333430
  platform: nucleo_h743zi
  product: ST-link
  runner: openocd
  serial: COM18
```
OR
```
- connected: true
  id: 004200224741500720383733
  platform: nucleo_u575zi_q
  product: ST-link
  runner: openocd
  serial: COM6

- connected: true
  id: 0024001E3532511131333430
  platform: nucleo_h743zi
  product: ST-link
  runner: openocd
  serial: COM18
```

Error Log:

```
Open On-Chip Debugger 0.12.0 (2023-01-14-23:37)

Licensed under GNU GPL v2

For bug reports, read

    http://openocd.org/doc/doxygen/bugs.html

Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD

init

Info : clock speed 1800 kHz

Info : STLINK V3J10M3 (API v3) VID:PID 0483:374E

Info : Target voltage: 3.293617

Warn : UNEXPECTED idcode: 0x0be12477

Error: expected 1 of 1: 0x6ba02477

in procedure 'init'

FATAL ERROR: command exited with status 1: 'C:/Program Files/openocd-v0.12.0-i686-w64-mingw32/bin/openocd.exe' -s 'C:/blr_deps/zephyr-v3.6.0/zephyr/boards/arm/nucleo_h743zi\support' -s C:/blr_deps/zsdk/zephyr-sdk-0.16.5/sysroots/AMD64-pokysdk-linux/usr/share/openocd/scripts -f 'C:/blr_deps/zephyr-v3.6.0/zephyr/boards/arm/nucleo_h743zi\support\openocd.cfg' '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase C:/blr_deps/zephyr-v3.6.0/zephyr/scripts/twister-out/nucleo_h743zi/examples.services.Alarm_example/zephyr/zephyr.hex' -c 'reset halt' -c 'reset run' -c shutdown
```

Explanatory Note:
During the build process for the Twister on the nucleo_h743zi platform, the following command was executed:
`python twister --device-testing --hardware-map map.yml -T C:\edge-rtos-babelfish-v3-repo\edge-rtos-babelfish-v3\examples\services\Alarm_example -c -p nucleo_h743zi`
The hardware map (map.yml) specifies the connected devices and their configurations. However, an error occurred during the initialization phase of the Open On-Chip Debugger (OpenOCD). The error log indicates an unexpected ID code (0x0be12477) was encountered, whereas the expected ID code was 0x6ba02477. This discrepancy caused the initialization procedure to fail, resulting in a fatal error.

Additional Observations:

1. I've observed that the hardware-map file works with one combination of STM32 H7 and U5, but it causes a flash error with another combination. In the problematic setup, the ST firmware for both H7 and U5 is V3J10M3.
2. I have two STM32 boards connected to my system: one STM32H7 and one STM32U5. However, I am trying to flash only the H7 board using the COM port and platform.

Given that this setup and command are supported and used by Zephyr, could you please provide guidance on resolving this ID code mismatch issue?

RadhikaMaheshwari-eaton, 2025-01-07:
Hi @erwango @FRASTM @gautierg-st Any updates on the above query?

RadhikaMaheshwari-eaton, 2025-02-18:
Hi @erwango @FRASTM @gautierg-st @nashif @kartben Any updates on the above query?

erwango, 2025-02-18:
@RadhikaMaheshwari-eaton Did you try to use STM32CubeProgrammer runner ?

```diff
\- connected: true
  id: 004200224741500720383733
  platform: nucleo_u575zi_q
  product: ST-link
-  runner: openocd
+ runner: stm32cubeprogrammer
  serial: COM6
```

RadhikaMaheshwari-eaton, 2025-02-18:
Have not tested with runner: stm32cubeprogrammer, will check and keep you posted.