zephyrproject-rtos/zephyr #82339: TELIT ME310G1 erroneous +CMUX command

rickou, 2024-11-29:
**Describe the bug**
ME310 get stuck on init after a failure at the AT+CMUX command step.

**To Reproduce**
Use a Telit ME310G1-WW modem with fw
AT#SWPKGV
37.00.217-P0C.210001
M0C.200006
P0C.210001
A0C.210000

**Expected behavior**
no error and continue initialisation in muxed mode

**Logs and console output**
```
[00:01:01.128,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGMR
[00:01:01.161,000] <dbg> modem_chat: modem_chat_log_received_command:  M0C.200006
[00:01:01.161,000] <dbg> modem_chat: modem_chat_script_next: telit_mex10g1_init_chat_script: step: 26
[00:01:01.161,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:01:01.161,000] <dbg> modem_chat: modem_chat_script_next: telit_mex10g1_init_chat_script: step: 27
[00:01:01.161,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CFUN=1
[00:01:01.838,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:01:01.838,000] <dbg> modem_chat: modem_chat_script_next: telit_mex10g1_init_chat_script: step: 28
[00:01:01.838,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CMUX=0,0,5,127,10,3,30,10,2
[00:01:01.865,000] <dbg> modem_chat: modem_chat_on_unknown_command_received: +CME ERROR: 4
[00:01:02.138,000] <dbg> modem_chat: modem_chat_script_stop: telit_mex10g1_init_chat_script: complete
[00:01:02.138,000] <dbg> modem_cellular: modem_cellular_log_event: event script success
[00:01:02.138,000] <dbg> modem_cellular: modem_cellular_log_event: event bus closed
[00:01:02.138,000] <dbg> modem_cellular: modem_cellular_log_state_changed: switch from run init script to connect cmux
[00:01:02.238,000] <dbg> modem_cellular: modem_cellular_log_event: event timeout
[00:01:02.238,000] <dbg> modem_cellular: modem_cellular_log_event: event bus opened
```

**Additional context**
refer to #78890 

The Telit document 80617ST10991A Rev. 15 specify only 
AT+CMUX=<mode>[,<subset>[,<port_speed>[,<N1>]]]

Solution is to shorten the command to 4 parameters.
(i don't know how to push the fix ??)

jeffwelder-ellenbytech, 2024-12-03:
My experience on the ME910G1-WW was the CMUX command takes a different amount of parameters based on firmware version of the modem.

Afaik it's undocumented by Telit.
If a change is made in this area it might be wise to check the solution on multiple firmwares.

rickou, 2024-12-03:
You're right, but i received an answer from Telit this week:
> In the ME310G1 family, the +CMUX command doesn't support the extra fields "T1","N2","T2" and "T3":

Checking the r13, the release used by [Erlkoenig90](https://github.com/Erlkoenig90) in #78890, the extra parameters are the default ones.
So, no prolems to simply remove the extra parameters.

In all cases, we need to do something or new modem FW do not works anymore.

closed 2025-02-16 by github-actions[bot]