zephyrproject-rtos/zephyr #82361: Add `pending` label to issues and PR

rruuaanng, 2024-11-30:
**Is your enhancement proposal related to a problem? Please describe.**

As far as I know, there doesn't seem to be a label for pending issues or PRs, but I think it's worth adding it.

**Describe the solution you'd like**

I think its name should be **pending**, or a more distinctive **delay**

**Describe alternatives you've considered**

**Additional context**

henrikbrixandersen, 2024-11-30:
We use milestones for this.

rruuaanng, 2024-11-30:
But labels are more intuitive, aren't they?

rruuaanng, 2024-12-02:
@nashif @cfriedt @kartben  WDYT?

nashif, 2024-12-10:
I am not following what is this about and why we need a label? What problem will it solve exactly? Please provide more details and background.

rruuaanng, 2024-12-11:
Currently, we usually use milestones for delayed plans. However, I think milestones have a few downsides compared to using a new label:
1. They're not as clear. When we categorize issues, they’re often tied to a specific architecture or subsystem, which makes them stand out because of a clear label. But for plans that are being discussed and expected to be delayed, there isn't a very obvious label indicating that the issue is delayed. This makes it harder for others to see, so they might either engage with it or overlook it. (Honestly, I don't think milestones get much attention. More people use labels for classification rather than milestones.)
2. Use a label will make this behavior more visible.
3. Milestones don't really reflect the actual status of a task. They usually indicate that a task has been included in a specific plan, but they don't show whether it's been delayed, lacks certain materials, or is waiting for support from a specific maintainer. For example, if we need to introduce a new interrupt API, and it's included in the 1.0.0 milestone, but we’re missing comments from the interrupt subsystem maintainer, we might have to delay the task. This is where the `pending` label or (`delay` label) becomes really useful. It allows us to take advantage of both the milestone classification system and the label classification system.