zephyrproject-rtos/zephyr #82366: tests: input.api.thread.input_api : imx93_evk : build failure

hakehuang, 2024-11-30:
**Describe the bug**

<html>
<body>

CASE_NAME | ZEPHYR_VERSION | RESULT | PLATFORM | TEST_PATH
-- | -- | -- | -- | --
input.api.thread.input_api.report_apis | v4.0.0-1231-g6c7b38f5fa1c | Build failure | imx93_evk/mimx9352/a55 | input.api
input.api.thread.input_api.sequence_thread | v4.0.0-1231-g6c7b38f5fa1c | Build failure | imx93_evk/mimx9352/a55 | input.api
input.api.thread.input_api.synchronous | v4.0.0-1231-g6c7b38f5fa1c | Build failure | imx93_evk/mimx9352/a55 | input.api
input.api.synchronous.input_api.report_apis | v4.0.0-1231-g6c7b38f5fa1c | Build failure | imx93_evk/mimx9352/a55 | input.api
input.api.synchronous.input_api.sequence_thread | v4.0.0-1231-g6c7b38f5fa1c | Build failure | imx93_evk/mimx9352/a55 | input.api
input.api.synchronous.input_api.synchronous | v4.0.0-1231-g6c7b38f5fa1c | Build failure | imx93_evk/mimx9352/a55 | input.api

</body>
</html>

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - imx93_evk/mimx9352/a55 
 - imx93_evk/mimx9352/m33
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 
```
f812de20266a40f3bf685b2aa31e6264ab6d697b is the first bad commit
commit f812de20266a40f3bf685b2aa31e6264ab6d697b
Author: Gerard Marull-Paretas <gerard@teslabs.com>
Date:   Tue Nov 26 20:48:56 2024 +0100

    drivers: input: gpio_keys: select GPIO
    
    Instead of depending on GPIO. This allows to just enable
    CONFIG_INPUT=y + DT node in the application layer. The same
    pattern is nowadays followed by most drivers.
    
    Signed-off-by: Gerard Marull-Paretas <gerard@teslabs.com>

 drivers/input/Kconfig.gpio_keys | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)
```

**To Reproduce**

```
west twister -p imx93_evk/mimx9352/m33 -s tests/subsys/input/api/input.api.synchronous
```

**Expected behavior**

**Impact**

**Logs and console output**

```
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/toolchain/gcc.h:87:36: error: static assertion failed: "zephyr-code must be specified to use the input-gpio-keys driver"
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                    ^~~~~~~~~~~~~~
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.0.0-1231-g6c7b38f5fa1c

mmahadevan108, 2024-12-05:
@JiafeiPan, can you please help take a look at this build failure.

JiafeiPan, 2024-12-06:
PR #82586 will fix this issue.

closed 2024-12-09 by kartben