zephyrproject-rtos/zephyr #82370: modem: the UART exchange between the MCU and the modem is disrupted after rebooting the MCU or modem.

dmkuzmin, 2024-11-30:
**Describe the bug**

bjarki-andreasen, 2024-12-02:
What does your hw flowcontrol look like? silence like this could easily be RTS/DTS pin which is just floating, sometimes allowing flow, sometimes not :)

dmkuzmin, 2024-12-02:
> What does your hw flowcontrol look like? UART silence like this could easily be RTS/DTS pin which is just floating, sometimes allowing flow, sometimes not :) The random char issue I don't what's about though, maybe a UART driver quirk

I tried to turn on and off the hw-flow-control option in board dts.
Configuring UART in pinctrl:
```
        uart1_default: uart1_default {
		group1 {
			pinmux = <UART1_TX_GPIO6>,
				 <UART1_RTS_GPIO2>;
			bias-pull-up;
			drive-push-pull;
			output-enable;
			output-high;
		};
		group2 {
			pinmux = <UART1_RX_GPIO10>,
				 <UART1_CTS_GPIO1>;
			bias-pull-up;
			input-enable;
		};
	};
 
	uart1_sleep: uart1_sleep {
		group1 {
			pinmux = <UART1_TX_GPIO6>,
				 <UART1_RTS_GPIO2>;
			bias-pull-up;
			drive-push-pull;
			output-enable;
			output-high;
		};
		group2 {
			pinmux = <UART1_RX_GPIO10>,
				 <UART1_CTS_GPIO1>;
			bias-pull-up;
			input-enable;
		};
	};
```

bjarki-andreasen, 2024-12-02:
Just to eliminate potential problems, the modem uart backend is extensively used and tested, so I would expect it to be a UART device driver issue... Could you try running this test suite with your board? it requires jumping TX to RX, but it can show if it is an issue with UART: https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/subsys/modem/backends/uart

run them using:
`west build -p -b <board> <workspace>/zephyr/subsys/modem/backends/uart -T modem.backends.uart.async`
`west build -p -b <board> <workspace>/zephyr/subsys/modem/backends/uart -T modem.backends.uart.isr`

dmkuzmin, 2024-12-02:
I closed the RX and TX. I don't understand what happened here, but apparently, according to the FAIL labels, everything is bad :)

```
=================
ceive
TX: 2,2
TX ACC: 2
RX: 2
TX: 8,8
TX ACC: 10
RX: 8
TX: 32,32
TX ACC: 42
RX: 32
TX: 128,128
TX ACC: 170
RX: 128
TX: 512,512
TX ACC: 682
RX: 512
TX: 20W: Receive buffer overrun
48,1024
TX ACC: 1706

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/modem/backends/uart/src/main.c:202: modem_backend_uart_suite_test_transmit_receive: Received data is corrupted

 FAIL - test_transmit_receive in 3.201 seconds
===================================================================
START - test_transmit_receive
TX: 2,2
TX ACC: 2

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/modem/backends/uart/src/main.c:202: modem_backend_uart_suite_test_transmit_receive: Received data is corrupted

 FAIL - test_transmit_receive in 1.601 seconds
===================================================================
START - test_transmit_receive
TX: 2,2
TX ACC: 2

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/modem/backends/uart/src/main.c:202: modem_backend_uart_suite_test_transmit_receive: Received data is corrupted

 FAIL - test_transmit_receive in 1.601 seconds
===================================================================
TESTSUITE modem_backend_uart_suite failed.
Running TESTSUITE modem_backend_uart_suite
===================================================================
START - test_transmit_receive
TX: 2,2
TX ACC: 2

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/modem/backends/uart/src/main.c:202: modem_backend_uart_suite_test_transmit_receive: Received data is corrupted

 FAIL - test_transmit_receive in 1.601 seconds
===================================================================
START - test_transmit_receive
TX: 2,2
TX ACC: 2

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/modem/backends/uart/src/main.c:202: modem_backend_uart_suite_test_transmit_receive: Received data is corrupted

 FAIL - test_transmit_receive in 1.601 seconds
===================================================================
START - test_transmit_receive
TX: 2,2
TX ACC: 2

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/modem/backends/uart/src/main.c:202: modem_backend_uart_suite_test_transmit_receive: Received data is corrupted

 FAIL - test_transmit_receive in 1.601 seconds
[... 247 lines cut ...]
D6 57 44 2D 62 F3 B0 29 AE 4F DC E5 BA 6B C8 61  .WD-b..).O...k.a
86 47 74 9D 12 E3 E0 99 5E 3F 0C 55 6A 5B F8 D1  .Gt.....^?.Uj[..
36 37 A4 0D C2 D3 10 09 0E 2F 3C C5 1A 4B 28 41  67......./<..K(A
E6 27 D4 7D 72 C3 40 79 BE 1F 6C 35 CA 3B 58 B1  .'.}r.@y..l5.;X.
96 17 04 ED 22 B3 70 E9 6E 0F 9C A5 7A 2B 88 21  ....".p.n...z+.!
46 07 34 5D D2 A3 A0 59 1E FF CC 15 2A 1B B8 91  F.4]...Y....*...
F6 F7 64 CD 82 93 D0 C9 CE EF FC 85 DA 0B E8 01  ..d.............
A6 E7 94 3D 32 83 00 39 7E DF 2C F5 8A FB 18 71  ...=2..9~.,....q
56 D7 C4 AD E2 73 30 A9 2E CF 5C 65 3A EB 48 E1  V....s0...\e:.H.
06 C7 F4 1D 92 63 60 19 DE BF 8C D5 EA DB 78 51  .....c`.......xQ
B6 B7 24 8D 42 53 90 89 8E AF BC 45 9A CB A8 C1  ..$.BS.....E....
66 A7 54 FD F2 43 C0 F9 3E 9F EC B5 4A BB D8 31  f.T..C..>...J..1
16 97 84 6D A2 33 F0 69 EE 8F 1C 25 FA AB 08 A1  ...m.3.i...%....
C6 87 B4 DD 52 23 20 D9 9E 7F 4C 95 AA 9B 38 11  ....R# ...L...8.
76 77 E4 4D 02 13 50 49 4E 6F 7C 05 5A 8B 68 81  vw.M..PINo|.Z.h.
26 67 14 BD B2 03 80 B9 FE 5F AC 75 0A 7B 98     &g......._.u.{. 
A3 A0                                            ..              
A3 A0                                            ..              
A3 A0                                            ..              
A3 A0                                            ..              
A3 A0                                            ..              
A3 A0                                            ..              
A3 A0                                            ..              
A3 A0                                            ..              
```

dmkuzmin, 2024-12-03:
It seems to me that somewhere in modem_backend_uart_isr.c there is a missing call to uart_irq_tx_complete()

bjarki-andreasen, 2024-12-03:
`CONFIG_DMA=y` is a dependency of a specific UART implementation/HW, so the UART driver has to select it:
```
config FOO_UART
        select CONFIG_DMA=y if UART_ASYNC_API
```

> It seems to me that somewhere in modem_backend_uart_isr.c there is a missing call to uart_irq_tx_complete()

The test suite runs fine on STM32 and NRF hardware, so likely not the case, but maybe those drivers can do without? not sure :)

dmkuzmin, 2024-12-03:
UART2 also works stably after 10 reboots

dmkuzmin, 2024-12-04:
I have solved the problem with the reboot. Please add the TX FIFO reset to the ..\zephyr\drivers\serial\uart_esp32.c file after the RX FIFO reset (line 269).  As a reference file, you can use ..\modules\hal\espressif\components\driver\uart\uart.c file. The uart_param_config function has a TX FIFO reset.

Next, about the case when I put the modem to sleep (pm suspend mode) and wake it up (pm resume modem). There is a missing RX FIFO reset call in some place, as the controller accepts garbage and cannot react to the response.

Nothing has changed with the use of UART ASYNC so far

dmkuzmin, 2024-12-04:
New additional information about asynchronous exchange. Asynchronous exchange only worked on UART0. The exchange via DMA does not work on UART1 and UART2. I think it has something to do with the architecture of connecting UART controllers to DMA via a common UHCI0 node

bjarki-andreasen, 2024-12-04:
This seems to be more of a UART problem rather than modem problem, I have added labels as such to hopefully get some input from some ESP32 engineers :)

@sylvioalves Maybe you can take a look? seems to be some instability in `uart_esp32.c` maybe :)

sylvioalves, 2024-12-13:
Fix in #82967

dmkuzmin, 2025-02-12:
Unfortunately, the problem with entering and exiting modem sleep mode has not been solved, where the controller takes garbage upon waking up and needs to reset the RX FIFO, as it seems to me, but I cannot guess where to insert it.

Secondly, there is a problem with asynchronous mode operation. DMA is enabled only when using UART0

Or is it necessary to create a new issue for these problems?