zephyrproject-rtos/zephyr #82384: Reduce ROM size of USB_EPStartXfer

KozhinovAlexander, 2024-12-01:
**Is your feature request related to a problem? Please describe.**
`USB_EPStartXfer` for stm32g4 has 1440 KiBi text section, which is huge in my opinion. 

**Describe the solution you'd like**
The size of `USB_EPStartXfer`  shall be reduced or the reason for such text section size shall be mentioned.

**Describe alternatives you've considered**
nonr

**Additional context**
To reproduce the behaviour any USB sample for stm32g4 could be taken. Afterwards execute `ninja -C build rom_report > rom_report.log`.

In my case I am getting following values:

```
ninja: Entering directory `app/build'
[0/1] cd $HOME/Documents/dev/cannectivity/app/build &&  $HOME/Documents/dev/zephyrproject/.zephyr_venv/bin/python3  $HOME/Documents/dev/zephyrproject/zephyr/scripts/footprint/size_report -k  $HOME/Documents/dev/cannectivity/app/build/zephyr/zephyr.elf -z $HOME/Documents/dev/zephyrproject/zephyr -o  $HOME/Documents/dev/cannectivity/app/build --workspace= $HOME/Documents/dev/zephyrproject -d 99 rom
WARN: Symbol 'z_tls_current' section '' is not in RAM or ROM.
Path                                                                                             Size       %  Address    Section    
==========================================================================================================================================
Root                                                                                            61146 100.00%  - 
├── (hidden)                                                                                     8088  13.23%  - 
├── (no paths)                                                                                  53056  86.77%  - 
│   ├── AHBPrescTable                                                                              16   0.03%  0x0800defd rodata
│   ├── APBPrescTable                                                                               8   0.01%  0x0800def5 rodata
│   ├── CSWTCH.490                                                                                 20   0.03%  0x0800bde8 rodata
│   ├── HAL_PCD_DataInStageCallback.constprop.0                                                    36   0.06%  0x0800b721 text
│   ├── HAL_PCD_DataOutStageCallback.constprop.0                                                   56   0.09%  0x08007579 text
│   ├── HAL_PCD_SuspendCallback.constprop.0                                                        20   0.03%  0x080075b1 text
│   ├── LL_RCC_HSI_IsReady                                                                         16   0.03%  0x08005ee5 text
│   ├── LL_SetFlashLatency                                                                        220   0.36%  0x08006031 text
│   ├── LL_TIM_ClearFlag_CC1                                                                        8   0.01%  0x0800a805 text
│   ├── LL_TIM_ClearFlag_CC2                                                                        8   0.01%  0x0800a80d text
│   ├── LL_TIM_ClearFlag_CC3                                                                        8   0.01%  0x0800a815 text
│   ├── LL_TIM_ClearFlag_CC4                                                                        8   0.01%  0x0800a81d text
│   ├── LL_TIM_DisableIT_CC1                                                                       10   0.02%  0x0800a82f text
│   ├── LL_TIM_DisableIT_CC2                                                                       10   0.02%  0x0800a843 text
│   ├── LL_TIM_DisableIT_CC3                                                                       10   0.02%  0x0800a857 text
│   ├── LL_TIM_DisableIT_CC4                                                                       10   0.02%  0x0800a86b text
│   ├── LL_TIM_EnableIT_CC1                                                                        10   0.02%  0x0800a825 text
│   ├── LL_TIM_EnableIT_CC2                                                                        10   0.02%  0x0800a839 text
│   ├── LL_TIM_EnableIT_CC3                                                                        10   0.02%  0x0800a84d text
│   ├── LL_TIM_EnableIT_CC4                                                                        10   0.02%  0x0800a861 text
│   ├── LL_TIM_OC_GetCompareCH1                                                                     4   0.01%  0x0800a7f5 text
│   ├── LL_TIM_OC_GetCompareCH2                                                                     4   0.01%  0x0800a7f9 text
│   ├── LL_TIM_OC_GetCompareCH3                                                                     4   0.01%  0x0800a7fd text
│   ├── LL_TIM_OC_GetCompareCH4                                                                     4   0.01%  0x0800a801 text
│   ├── LL_TIM_OC_SetCompareCH1                                                                     4   0.01%  0x0800a7e5 text
│   ├── LL_TIM_OC_SetCompareCH2                                                                     4   0.01%  0x0800a7e9 text
│   ├── LL_TIM_OC_SetCompareCH3                                                                     4   0.01%  0x0800a7ed text
│   ├── LL_TIM_OC_SetCompareCH4                                                                     4   0.01%  0x0800a7f1 text
│   ├── SEGGER_RTT_WriteSkipNoLock.constprop.0                                                    112   0.18%  0x0800760d text
│   ├── SystemCoreClock                                                                             4   0.01%  0x20000f1c datas
│   ├── USB_ActivateEndpoint                                                                      776   1.27%  0x08006cb5 text
│   ├── USB_EPStartXfer                                                                          1440   2.36%  0x0800ae4d text
```

KozhinovAlexander, 2024-12-01:
@erwango I am not sure if it could be considered as a bug, cause everything is working fine. Thus feel free to move in an appropriate category.

erwango, 2024-12-02:
@KozhinovAlexander I'd suggest to raise directly in https://github.com/STMicroelectronics/STM32CubeG4 and follow up from there.

KozhinovAlexander, 2024-12-02:
> @KozhinovAlexander I'd suggest to raise directly in https://github.com/STMicroelectronics/STM32CubeG4 and follow up from there.

Thanks I will move it there and link it here.

KozhinovAlexander, 2024-12-02:
> @KozhinovAlexander I'd suggest to raise directly in https://github.com/STMicroelectronics/STM32CubeG4 and follow up from there.

Moved to: https://github.com/STMicroelectronics/STM32CubeG4/issues/52
Closed since moved to better matching thread

closed 2024-12-02 by KozhinovAlexander