zephyrproject-rtos/zephyr #82391: Twister with device-testing cannot run tests on target on Windows environment

Dat-NguyenDuy, 2024-12-02:
**Describe the bug**

When trying to run twister with --device-testing on Windows, all tests being skipped.

Is this a regression? Yes: this PR https://github.com/zephyrproject-rtos/zephyr/pull/79174 causes the issue
Quick workaround: 
```
--- a/scripts/pylib/twister/twisterlib/testinstance.py
+++ b/scripts/pylib/twister/twisterlib/testinstance.py
@@ -263,11 +263,11 @@ class TestInstance:
         simulator = self.platform.simulator_by_name(simulation)
         if os.name == 'nt':
             # running on simulators is currently supported only for QEMU on Windows
-            if (not simulator) or simulator.name not in ('na', 'qemu'):
+            if simulator and simulator.name not in ('na', 'qemu'):
                 return False

             # check presence of QEMU on Windows
-            if simulator.name == 'qemu' and 'QEMU_BIN_PATH' not in os.environ:
+            if simulator and simulator.name == 'qemu' and 'QEMU_BIN_PATH' not in os.environ:
                 return False
```

**To Reproduce**
Run this command:
`west twister -p frdm_k64f --short-build-path -M -N --device-testing --device-serial-baud 115200 --device-serial COM33 -O D:\o -T .\samples\hello_world\
`

**Expected behavior**
Twister with device-testing can run tests on target on Windows environment

**Impact**
Twister with device-testing **cannot** run tests on target on Windows environment

**Logs and console output**
```
INFO    - Using Ninja..
INFO    - Zephyr version: v4.0.0-1261-g0a80c780cb65
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report D:\o\testplan.json

Device testing on:

| Platform   | ID   | Serial device   |
|------------|------|-----------------|

Junction created for D:\o\twister_links\test_0 <<===>> D:\o\frdm_k64f_mk64f12\samples\hello_world\sample.basic.helloworld
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue

INFO    - 1 test scenarios (1 configurations) selected, 1 configurations filtered (1 by static filter, 0 at runtime).
INFO    - 0 of 0 executed test configurations passed (0.00%), 0 built (not run), 0 failed, 0 errored, with no warnings in 7.10 seconds.
INFO    - 0 of 0 executed test cases passed (0.00%) on 0 out of total 871 platforms (0.00%).
INFO    - 0 test configurations executed on platforms, 0 test configurations were only built.
```
**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 66df541d6c8d2e192730b15bfcd29be01332f40e

**Additional context**

nashif, 2024-12-03:
#82020 could also be related.

ithinuel, 2024-12-03:
Oh, indeed. @Dat-NguyenDuy would you mind creating a PR with your fix please ?

closed 2024-12-04 by nashif