zephyrproject-rtos/zephyr #82395: stm32wb_ble did not working

athulksriod, 2024-12-02:
zephyr version :3.7.0  

sample : zephyr\boards\stm32\power_mgmt\stm32wb_ble

board : nucleo-wb55

Hi all , this is i got when running the stm32wb_ble sample code #zephyr #stm32wb55 #ble
*** BStarting Beacon Demo

[00:00:03.127,000] �[1;31m os: ***** USAGE FAULT *****�[0m

[00:00:03.127,000] �[1;31m os: Illegal use of the EPSR�[0m

[00:00:03.127,000] �[1;31m os: r0/a1: 0x0800c69c r1/a2: 0x20003108 r2/a
3: 0x00080000�[0m

[00:00:03.127,000] �[1;31m os: r3/a4: 0x00000000 r12/ip: 0x200002e0 r14/l
r: 0x08005b21�[0m

[00:00:03.127,000] �[1;31m os: xpsr: 0x20000000�[0m

[00:00:03.127,000] �[1;31m os: Faulting instruction address (r15/pc): 0x000
00000�[0m

[00:00:03.127,000] �[1;31m os: >>> ZEPHYR FATAL ERROR 35: Unknown error on
CPU 0�[0m

[00:00:03.127,000] �[1;31m os: Current thread: 0x20000a80 (unknown)�[0m

[00:00:03.186,000] �[1;31m os: Halting system�[0m

i update the m0 core with stm32wb5x_BLE_HCILayer_fw V1.15.0 but the code getting stuck in FUNC_NORETURN __weak void arch_system_halt(unsigned int reason) function the reason is 35

/* LCOV_EXCL_START */
FUNC_NORETURN __weak void arch_system_halt(unsigned int reason)
{
ARG_UNUSED(reason);

/* TODO: What's the best way to totally halt the system if SMP
 * is enabled?
 */

(void)arch_irq_lock();
for (;;) {
	/* Spin endlessly */
}
}

in this infinite for loop .

@erwango

erwango, 2024-12-02:
@athulksriod Why raising 2 issues ?
Also, please use the template and provide information required to provide support.

erwango, 2024-12-02:
Set under `waiting for response` until bug template is used and correctly filled. 
(Reason is we can't start analysis before we're sure to have all required information in order to avoid losing time).

erwango, 2024-12-03:
@athulksriod Please use v3.7.1, there is a fix on for STM32WB BLE.

athulksriod, 2024-12-04:
ok i got it thanks

closed 2024-12-04 by athulksriod