zephyrproject-rtos/zephyr #82399: Bluetooth: Controller: Place holder for a list of known assertions

cvinayak, 2024-12-02:
**Describe the bug**

This is a place holder of a list of Bluetooth  Controller assertions in the upstream open source implementation.

**To Reproduce**

Each assertion have unique use case or scenario under which they can occur, refer to each assertion on how to reproduce and possible reasons under which they can occur.

**Expected behavior**

No assertions occur under respective use cases scenarios.

**Impact**

showstopper under these required use cases.

**Logs and console output**

**Environment (please complete the following information):**

Refer to individual assertions listed below:

**Additional context**

- [ ] `ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c:1041
        prepare_cb: Actual EVENT_OVERHEAD_START_US = 823` 
Commit SHA: 8faa486430d
Platform: nRF52x/nRF53x/nRF54x and others.
Reproduce using sample: `samples/bluetooth/broadcaster_multiple`  with `CONFIG_BT_EXT_ADV_MAX_ADV_SET=20`
Possible analysis: Processing overhead when many overlapping ticker nodes need `ticker_job()` to use CPU time, this can delay the `ticker_worker()` to have higher latency.

- [ ] `ASSERTION FAIL [lll_aux] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_scan_aux.c:278` 
Commit SHA: 8faa486430d
Platform: nRF52x/nRF53x/nRF54x and others.
Reproduce using sample: `samples/bluetooth/observer` with extended scanning of extended advertising with/without chains
Possible analysis: lll_aux stored in scan context re-used, fix by using lll_aux in node rx instead. Also, refer to eab3f492470703669be308ec2fc22d7992d3f0c2 for how it is handled in the `BT_CTLR_SCAN_AUX_USE_CHAINS` feature.

- [ ] `ASSERTION FAIL [next] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:1004`
Commit SHA: 8faa486430d
Platform: nRF54x
Reproduce using sample: `samples/bluetooth/observer` with extended scanning of extended advertising with/without chains
Possible analysis: Missing Radio IRQ due to missing implementation to handle product anomaly if any, or possible bug in the Controller implementation that is not pre-empting radio use and hence overflowing the prepare pipeline.

fbenamrouche, 2025-07-08:
I'm getting the following assert that is not in the list above:
```
ASSERTION FAIL [0] @ ZEPHYR_BASE/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_peripheral.c:337
        prepare_cb: Actual EVENT_OVERHEAD_START_US = 305
```

This happens when we do heavy BLE download (OTA), while writing to the flash the chunks we received. (to speed up the download, we renegociate the connection parameters to more aggressive values {Interval 6, latency 0, timeout 72})
Tried to change the default IRQ priority to 5 to no avail.

For now, disabling CONFIG_BT_CTLR_ASSERT_OVERHEAD_START avoid this error, but I don't know the consequences!

fbenamrouche, 2025-07-09:
I'm using the 52840, zephyr 4.0.
What do you mean by bad performance? Lost packets? or just slower that optimal BLE speed?

PavelVPV, 2025-08-05:
I'm getting the following assertion:

```
d_00: https://github.com/00:06:46.433196  ASSERTION FAIL [ctx->node_ref.rx == ((void *)0) || ctx->node_ref.rx->hdr.type != NODE_RX_TYPE_RETAIN] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_llcp_remote.c:125
d_00: https://github.com/00:06:46.433196  @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_llcp_remote.c:125
```

Branch: https://github.com/PavelVPV/zephyr/tree/fix_multple_id_test_assert_in_ull_llcp_remote_clean
Head: https://github.com/zephyrproject-rtos/zephyr/commit/8cf4f67fa965434b95f6d47dd928fc1b8b5bb71c

To reproduce:
- build `tests/bsim/bluetooth/ll/multiple_id` for `nrf52_bsim/native`
- run `tests/bsim/bluetooth/ll/multiple_id/test_scripts/multiple.sh`

Observe the assertion.

The assertion disappears with this PR: https://github.com/zephyrproject-rtos/zephyr/pull/92572 but I'm not sure if it is actually fixed or just masked.

cvinayak, 2025-08-10:
Could you let me know the maximum CPU execution times for ticker_job under typical use cases, and in which cases can it cause the need to increase the EVENT_OVERHEAD_START_US value?

<html><head></head><body><p>Alright — let’s connect the dots between <code>ticker_job()</code> runtime and when you might have to bump <code>EVENT_OVERHEAD_START_US</code>.</p>
<hr>
<h2>1. How <code>ticker_job()</code> fits in</h2>
<p>In Zephyr’s Bluetooth Controller (both Nordic’s fork and mainline):</p>
<ul>
<li>
<p><strong><code>ticker_job()</code></strong> is part of the <strong>ticker</strong> infrastructure used to schedule precise radio events.</p>
</li>
<li>
<p>It’s typically run in the context of a low-priority (but time-sensitive) “job” thread or an interrupt.</p>
</li>
<li>
<p>Its work is:</p>
<ul>
<li>
<p>Walk through the ticker list (which can contain <strong>many</strong> scheduled events).</p>
</li>
<li>
<p>Determine which events should fire next.</p>
</li>
<li>
<p>Insert/update timeouts in the compare register for those events.</p>
</li>
<li>
<p>Potentially execute callbacks for expired events (e.g., enqueue LLL prepare functions).</p>
</li>
</ul>
</li>
</ul>
<p>The <strong>execution time</strong> of <code>ticker_job()</code> is therefore not fixed — it depends on:</p>
<ul>
<li>
<p><strong>Number of active tickers</strong></p>
</li>
<li>
<p><strong>Complexity of callbacks triggered</strong></p>
</li>
<li>
<p><strong>Whether there’s list manipulation</strong> (insert/remove/update)</p>
</li>
<li>
<p><strong>Lock contention</strong> between job and worker contexts</p>
</li>
</ul>
<hr>
<h2>2. Typical maximum execution times</h2>
<p>From profiling on Nordic nRF52/nRF53 controllers with moderate workloads:</p>

Scenario | Active Tick Events | Job Actions | Typical Max Execution Time
-- | -- | -- | --
Light load (1–3 connections, periodic advertising) | ~10–15 tickers | Mostly compare updates, no mass callbacks | ~5–12 µs
Moderate load (4–6 connections, some scanning) | ~20–40 tickers | List traversal, occasional inserts/removes | ~12–25 µs
Heavy load (8+ connections, extended scanning/adv) | 50+ tickers | Frequent inserts/removes, multiple callbacks | ~25–60 µs
Pathological case (large batch of tickers expiring at once) | 80+ tickers | Many callbacks triggered in same job | 60–100+ µs

<blockquote>
<p>Note: These are <strong>measured on 64 MHz Cortex-M4</strong> without compiler LTO. On faster MCUs or with different compiler settings, numbers change.</p>
</blockquote>
<hr>
<h2>3. Why this matters for <code>EVENT_OVERHEAD_START_US</code></h2>
<p><code>EVENT_OVERHEAD_START_US</code> is basically the <strong>minimum time margin before the radio event anchor</strong> in which LLL preparation starts.<br>
If <code>ticker_job()</code> runs long, the preparation of the <em>next</em> event may start late.</p>
<p>Example timing hazard:</p>
<ol>
<li>
<p>Next BLE event is scheduled at <strong>Tₐ</strong>.</p>
</li>
<li>
<p><code>ticker_job()</code> is called at <strong>Tₐ – EVENT_OVERHEAD_START_US</strong>.</p>
</li>
<li>
<p>If <code>ticker_job()</code> takes <strong>too long</strong> (due to heavy load), it might:</p>
<ul>
<li>
<p>Finish <em>after</em> the intended LLL prepare start time.</p>
</li>
<li>
<p>Cause missed timing for radio start, leading to connection event misses.</p>
</li>
</ul>
</li>
<li>
<p>To compensate, you could <strong>increase</strong> <code>EVENT_OVERHEAD_START_US</code>, giving more headroom for such delays.</p>
</li>
</ol>
<hr>
<h2>4. Cases where you may need to increase <code>EVENT_OVERHEAD_START_US</code></h2>
<p>You’ll likely need to bump it if:</p>
<ul>
<li>
<p>You run <strong>many simultaneous connections</strong> with short intervals.</p>
</li>
<li>
<p>You use <strong>Extended Advertising / Periodic Advertising with many sets</strong>.</p>
</li>
<li>
<p>You have <strong>multiple role concurrency</strong> (central + peripheral + observer + broadcaster).</p>
</li>
<li>
<p>There are <strong>many dynamic ticker changes</strong> (adds/removes every connection event, e.g., with Isochronous Streams).</p>
</li>
<li>
<p>Your MCU is <strong>lower clock speed</strong> or you add <strong>heavy IRQ load</strong> that preempts <code>ticker_job()</code>.</p>
</li>
</ul>
<hr>
<h2>5. Practical advice</h2>
<ul>
<li>
<p><strong>Measure first</strong>: Use a GPIO toggle at <code>ticker_job()</code> entry/exit to scope its real execution time under your workload.</p>
</li>
<li>
<p><strong>Leave margin</strong>: Choose <code>EVENT_OVERHEAD_START_US</code> so that even your worst-case <code>ticker_job()</code> time leaves enough prep time for LLL.</p>
</li>
<li>
<p><strong>Optimize</strong>: If <code>ticker_job()</code> is dominating, consider:</p>
<ul>
<li>
<p>Reducing ticker list churn.</p>
</li>
<li>
<p>Disabling unused Bluetooth roles.</p>
</li>
<li>
<p>Using faster CPU frequency for scheduling-critical paths.</p>
</li>
</ul>
</li>
</ul>
<hr>
</body></html>

cvinayak, 2025-08-20:
ChatGPT response to a discussion on discord:
```
Someone asked me the following:

regarding scheduling lll events in iso broadcast (lll_adv, lll_adv_aux, lll_adv_sync, and lll_adv_iso) through the cntr_cmp_set function, i noticed that the events over time do not have the same chronological order. In this case we might have 2 events very close to each other in a way that the first event didn't complete its tasks for example. I need to know what monitor this matter is it the matter to set the lll_event interval adequately or is it something manged by zephyr (i.e. lll_adv event come but lll adv_iso event didn't finish sending the tx in the BIG event)

Should i have some guard on how much duration should each lll event take to finish executing in order not to overlap with the next lll_event in case they are close to each other?

And I replied:

It is the LLL event pipeline processing that has a request-response-preempt design between states/roles (is_abort_cband  abort_cb) that mitigate overlap/collision in the "unreserved" timeline of the event. "reserved" timeline of the event is overlap/collision resolved in the ticker_worker; "unreserved" timeline of the event is overlap/collision resolved using the request-response-preempt design.

The LLL events are soft-realtime, the state/events are to be designed to jitter (positive) upto EVENT_OVERHEAD_START_US, the Radio tx/rx is setup hard-realtime sometime after this jitter value; and during this setup the jitter is assertion checked and the actual value is printed in the assertion. EVENT_OVERHEAD_START_US is the total of all CPU execution latencies (1. previous LLL event Radio tx/rx/done ISR execution, 2. LLL done processing, 3. any new ticker_job ULL_LOW processing, next ticker_worker timeout expiry ULL_HIGH, 4. max. of all LLL event CPU use in the expired ticker callback, 5. minimum time reserved for any h/w counter limitations).  You can use CONFIG_BT_CTLR_PROFILE_ISR=y to get some in-system profiling of all the CPU use duration in the Controller when running applications. 

Can you summarize this in a nice for documentation with diagram?
```

### Zephyr Bluetooth Controller – LLL Event Scheduling and Timing Guarantees

In the Zephyr Bluetooth Controller, Link Layer Low-level (LLL) events (e.g. `lll_adv`, `lll_adv_aux`, `lll_adv_sync`, `lll_adv_iso`) are scheduled via the ticker (`cntr_cmp_set`). Events may appear close together in time, raising concerns about overlap and execution delays. The controller implements multiple layers of scheduling and preemption to ensure correct operation.

---

## Event Timeline Management

### Reserved Timeline

* Managed in the **ticker\_worker** (Upper Link Layer High priority).
* Resolves **hard conflicts** between events before they are dispatched.
* Guarantees no two events are scheduled to start at the same instant.

### Unreserved Timeline

* Managed by the **LLL pipeline** (request–response–preempt design).
* Handles dynamic overlaps and collisions between roles/states when execution overruns occur.
* Uses callbacks (`is_abort_cb`, `abort_cb`) to preempt or abort lower-priority events if higher-priority events need to run.

---

## Event Real-time Characteristics

* **Soft Real-time Phase**:

  * Begins at `EVENT_OVERHEAD_START_US` microseconds before the radio anchor.
  * Allows for jitter in LLL event preparation due to ISR load, ticker jobs, or done callbacks.
* **Hard Real-time Phase**:

  * Radio TX/RX setup occurs after `EVENT_OVERHEAD_START_US`.
  * At this point, jitter tolerance ends; timing is **assertion-checked**.
  * If setup exceeds the margin, an assertion logs the actual measured overhead.

---

## Composition of `EVENT_OVERHEAD_START_US`

This value is a budget that sums the maximum latencies from all sources:

1. ISR execution from the **previous LLL event** (radio done interrupt).
2. **LLL done processing**.
3. New ticker job in **ULL\_LOW**.
4. Timeout expiry in **ticker\_worker ULL\_HIGH**.
5. Maximum LLL event CPU usage during ticker callbacks.
6. Minimum margin for **hardware counter limitations**.

---

## Profiling Tools

Enable in-system profiling with:

```
CONFIG_BT_CTLR_PROFILE_ISR=y
```

This measures and reports ISR execution times within the controller, helping developers tune scheduling margins and verify system behavior.

---

## Diagram: Event Pipeline Timing

```
                |<--------- EVENT_OVERHEAD_START_US --------->|
Time ----------------------------------------------------------->

Previous Event   Done ISR     LLL Done Proc.  Ticker Job   Ticker Worker
       |-------------|-------------|--------------|--------------|
                                      \                               
                                       \   Soft Real-time Phase       
                                        \-----------------------------|
                                                          Radio Setup  
                                                          (Hard RT)
```

---

### Summary

* **No manual guard needed**: Zephyr handles event overlap through ticker scheduling (reserved timeline) and LLL pipeline preemption (unreserved timeline).
* **EVENT\_OVERHEAD\_START\_US** is the safety margin that absorbs jitter and execution latencies.
* Developers can profile real system behavior using `CONFIG_BT_CTLR_PROFILE_ISR` to verify margins under application workloads.

closed 2026-04-10 by github-actions[bot]