zephyrproject-rtos/zephyr #82403: DOC: Documentation build finished with problems, 1532 warnings

butok, 2024-12-02:
**Describe the bug**

Trying to build the Zephyr HTML documentation.
Documentation build finished with problems, 1532 warnings (with warnings treated as errors)

**To Reproduce**
Use Windows 11 OS.
1. Install all documentation tools and requirements according to 
    https://docs.zephyrproject.org/latest/contribute/documentation/generation.html
3. Go to the zephyr doc directory
    `cd \zephyr\doc`
4. Use cmake to configure a Ninja-based build system:
    `cmake -GNinja -B_build .`
6. Enter the build directory
   `cd _build`
8. To generate HTML output, run ninja on the generated build system:
    `ninja html`
10. HTML generation is successful, but it reports that is failed cause of warnings:

```
build finished with problems, 1532 warnings (with warnings treated as errors).

FAILED: CMakeFiles/html C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/CMakeFiles/html
C:\Windows\system32\cmd.exe /C "cd /D C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build && C:\Strawberry\c\bin\cmake.exe -E env "DOXYGEN_EXECUTABLE=C:/Program Files/doxygen/bin/doxygen.exe" DOT_EXECUTABLE=C:/ProgramData/chocolatey/bin/dot.exe DOCS_HTML_DIR=C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/html OUTPUT_DIR=C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/html C:/Users/nxa12205/AppData/Local/Programs/Python/Python310/Scripts/sphinx-build.exe -b html -c C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc -d C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/doctrees -w C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/html.log -t development -j auto -W --keep-going -T C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/src C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/html"
ninja: build stopped: subcommand failed.
```

**Expected behavior**
Successful HTML generation 

**Logs and console output**
```
C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build>ninja html
[0/2] Generating Devicetree bindings documentation...
[1/2] Running Sphinx HTML build...Running Sphinx v8.1.3
loading translations [en]... done
Building Kconfig database...... done
Preparing Doxyfile...
Checking if Doxygen needs to be run...
Running Doxygen...
Syncing Doxygen output...
building [mo]: targets for 0 po files that are out of date
writing output...
building [html]: targets for 2184 source files that are out of date
updating environment: [new config] 4248 added, 0 changed, 0 removed
reading sources... [100%] snippets/xen_dom0/README
WARNING: driversadcadc_ad559x.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\adi,ad559x-adc.rst:21'
WARNING: driversadcadc_max32.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\adi,max32-adc.rst:21'
WARNING: driversadcadc_ambiq.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\ambiq,adc.rst:21'
WARNING: driversadcadc_sam.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\atmel,sam-adc.rst:21'
WARNING: driversadcadc_sam_afec.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\atmel,sam-afec.rst:21'
WARNING: driversadcadc_sam0.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\atmel,sam0-adc.rst:21'
WARNING: driversadcadc_ene_kb1200.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\ene,kb1200-adc.rst:21'
WARNING: driversadcadc_esp32.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\espressif,esp32-adc.rst:21'
WARNING: driversadcadc_gd32.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\gd,gd32-adc.rst:21'
WARNING: driversadcadc_ifx_cat1.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\infineon,cat1-adc.rst:21'
WARNING: driversadcadc_xmc4xxx.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\infineon,xmc4xxx-adc.rst:21'
WARNING: driversadcadc_ite_it8xxx2.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\ite,it8xxx2-adc.rst:21'
WARNING: driversadcadc_ltc2451.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\lltc,ltc2451.rst:21'
WARNING: driversadcadc_max11102_17.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\maxim,max11102.rst:21'
WARNING: driversadcadc_max11102_17.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\maxim,max11103.rst:21'
WARNING: driversadcadc_max11102_17.c not found in zephyr at 'C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\build\dts\api\bindings\adc\maxim,max11105.rst:21'
WARNING: driversadcadc_max11102_17.c not found in zephyr at 
...

...
C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\samples\boards\st\bluetooth\interactive_gui\README.rst: document is referenced in multiple toctrees: ['samples/boards/st/bluetooth/index', 'samples/boards/st/index'], selecting: samples/boards/st/index <- samples/boards/st/bluetooth/interactive_gui/README
C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\samples\boards\st\power_mgmt\adc\README.rst: document is referenced in multiple toctrees: ['samples/boards/st/index', 'samples/boards/st/power_mgmt/index'], selecting: samples/boards/st/power_mgmt/index <- samples/boards/st/power_mgmt/adc/README
C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\samples\boards\st\power_mgmt\blinky\README.rst: document is referenced in multiple toctrees: ['samples/boards/st/index', 'samples/boards/st/power_mgmt/index'], selecting: samples/boards/st/power_mgmt/index <- samples/boards/st/power_mgmt/blinky/README
C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\src\samples\boards\st\power_mgmt\serial_wakeup\README.rst: document is referenced in multiple toctrees: ['samples/boards/st/index', 'samples/boards/st/power_mgmt/index'], selecting: samples/boards/st/power_mgmt/index <- samples/boards/st/power_mgmt/serial_wakeup/README
[... 33 lines cut ...]
Writing evaluated template result to C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\html\_static\js\versions.js
Writing evaluated template result to C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\html\_static\copybutton.js
copying static files: done
copying extra files...
copying extra files: done
copying assets: done
writing output... [100%] snippets/xen_dom0/README
generating indices... genindex done
writing additional pages... gsearch search done
copying images... [  9%] ../../../boards/arm/fvp_baser_aemv8r/doc/images/create-new-configuration-database_database-namecopying images... [  9%] ../../../boards/arm/fvp_baser_aemv8r/doc/images/create-new-configuration-database_shown-in-projcopying images... [  9%] ../../../boards/arm/fvp_baser_aemv8r/doc/images/create-new-model-configuration_model-use-V8R64-copying images... [ 10%] ../../../boards/arm/fvp_baser_aemv8r/doc/images/create-new-launch-configuration_context-menu.jpcopying images... [ 23%] ../../../boards/heltec/heltec_wireless_stick_lite_v3/doc/heltec_wireless_stick_lite_v3_pinout.wcopying images... [ 54%] ../../../boards/shields/sparkfun_carrier_asset_tracker/doc/img/sparkfun_carrier_asset_tracker.wcopying images... [100%] ../../services/zbus/images/zbus_publishing_process_example_HLP.svg
dumping search index in English (code: en)... done
dumping object inventory... done
sphinx-sitemap: sitemap.xml was generated for URL https://docs.zephyrproject.org/latest/ in C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build\html\sitemap.xml
build finished with problems, 1532 warnings (with warnings treated as errors).

FAILED: CMakeFiles/html C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/CMakeFiles/html
C:\Windows\system32\cmd.exe /C "cd /D C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build && C:\Strawberry\c\bin\cmake.exe -E env "DOXYGEN_EXECUTABLE=C:/Program Files/doxygen/bin/doxygen.exe" DOT_EXECUTABLE=C:/ProgramData/chocolatey/bin/dot.exe DOCS_HTML_DIR=C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/html OUTPUT_DIR=C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/html C:/Users/nxa12205/AppData/Local/Programs/Python/Python310/Scripts/sphinx-build.exe -b html -c C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc -d C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/doctrees -w C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/html.log -t development -j auto -W --keep-going -T C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/src C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/html"
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**

 - OS: Windows 11
 - Toolchain: Zephyr SDK 0.17
 - Commit SHA: 065bd322200ccf2f9d717fd258ec966a522ce733

marc-hb, 2025-09-03:
Reproduced with today's commit bfea9cf94f9a - documentation warnings definitely don't fix  themselves.

More annoyingly, even the incremental build has noise now, see more focused #95438 that I just filed. This used to be clean last time I was foolish enough to go and try to fix a typo.

marc-hb, 2025-09-04:
Thanks @kartben for
- #95433 
- #95436